m-labs/artiq #1988: core device execution time variation

jbqubit, 2022-11-07:
# Bug Report

## Issue Details
I observe large variations in the execution time of kernels on the core device (Kasli v1.x). While the standard deviation of the execution time may be small, in a small number of cases there are anomalous long processing delays. That is, the execution time has a long tail, implying non-gaussian statistics. This can be observed in code examples below. 

In my lab's experiment codebase we make heavy use of abstraction. We write modular kernel methods with sufficient built-in slack to reliably run without RTIOUnderflow (eg Doppler cooling, gates, detection). We combine these modular elements to produce large monolithic kernels. Unfortunately, the slack that works for the atomic modules often proves insufficient when many modules are called in succession. We suspect what's happening is a random walk in slack; with a sufficient number of reps  we hit RTIOUnderflow. 

### Actual (undesired) Behavior

Below are tables showing the execution time in ns for 80 successive kernel calls for a variety of kernel methods. At the end of each table are summary statistics that exclude the first two reps. The first two are excluded from the statistics because they are known to be reproducibly slower than successive reps. 

The example code relies on get_rtio_counter_mu() to put a bound on kernel execution time. I note that the non-determinism of `get_rtio_counter_mu()` method is [documented](https://m-labs.hk/artiq/manual/core_drivers_reference.html#artiq.coredevice.core.Core.get_rtio_counter_mu). In my test I put a bound on this in test_dummy(). The balance of uncertainty seems attributable to the core device microprocessor. 

Another manifestation of this is in https://github.com/m-labs/artiq/issues/1987. 

```
$ artiq_run repository/core_device_process_durations_jwb.py
### test_dummy ###
  0:  	1296,   1:   	312,   2:   	320,   3:   	320,   4:   	320,   5:   	320,   6:   	320,   7:   	320,
  8:   	320,   9:   	320,  10:   	320,  11:   	312,  12:   	320,  13:   	320,  14:   	312,  15:   	320,
 16:   	320,  17:   	320,  18:   	320,  19:   	320,  20:   	320,  21:   	320,  22:   	320,  23:   	312,
 24:   	320,  25:   	320,  26:   	320,  27:   	320,  28:   	320,  29:   	320,  30:   	312,  31:   	320,
 32:   	320,  33:   	312,  34:   	320,  35:   	320,  36:   	320,  37:   	320,  38:   	320,  39:   	312,
 40:   	312,  41:   	320,  42:   	320,  43:   	320,  44:   	320,  45:   	312,  46:   	320,  47:   	320,
 48:   	320,  49:   	320,  50:   	320,  51:   	320,  52:   	320,  53:   	320,  54:   	320,  55:   	320,
 56:   	320,  57:   	320,  58:   	320,  59:   	320,  60:   	320,  61:   	320,  62:   	312,  63:   	320,
 64:   	320,  65:   	320,  66:   	320,  67:   	320,  68:   	320,  69:   	320,  70:   	320,  71:   	320,
 72:   	320,  73:   	320,  74:   	320,  75:   	312,  76:   	320,  77:   	320,  78:   	312,  79:   	320,

mean:   	319, std:     	3, max deviation:     	8, max frac deviation: 0.025

### test_ttl_pulse ###
  0:  	4320,   1:   	920,   2:   	904,   3:   	896,   4:   	896,   5:   	896,   6:   	904,   7:   	904,
  8:   	904,   9:   	896,  10:   	904,  11:   	896,  12:   	896,  13:   	904,  14:   	896,  15:   	896,
 16:   	896,  17:   	904,  18:   	896,  19:   	896,  20:   	904,  21:   	896,  22:   	896,  23:   	904,
 24:   	896,  25:   	904,  26:   	904,  27:   	896,  28:   	896,  29:   	904,  30:   	904,  31:   	896,
 32:   	904,  33:   	904,  34:   	904,  35:   	896,  36:   	896,  37:   	904,  38:   	904,  39:   	896,
 40:   	896,  41:   	896,  42:   	896,  43:   	896,  44:   	896,  45:   	904,  46:   	904,  47:   	896,
 48:   	896,  49:   	904,  50:   	904,  51:   	896,  52:   	896,  53:   	896,  54:   	904,  55:   	896,
 56:   	896,  57:   	904,  58:   	904,  59:   	896,  60:   	904,  61:   	904,  62:   	904,  63:   	896,
 64:   	896,  65:   	904,  66:   	904,  67:   	904,  68:   	896,  69:   	896,  70:   	896,  71:   	904,
 72:   	904,  73:   	896,  74:   	896,  75:   	904,  76:   	904,  77:   	896,  78:   	896,  79:   	896,

mean:   	900, std:     	4, max deviation:     	8, max frac deviation: 0.009

### test_reset ###
  0:   5814680,   1:   4844920,   2:   4875376,   3:   4876640,   4:   4929560,   5:   5809232,   6:   4876648,   7:   4878136,
  8:   4876536,   9:   5859448,  10:   4915488,  11:   4825176,  12:   4876984,  13:   4879168,  14:   5910032,  15:   4876536,
 16:   4825952,  17:   4877464,  18:   4930624,  19:   5858080,  20:   4878960,  21:   4875568,  22:   4826144,  23:   5911616,
 24:   4863920,  25:   4872912,  26:   4882088,  27:   4876640,  28:   5859728,  29:   4876536,  30:   4878256,  31:   4876456,
 32:   4930040,  33:   5807816,  34:   4876368,  35:   4878208,  36:   4877904,  37:   5859864,  38:   4863872,  39:   4877456,
 40:   4877024,  41:   4876944,  42:   5859152,  43:   4879648,  44:   4876496,  45:   4877056,  46:   4928392,  47:   5808752,
 48:   4878328,  49:   4876344,  50:   4877624,  51:   5858760,  52:   4913240,  53:   4879072,  54:   4876544,  55:   4826096,
 56:   5911192,  57:   4876440,  58:   4878592,  59:   4877904,  60:   4876592,  61:   5859632,  62:   4876592,  63:   4877936,
 64:   4861432,  65:   4945680,  66:   5807392,  67:   4876568,  68:   4877800,  69:   4878480,  70:   5857784,  71:   4862752,
 72:   4878096,  73:   4877064,  74:   4878816,  75:   5858208,  76:   4878056,  77:   4876640,  78:   4877704,  79:   4930336,

mean:   5079931, std:	395087, max deviation:   1086440, max frac deviation: 0.214

### test_simple_float ###
  0:  	1704,   1:   	336,   2:   	320,   3:   	320,   4:   	320,   5:   	320,   6:   	320,   7:   	320,
  8:   	320,   9:   	320,  10:   	312,  11:   	320,  12:   	320,  13:   	312,  14:   	312,  15:   	320,
 16:   	320,  17:   	312,  18:   	312,  19:   	320,  20:   	320,  21:   	320,  22:   	320,  23:   	312,
 24:   	320,  25:   	320,  26:   	320,  27:   	320,  28:   	320,  29:   	320,  30:   	320,  31:   	320,
 32:   	320,  33:   	320,  34:   	320,  35:   	320,  36:   	312,  37:   	320,  38:   	312,  39:   	312,
 40:   	312,  41:   	320,  42:   	320,  43:   	320,  44:   	320,  45:   	320,  46:   	312,  47:   	312,
 48:   	320,  49:   	312,  50:   	312,  51:   	320,  52:   	320,  53:   	312,  54:   	312,  55:   	320,
 56:   	320,  57:   	312,  58:   	312,  59:   	320,  60:   	312,  61:   	312,  62:   	320,  63:   	312,
 64:   	312,  65:   	320,  66:   	320,  67:   	320,  68:   	320,  69:   	320,  70:   	320,  71:   	320,
 72:   	320,  73:   	320,  74:   	312,  75:   	320,  76:   	320,  77:   	312,  78:   	312,  79:   	320,
[... 173 lines cut ...]

    @kernel
    def test_dataset(self):
        # Note that set_dataset is decorated with @rpc(flags={"async"})
        self.set_dataset('ds0', 1.1, broadcast=True)
        return self.mindelay

```
### Expected Behavior

* Core device execution time variation is manageable if it's bounded. In ARITQ7 it does not appear to be well bounded. 
* Presently we add huge delays to each experiment repetition to patch over this class of problem. This solution comes at the cost of much reduced experiment duty cycle. 
* Subjectively this problem seems more prevalent in ARTIQ 7 than previous ARTIQ releases. Is it related to the switch to VexRiscv from mor1kx? Are there benchmarks for execution variation in the ARTIQ codebase? It would be helpful to track this in order to detect performance regressions.  

### Questions
* Why are the first 1 (or 2) reps of each rep of a loop so much slower than subsequent reps?
* Would a bare VexRISCv uP have deterministic execution? 
* What are the leading factors that cause the ARTIQ MiSOC to suffer from non-determinism? 

### System

* Operating System: Ubuntu 
* ARTIQ version: latest commit of tag version 7
* Environment is nix shell
* Hardware involved: Kasli v1.x

jbqubit, 2022-11-09:
I assume you beam BRAM not SRAM.

> Can you use ARTIQ_DUMP_ELF to get the binary for the largest kernel in your experiment and see if it could fit?

The constraints for code complied to ELF are extreme. 
* raise ValueError("Experiment entry point must be a kernel")
* raise ValueError("Experiment must not use RPC")
I can't refactor my code base subject to these constraints. I depend too much on many higher level ARTIQ features.

I can see this a problem inherent in SDRAM ([link](https://arxiv.org/pdf/1810.07059.pdf)). Have you looked into using [RLDRAM](https://en.wikipedia.org/wiki/RLDRAM)? The arxiv [paper](https://arxiv.org/pdf/1810.07059.pdf) describes a memory controller design that predictably manages accesses to the RLDRAM.

jbqubit, 2022-11-10:
> The constraints you mention are specific to artiq_compile and not ARTIQ_DUMP_ELF, the latter has no restrictions.

I'm not familiar with [this](https://github.com/m-labs/artiq/blob/56e6b1428c954108f61f395758c77e9f3517e33a/artiq/compiler/targets.py) feature of ARTIQ. Please give an example of how of how to use it to do an ELF dump. There are a lot of options for the Target class... 

> The problem is inherent to all DRAM, the paper only mentions a reduction of the latency variation, not an elimination. And the vast majority of boards don't support RLDRAM.

A 10X reduction in latency variation would be a big win. Sinara builds it's own boards so that's not necessarily a roadblock. I guess it would entail writing another memory controller. @gkasprow Have you worked with RLDRAM?

dnadlinger, 2022-11-10:
> Please give an example of how of how to use it to do an ELF dump.

Just set the `ARTIQ_DUMP_ELF` environment variable to a string (which will become the file name), either when calling `artiq_run`, or when starting `artiq_master` (the latter IIRC will just write to the result file directory).

jbqubit, 2022-11-15:
I followed the advice of @dnadlinger for a typical experiment in my lab. Here's the size. 

```
$ ls -lh
total 1.8M
-rw-r--r--   1 iirabi iirabi  33K Nov 14 17:38 artiq_dump.elf
```

How much BRAM is typically available on a Kasli running ARTIQ? Is there an established path for fully loading a kernel into BRAM?

jbqubit, 2022-11-15:
Is there a way to do an initial test and see how the latency variation improves? Has the latency variation for Zynq been evaluated yet? It may be that both core device variants would benefit from use of BRAM for sufficiently simple kernels.