m-labs/artiq #1994: AD9910: single tone profile not changed properly

MYCyw, 2022-11-13:
# Bug Report

## One-Line Summary

After changing single tone profile, the parameters of DDS  don't correspond with that of the selected profile.

## Issue Details

### Steps to Reproduce

```python
from artiq.experiment import *

class DDSTestSet(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.cpld = self.get_device("urukul3_cpld")
        self.dds = self.get_device("urukul3_ch0")

    @kernel
    def run(self):
        self.core.reset()
        self.cpld.init()
        self.dds.init()
        delay(1 * us)
        # set 8 profiles
        for profile_n in range(0, 8):
            self.cpld.set_profile(profile_n)
            self.dds.set(
                frequency=profile_n * 11 * MHz,
                amplitude=profile_n * 0.11,
                profile=profile_n,
                ram_destination=-1,
            )
            delay(1 * us)
            profile_tuple = self.dds.get(profile=profile_n)
            self.get_profile(profile_tuple)
            delay(1 * ms)

        # turn on dds
        self.dds.sw.on()
       # change profile 
        for profile_n in range(0, 8):
            self.cpld.set_profile(profile_n)
            delay(1 * ms)
            profile_tuple = self.dds.get(profile=profile_n)
            delay(1 * ms)
            self.get_profile(profile_tuple)
            delay(3 * ms)

        self.dds.sw.off()

    @rpc(flags={"async"})
    def get_profile(self, profile_tuple):
        print(profile_tuple)

```

### Expected Behavior

DDS parameters should change corresponding to the selected profile.

### Actual (undesired) Behavior

Profile 2 and profile 6 get wrong parameters: profile 2 get the same parameters as profile 3, and profile 6 get the same parameters as profile 7.

[image]

[image]

I also find that if I change the profile to some other profile before change it to the intended one can cause different results.

If I change the profile to 0, 3 or 7 in advance, then the result is as expected.

```python
from artiq.experiment import *

class DDSTestSet(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.cpld = self.get_device("urukul3_cpld")
        self.dds = self.get_device("urukul3_ch0")

    @kernel
    def run(self):
        self.core.reset()
        self.cpld.init()
        self.dds.init()
        delay(1 * us)
        # set 8 profiles
        for profile_n in range(0, 8):
            self.cpld.set_profile(profile_n)
            self.dds.set(
                frequency=profile_n * 11 * MHz,
                amplitude=profile_n * 0.11,
                profile=profile_n,
                ram_destination=-1,
            )
            delay(1 * us)
            profile_tuple = self.dds.get(profile=profile_n)
            self.get_profile(profile_tuple)
            delay(1 * ms)

        # turn on dds
        self.dds.sw.on()

        # change profile
        for profile_n in range(0, 8):
            self.cpld.set_profile(7) # change profile to 0, 3 or 7 before the following line
            self.cpld.set_profile(profile_n)
            delay(1 * ms)
            profile_tuple = self.dds.get(profile=profile_n)
            delay(1 * ms)
            self.get_profile(profile_tuple)
            delay(3 * ms)

        self.dds.sw.off()

    @rpc(flags={"async"})
    def get_profile(self, profile_tuple):
        print(profile_tuple)

```
[image]

* However, if I change it to profile 1 before, then the result is: [image]

* If change it to profile 2: [image]

* Change it to profile 4: [image]

* Profile 5: [image]

* Profile 6: [image]

### Your System (omit irrelevant parts)

* Operating System: DebianBookworm
* ARTIQ version: 7.8125.d039dd4
* Version of the gateware and runtime loaded in the core device: 7.0.beta
* Hardware involved: urukul

MYCyw, 2022-12-08:
Solved by adding `self.cpld.io_update.pulse_mu(8)` after `self.cpld.set_profile(profile_n)`

closed 2022-12-08 by MYCyw

dhslichter, 2022-12-08:
@MYCyw you should double-check the timing on this. The datasheet claims that the changing of the profile pins alone should be sufficient to update the profiles, but our experience at NIST with the AD9914 (which I am guessing uses similar logic) was that if you attempted to change more than one profile pin at once it didn't work properly. There is also the issue of making sure the profile pin changes meet the setup timing criteria based on SYNC_CLK. See p.41 of the datasheet https://www.analog.com/media/en/technical-documentation/data-sheets/ad9910.pdf

clayton-ho, 2023-01-13:
Ran into this problem recently despite being able to switch between 6 profiles quickly during a pulse sequence, and found it was because we were profile switching in a regular kernel function. After testing, we believe that it's not an issue if you record any sequences involving profile switching onto core DMA.

For reference, we have AD9910 Urukuls (hardware rev 1.5) in 2x EEM mode.

dnadlinger, 2023-01-13:
@clayton-ho: Not sure this makes much sense on the face of it – regular event submission and DMA should lead to exactly the same output (if it successfully executes on both).