m-labs/artiq #2022: Long range DRTIO: timeout attempting to get remote buffer space

marmeladapk, 2022-12-08:
# Bug Report

## One-Line Summary

I connected two Kaslis with a long fiber (~20 km master->satellite, ~25 km satellite->master) and while the link setups mostly correctly I get this error: `timeout attempting to get remote buffer space`.

## Issue Details

AFAICT the link works fine and I can run an experiment which changes state on DIO on both Kaslis. Latency between edges on DIO is around 98.3 us. Only when I try to use loops to introduce more events then I get `write overflow` on satellite and `timeout attempting to get remote buffer space` on master.

Master logs (setup):
```
[    13.936935s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[    13.943128s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 0
[    14.150352s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] error(s) found (0x04):
[    14.156203s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] timeout attempting to get remote buffer space
```

On satellite nothing seems out of the ordinary until I run an experiment with more events.

Both fibers have been tested with IBERT to check if digital communication works.

Can timeout be increased? Can it have some negative effects in other areas?

Paging @dhslichter, as he was also interested in long range DRTIO.

### Your System (omit irrelevant parts)

* Operating System: Linux
* ARTIQ version: 5985595
* Version of the gateware and runtime loaded in the core device: 5985595
* Hardware involved:
  * 2x Kasli v2.0.1
  * 2x DIO-BNC
  * 20 and 25 km fiber
  * 10G fiber transceiver

jordens, 2022-12-08:
Why are the fibers asymmetric? Is that the BiDi refractive index difference or are they physically different?

marmeladapk, 2022-12-08:
@jordens They are physically different. That's what was available to me.

dhslichter, 2022-12-08:
We are interested in long distance DRTIO but we would use matched fibers and transceivers for this. With 5 km difference in fiber lengths, I could imagine that would make all kinds of things not work properly...

marmeladapk, 2022-12-08:
@dhslichter AFAICT link assymetry shouldn't play any part in here. I didn't see any latency measurements in code (but I also wasn't looking very hard), and I think that both master and satellite care only about round trip time. For this error timeout seems to be hardcoded: [rt_controller_master.py#L106](https://github.com/m-labs/artiq/blob/master/artiq/gateware/drtio/rt_controller_master.py#L106).

marmeladapk, 2022-12-20:
Increasing the timeout fixed the issue and after a while I didn't see anything wrong in the logs.