m-labs/artiq #2023: `set_dataset` with `archive=False` throws `KeyError` when accessing

ThorstenGroh, 2022-12-08:
Running 
```python
class Test(EnvExperiment):
    def prepare(self):
        self.set_dataset("x", 1, archive=False)

    def run(self):
        print(self.get_dataset("x"))
```
throws a `KeyError`
```python
root:Terminating with exception (KeyError: "'x'")
  File "C:\tools\miniconda3\envs\artiq-7\lib\site-packages\artiq\master\worker.py", line 229, in _handle_worker_requests
    data = func(*obj["args"], **obj["kwargs"])
  File "C:\tools\miniconda3\envs\artiq-7\lib\site-packages\artiq\master\databases.py", line 62, in get
    return self.data.raw_view[key][1]
KeyError: "'x'"
```
If I run the same code with `archive=True`, which is the default everything works as expected.

I am running ARTIQ-7 version 7.8123.3038639.

thomasfire, 2023-01-09:
> Arguably, if a value should neither be broadcasted, persisted, or archived, then it should not be a dataset.

Shouldn't be the `set` function be symmetrical to `get` function? `get` function clearly says it wouldn't expect NOP in any case:
```py
    def get(self, key, archive=False):
        if key in self.local:
            return self.local[key]
        
        data = self.ddb.get(key)
        if archive:
            if key in self.archive:
                logger.warning("Dataset '%s' is already in archive, "
                               "overwriting", key, stack_info=True)
            self.archive[key] = data
        return data
```

ThorstenGroh, 2023-04-27:
This was a workaround to get our own data storage running. Rewrote `set_dataset` for my needs now. Closing this.

closed 2023-04-27 by ThorstenGroh