m-labs/artiq #2028: Satellite system doesn't get uplink clock.

nkrackow, 2022-12-13:
# Bug Report

## One-Line Summary

Building a bitstream for a satellite with an ARTIQ version after 47581e0de9ec135a82a1448d8966111dc03545ee leads to the satellite not getting a link to the master. 

## Issue Details

I'm pretty sure 0775ae1c191d796c5ea9d177be3723d4c528a897, 8f3d06a51592b21d38583f67aaf691c34460d1ea are the culprit. 
If I find time tomorrow I'll check this.
The info output about the io expander is a fix for an unrelated hardware change that is not in upstreamed in ARTIQ yet. 
I verified that it has nothing to do with this problem by testing a verbatim ARTIQ earlier but I don't have the console out of that right now.

### Expected Behavior

```
[     0.000003s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005908s]  INFO(satman): software ident 7.0.beta;stockholmsatellite
[     0.012445s]  INFO(satman): gateware ident 7.0.beta;stockholmsatellite
[     0.029525s]  INFO(board_misoc::io_expander): MCP23017 io expander 0 not found. Checking for PCA9539.
[     0.058069s]  INFO(board_misoc::io_expander): MCP23017 io expander 1 not found. Checking for PCA9539.
[     0.456029s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     2.408998s]  INFO(board_artiq::si5324):   ...locked
[     2.797873s]  INFO(satman): uplink is up, switching to recovered clock
[     2.830818s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     4.559314s]  INFO(board_artiq::si5324):   ...locked
[     7.885235s]  INFO(board_artiq::si5324::siphaser): calibration successful, lead: 198, width: 433 (347deg)
[     8.235495s]  INFO(satman): TSC loaded from uplink
[     8.307408s]  INFO(satman): rank: 1
[     8.309507s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; 3: 3 0; }
```

### Actual (undesired) Behavior

```
[     0.000003s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005908s]  INFO(satman): software ident 7.0.beta;stockholmsatellite
[     0.012445s]  INFO(satman): gateware ident 7.0.beta;stockholmsatellite
[     0.029525s]  INFO(board_misoc::io_expander): MCP23017 io expander 0 not found. Checking for PCA9539.
[     0.058069s]  INFO(board_misoc::io_expander): MCP23017 io expander 1 not found. Checking for PCA9539.
[     0.456029s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     2.376948s]  INFO(board_artiq::si5324):   ...locked
```
It never gets an uplink connection after this.

### Setup

No flash config file on the satellite.

nkrackow, 2022-12-13:
https://github.com/m-labs/artiq/commit/8f3d06a51592b21d38583f67aaf691c34460d1ea#diff-0c99ae49963f6b70bf1690eaba102795097c1089358b20dcc83ee82422cca57cR120-R121
I don't fully understand this comment but maybe that has something to do with it? @dnadlinger

dnadlinger, 2022-12-13:
Your guess isn't unreasonable, but I'm also not sure how the commits would be related – AFAIK they should only affect the master.

dnadlinger, 2022-12-13:
What are the master's clock settings (that this satellite is supposed to connect to?).

dnadlinger, 2022-12-13:
To be clear, the changes only (intentionally) affect Kasli v2 when used as a master, making it such that internal/external clocks can actually be selected at runtime.

jordens, 2022-12-13:
How do you come to the impression that the changes are limited to standalone/master? Maybe this was tested.
https://github.com/m-labs/artiq/commit/0775ae1c191d796c5ea9d177be3723d4c528a897#diff-3ed5c793f81f4728447227870960117251dafbb1869a0ff428661c7d97fd5384R221
in the context of
https://github.com/m-labs/artiq/blob/874d298ceb68ffa776592f855365ca5e2e885575/artiq/firmware/satman/main.rs#L452
and
https://github.com/m-labs/artiq/blob/874d298ceb68ffa776592f855365ca5e2e885575/artiq/firmware/satman/main.rs#L521
looks like an impact to me.

dnadlinger, 2022-12-17:
Yep, that was just a failure of Find All Uses (and apparently, also my subsequent `rg`ing).

dnadlinger, 2022-12-17:
Reopened this until somebody can confirm the issue fixed, as I can't test this locally until next week.

nkrackow, 2022-12-19:
I've now tested 1dd0d3432c59912411cfde10adf08a4d57977d45 on a Kasli V2.0.2 satellite and master and all seems to work as before on both sides.

commit ceabeb8d (2022-12-17): firmware: Fix Si5324 initialisation for satellites

Commit 740543d4e28424 had unintentionally broken DRTIO
satellites, as si5324::setup is also used there. This
imports setup_si5324_as_synthesizer() from artiq-zynq,
where the input selection was already explicitly done.

GitHub: Fixes #2028.
commit 1db3a42a (2022-12-18): firmware: Fix Si5324 initialisation for satellites

Commit 740543d4e28424 had unintentionally broken DRTIO
satellites, as si5324::setup is also used there. This
imports setup_si5324_as_synthesizer() from artiq-zynq,
where the input selection was already explicitly done.

GitHub: Fixes #2028.
closed 2022-12-19 by dnadlinger