m-labs/artiq #2041: Idle kernel runs without reboot after flashing

thomasfire, 2023-01-03:
# Bug Report

## One-Line Summary

Idle kernel runs without reboot after flashing, which is not desired behavior.

## Issue Details

### Steps to Reproduce

1. Compile and flash idle kernel
2. run any experiment

### Expected Behavior

Idle kernel is not being run until Kasli to be rebooted.

### Actual (undesired) Behavior

Idle kernel is being run.

dnadlinger, 2023-01-03:
Is it clear that this is undesirable? The ability to update idle kernels on the fly seems quite useful to me (though we don't typically use idle kernels for much, so this isn't a strongly held opinion).

SimonRenblad, 2024-09-19:
Fixed by #2569 and https://git.m-labs.hk/M-Labs/artiq-zynq/pulls/325. I think we can close this.