m-labs/artiq #2067: Satman: with stack guard, flushing L2 cache causes a panic

Spaqin, 2023-03-21:
# Bug Report

## One-Line Summary

With the stack guard enabled, trying to clear L2 cache causes an ``Exception(LoadFault)``.

## Issue Details

### Steps to Reproduce

1. Step 1. Add a call to ``board_misoc::cache::flush_L2_cache()`` anywhere in Satman firmware.
2. Step 2. Compile, run.
3. Step 3. Observe ``Exception(LoadFault)``.

### Expected Behavior

L2 cache is cleared, no panic.

### Actual (undesired) Behavior

With L2 cache flush inserted right after gateware ident:

```
Booting from flash...
Starting firmware.
[     0.000007s]  INFO(satman): ARTIQ satellite manager starting...
[     0.005886s]  INFO(satman): software ident 8.0.beta;tst-sat-1.1
[     0.011799s]  INFO(satman): gateware ident 8.0.beta;tst-sat-1.1
@ 0x40000b80
+0000: 00052603 fff58593 00450513 fe059ae3
+0010: e0003537 00052503 0ff57513 00100593
+0020: 00b50463 6880306f 4001a537 06052503
+0030: 00300593 08b56663 40016537 b0850513
panic at satman/main.rs:637:5: exception Exception(LoadFault) at PC 0x40000b80, trap value 0x4001b000
```

Panics.
If ``pmp::init_stack_guard`` is commented out, the cache flush works fine.

Implementing ``boot::start_user`` as runtime does (and cache flushes on runtime fine) does not alleviate the issue either.

### Your System (omit irrelevant parts)

* Hardware involved: Kasli 1.1

occheung, 2023-03-28:
Just wondering if this is similar to #1775. (In the sense of firmware size vs address range touched by flushing L2)

Spaqin, 2023-04-20:
> Just wondering if this is similar to #1775. (In the sense of firmware size vs address range touched by flushing L2)

I somehow missed your comment :sweat_smile: yeah that was exactly it