m-labs/artiq #2100: editable installation bug for pip 23.1.2

fanmingyu212, 2023-06-01:
# Bug Report

## One-Line Summary

Imports do not work after `pip install -e artiq` with pip 23.1.2.

## Issue Details

### Steps to Reproduce

1. Clone `artiq` to a local folder.
2. Run `pip install -e artiq` in the parent folder.
3. In IPython, run `import artiq` and `artiq.__artiq_dir__`. 

### Expected Behavior

It outputs the installation directory. This works in pip 23.0.1.

### Actual (undesired) Behavior

It raises an `AttributeError`. Maybe this is raised to 

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: 7

thomasfire, 2023-06-02:
But do we support the installation via pip? I never saw it on our manuals as a way to install artiq... What is the situation with conda?

fanmingyu212, 2023-06-02:
> But do we support the installation via pip? I never saw it on our manuals as a way to install artiq... What is the situation with conda?

From a user's perspective I think supporting the most popular python package index may be a good idea. It is definitely important for us: Conda often does not have the latest packages and pip and Conda don't work great together. Historically I think artiq has unofficially supported pip for a long time, and maybe it is a good idea to keep doing so?

fanmingyu212, 2023-09-15:
This was due to implementation of PEP 517 and an error in our configuration. ARTIQ works fine with pip install.

closed 2023-09-15 by fanmingyu212