m-labs/artiq #2112: SU Servo initialisation problems together with stand-alone Urukul

MagicCreed, 2023-06-09:
# Bug Report

## One-Line Summary

Problems initialising stand-alone Urukul AD9910 with SU-Servo in the same kernel call.

## Issue Details

### Steps to Reproduce
Ran the following block of code
```py
class DDSTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul2_ch0")
        self.setattr_device("urukul2_cpld")

        self.setattr_device("suservo0")
        self.setattr_device("urukul0_cpld")
        self.setattr_device("suservo0_ch0")

    @kernel
    def run(self):
        self.core.reset()

        ################### Standalone Urukul ###################
        self.urukul2_cpld.init()
        self.urukul2_ch0.init()
        self.urukul2_ch0.sw.on()
        self.urukul2_ch0.set_att(0.0)
        self.urukul2_ch0.set(frequency=230 * MHz, phase=0.0, amplitude=1.0)
        #####################################################

        ################### SUServo Urukul ###################
        self.suservo0.init()
        self.suservo0.set_config(enable=1)

        self.suservo0.set_pgia_mu(0, 1)
        self.urukul0_cpld.set_att(0, 0.0)

        self.suservo0_ch0.set_iir(profile=0, adc=0, kp=0.05, ki=-10000.0, g=0.0)
        self.suservo0_ch0.set_dds(profile=0, frequency=70 * MHz, offset=-0.15)
        self.suservo0_ch0.set(en_out=1, en_iir=1, profile=0)
        #####################################################
```
### Expected Behavior

Both Urukuls should initialise and output their respective set frequencies.

### Actual (undesired) Behavior
```
ERROR:worker(4178,DDS_test.py):root:While compiling <repository>\scratch\DDS_test.py
<synthesized>:1:1-1:62: error: host object has an attribute 'io_update' of type <instance artiq.coredevice.urukul._RegIOUpdate {
                __objectid__: numpy.int32
        }>, which is different from previously inferred type <instance artiq.coredevice.ttl.TTLOut {
                __objectid__: numpy.int32
        }> for the same attribute
`<artiq.coredevice.urukul.CPLD object at 0x000001814E69A260>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\scratch\DDS_test.py:15:1: note: expanded from here
def run(self):
^
```
If the SU Servo code block is swapped with the stand-alone Urukul such that the former is run first , we get the following error message instead:
```
WARNING:worker(4181,DDS_test.py):root:While compiling <repository>\scratch\DDS_test.py
<synthesized>:1:2-1:65: warning: object <artiq.coredevice.ad9910.AD9910 object at 0x000002405D26ADA0> of type <instance artiq.coredevice.ad9910.AD9910> does not declare attribute(s) 'sw' as kernel invariant, but other objects of the same type do; the invariant annotation on other objects will be ignored
[`<artiq.coredevice.ad9910.AD9910 object at 0x000002405D26ADA0>`, `<artiq.coredevice.ad9910.AD9910 object at 0x000002405D26AEF0>`]
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\scratch\DDS_test.py:15:1: note: expanded from here
def run(self):
^
<artiq>\coredevice\suservo.py:108:19-108:29: note: while inferring a type for an attribute 'ddses' of a host object
            dds = self.ddses[i]
                  ^^^^^^^^^^
WARNING:worker(4181,DDS_test.py):root:While compiling <repository>\scratch\DDS_test.py
<synthesized>:1:67-1:130: warning: object <artiq.coredevice.ad9910.AD9910 object at 0x000002405D26AEF0> of type <instance artiq.coredevice.ad9910.AD9910> does not declare attribute(s) 'sw' as kernel invariant, but other objects of the same type do; the invariant annotation on other objects will be ignored
[`<artiq.coredevice.ad9910.AD9910 object at 0x000002405D26ADA0>`, `<artiq.coredevice.ad9910.AD9910 object at 0x000002405D26AEF0>`]
                                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\scratch\DDS_test.py:15:1: note: expanded from here
def run(self):
^
<artiq>\coredevice\suservo.py:108:19-108:29: note: while inferring a type for an attribute 'ddses' of a host object
            dds = self.ddses[i]
                  ^^^^^^^^^^
WARNING:worker(4181,DDS_test.py):root:While compiling <repository>\scratch\DDS_test.py
<synthesized>:1:1-1:64: warning: object <artiq.coredevice.ad9910.AD9910 object at 0x000002405D26ADA0> of type <instance artiq.coredevice.ad9910.AD9910> does not declare attribute(s) 'sw' as kernel invariant, but other objects of the same type do; the invariant annotation on other objects will be ignored
`<artiq.coredevice.ad9910.AD9910 object at 0x000002405D26ADA0>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\scratch\DDS_test.py:15:1: note: expanded from here
def run(self):
^
<artiq>\coredevice\suservo.py:308:15-308:23: note: while inferring a type for an attribute 'dds' of a host object
        ftw = self.dds.frequency_to_ftw(frequency)
              ^^^^^^^^
ERROR:worker(4181,DDS_test.py):root:While compiling <repository>\scratch\DDS_test.py
<synthesized>:1:2-1:65: error: host object has an attribute 'sync_data' of type <instance artiq.coredevice.ad9910.SyncDataUser {
                __objectid__: numpy.int32
        }>, which is different from previously inferred type <instance artiq.coredevice.ad9910.SyncDataEeprom {
                __objectid__: numpy.int32
        }> for the same attribute
[`<artiq.coredevice.ad9910.AD9910 object at 0x000002405D26ADA0>`, `<artiq.coredevice.ad9910.AD9910 object at 0x000002405D26AEF0>`]
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\scratch\DDS_test.py:15:1: note: expanded from here
def run(self):
^
```

If the SU Servo block of code and the stand-alone Urukul block of code are wrapped in their own kernel functions and called separately, no error occurs.
### Our System

* Operating System: Windows 10 22H2
* ARTIQ version: ARTIQ v7.8143.75d75cc
* Conda packages: [packagelist.txt](https://github.com/m-labs/artiq/files/11703191/packagelist.txt)
* Hardware involved: Kali-SoC, 1xUrukul AD9910 (v1.5.3), 2 x Urukul AD9910 (v1.5.3) connected to 1x Sinara 5108 Sampler (v2.3) in a SUServo

jordens, 2023-06-09:
https://github.com/m-labs/artiq/issues/1207

dnadlinger, 2023-06-09:
This is because both the suservo code and the normal Urukul code use the AD9910 class, but with different internal types. The ARTIQ compiler can't handle this, but you can just make a subclass for the suservo variant and use it for your suservo device DB entries. We typically do something like this:

```py
class AD9910(artiq.coredevice.ad9910.AD9910):
    passs
```

(though these days, we have a non-trivial version of this for suservo CPLD readback support)

MagicCreed, 2023-06-13:
Hi @dnadlinger,

We tried implementing the subclass for the AD9910 such that the suservo variant calls it instead of AD9910. But still run into this issue: 
```py
ERROR:worker(4178,DDS_test.py):root:While compiling <repository>\scratch\DDS_test.py
<synthesized>:1:1-1:62: error: host object has an attribute 'io_update' of type <instance artiq.coredevice.urukul._RegIOUpdate {
                __objectid__: numpy.int32
        }>, which is different from previously inferred type <instance artiq.coredevice.ttl.TTLOut {
                __objectid__: numpy.int32
        }> for the same attribute
`<artiq.coredevice.urukul.CPLD object at 0x000001814E69A260>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<repository>\scratch\DDS_test.py:15:1: note: expanded from here
def run(self):
```

I'm not sure if this is because we created the class wrong but would you possible provide an example of how this was solved? Will we need to have a new subclass for the CPLD as well?

MagicCreed, 2023-06-13:
Thank you for the help, we have now fixed the problem by creating a subclass for the suservo urukuls. We followed the instruction in the following forums:
https://forum.m-labs.hk/d/273-using-su-servo-mode-and-freestanding-urukul-in-the-same-experiment/3

closed 2023-06-13 by MagicCreed