m-labs/artiq #2116: SED: Lane exhaustion even with clearing RTIO via core.reset(), when too many events were scheduled to the future

systemofapwne, 2023-06-13:
# Bug Report

## One-Line Summary

When scheduling multiple events to the far future + jumping back, a `core.reset()` does not seem to clear future FIFO events

## Issue Details
In our system, we have a TTLOut, that shall go low ~5 minutes after it has been set to high. If it gets set to high again during that timeout, the 5 minute timer shall begin from the beginning again.
Our implementation looks similar to this:
```python
@kernel
def run(self):
    self.core.reset()   # Clear FIFO
    self.ttl0.on()
    now_t = now_mu()
    delay(5*60*s)       # Travel to the future
    self.ttl0.off()
    at_mu(now_t)        # Travel back to present
    
    # Do other stuff now
```
However running this kernel multiple times ends up in exhausting lanes of the FIFO, even though we expect `core.reset()` to clear FIFO.
The scheduled events (`ttl0.off()`) clearly do not get wiped and are even getting executed in the future nonetheless.

When this happens, the dashboard shows
`artiq.coredevice.comm_kernel: sequence error(s) reported during kernel execution`
and corelog shows 
`ERROR:firmware.runtime.rtio_mgt:RTIO sequence error involving channel 0x0005:unknown`

### Steps to Reproduce
Here is a minimalistic example experiment, that triggers this issue
```python
class Minimal(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl5")

    @kernel
    def run(self):
        ttl = self.ttl5
        ttl.output()
        for i in range(0,9):                       # Consumes all available 8 FIF lanes and throws error, when trying to add a 9th event.
            print(i)
            self.core.reset()   # Clear FIFO
            self.core.break_realtime()
            
            # Some sequence now, including setting the ttl high
            delay(0.1*s)
            ttl.on()
            delay(0.5*s)

            now_t = now_mu()
            delay(10*s)         # Jump far into the future
            ttl.off()           # Do future stuff here
            at_mu(now_t)        # Jump back to present

            # Eventually append other events...
```

### Expected Behavior
`core.reset()` should wipe all FIFO events

### Actual (undesired) Behavior
Future events on FIFO are not cleared by `core.reset()`

### Your System (omit irrelevant parts)
* Arch Linux
* ARTIQ version: ARTIQ v8.8322.b225717.beta
* Version of the gateware and runtime loaded in the core device:  8.8319.e9a153b.beta
* Hardware involved: Two artiq Kasli-crates with DRTIO (master/satellite).

Spaqin, 2023-06-14:
> Future events on FIFO are not cleared by core.reset()

That possibly has been fixed with a533f2a0cd6c116c0a2311f489365b82dc22baf3 - please check newer firmware version before sending a bug report for beta...

systemofapwne, 2023-06-14:
> > Future events on FIFO are not cleared by core.reset()
> 
> That possibly has been fixed with [a533f2a](https://github.com/m-labs/artiq/commit/a533f2a0cd6c116c0a2311f489365b82dc22baf3) - please check newer firmware version before sending a bug report for beta...

Thank you. Going to test recent firmware later that day and report back. You are absolutely right, I should have tested it with the recent beta build. Stay tuned!

systemofapwne, 2023-06-14:
After flashing a recent gateware, compiled against todays commit [115415d](https://github.com/m-labs/artiq/commit/115415d1204a2c69f41aa4c35e187f8a30855b1d), the issue seems to have vanished.

Thank you the feedback. I close this report now.

closed 2023-06-14 by systemofapwne