m-labs/artiq #2123: Panic at runtime/rtio_mgt.rs after repeated DMA usage

lriesebos, 2023-06-29:
# Bug Report

## One-Line Summary

Kasli panics sometimes when doing repeated DMA calls to a channel on a satellite.

## Issue Details

### Steps to Reproduce

My system has a single satellite. This satellite has a `TTLOut` that is used to run this function: https://gitlab.com/duke-artiq/dax/-/blob/master/dax/modules/rtio_benchmark.py#L479. This is a benchmark we use to measure performance of the Kasli. Sometimes the Kasli panics when we run that code.

The panic was not observed when running the same test on a `TTLOut` on the master, though I cannot test that exhaustively. 

The panic is visible in the UART logs of the master. The satellite does not report any UART messages when the master panics, except that the link is lost.

### Expected Behavior

No panic.

### Actual (undesired) Behavior

Panic.

```
[2023-06-29 16:40:31] [   771.652731s]  INFO(runtime::session): no connection, starting idle kernel
[2023-06-29 16:40:31] [   771.736134s]  INFO(runtime::kern_hwreq): resetting RTIO
[2023-06-29 16:40:31] [   771.790687s]  INFO(runtime::session): new connection from 192.168.1.100:33904
[2023-06-29 16:40:31] panic at runtime/rtio_mgt.rs:51:29: called `Result::unwrap()` on an `Err` value: Interrupted
[2023-06-29 16:40:31] backtrace for software version 7.8173.ff97675;[removed gateware id]:
[2023-06-29 16:40:31] 0x4003d29c
[2023-06-29 16:40:31] 0x4000af10
[2023-06-29 16:40:31] 0x4000a510
[2023-06-29 16:40:31] 0x40028088
[2023-06-29 16:40:31] 0x40027eb0
[2023-06-29 16:40:31] 0x400211d8
[2023-06-29 16:40:31] 0x40024e80
[2023-06-29 16:40:31] 0x4002cc4c
[2023-06-29 16:40:31] 0x4000e7c4
[2023-06-29 16:40:31] 0x4000e7b4
[2023-06-29 16:40:31] 0x40038c94
[2023-06-29 16:40:31] 0x4000770c
[2023-06-29 16:40:31] 0x40028f40
[2023-06-29 16:40:31] 0x40028ee8
[2023-06-29 16:40:31] 0x4003c414
[2023-06-29 16:40:31] halting.
[2023-06-29 16:40:31] use `artiq_coremgmt config write -s panic_reset 1` to restart instead
```

### Your System (omit irrelevant parts)

* Operating System: Linux (Ubuntu 22.04)
* ARTIQ version: 7.8173.ff97675
* Version of the gateware and runtime loaded in the core device: same as ARTIQ version
* Hardware involved: 2x Kasli 2.0 with master-satellite configuration, a DIO card on the satellite

dnadlinger, 2023-06-30:
What do the backtrace addresses resolve to? (You can use `addr2line` with the .elf file corresponding to the gateware build.)

lriesebos, 2023-06-30:
hope this helps!

```
0x4003d29c /nix/store/ky5j6106mjc6qgf1l4607z3bpg3yclgl-python3.9-artiq-7.8173.ff97675/lib/python3.9/site-packages/artiq/firmware/libunwind_backtrace/lib.rs:42
0x4000af10 runtime.7bvx51tr-cgu.0:?
0x4000a510 runtime.7bvx51tr-cgu.0:?
0x40028088 /nix/store/kw58p9214r5cacry19ivdi8n28lqbyji-rust-1.51.0-nightly-2021-01-28-c0b64d97b/lib/rustlib/src/rust/library/core/src/result.rs:1037
0x40027eb0 /nix/store/ky5j6106mjc6qgf1l4607z3bpg3yclgl-python3.9-artiq-7.8173.ff97675/lib/python3.9/site-packages/artiq/firmware/runtime/rtio_mgt.rs:59
0x400211d8 /nix/store/ky5j6106mjc6qgf1l4607z3bpg3yclgl-python3.9-artiq-7.8173.ff97675/lib/python3.9/site-packages/artiq/firmware/runtime/rtio_mgt.rs:316
0x40024e80 /nix/store/ky5j6106mjc6qgf1l4607z3bpg3yclgl-python3.9-artiq-7.8173.ff97675/lib/python3.9/site-packages/artiq/firmware/runtime/session.rs:548
0x4002cc4c /nix/store/ky5j6106mjc6qgf1l4607z3bpg3yclgl-python3.9-artiq-7.8173.ff97675/lib/python3.9/site-packages/artiq/firmware/runtime/session.rs:666
0x4000e7c4 /build/cargo-vendor-dir/fringe-1.2.1/src/arch/riscv32.rs:91
0x4000e7b4 /build/cargo-vendor-dir/fringe-1.2.1/src/arch/riscv32.rs:56
0x40038c94 /build/cargo-vendor-dir/fringe-1.2.1/src/arch/riscv32.rs:210
0x4000770c /nix/store/ky5j6106mjc6qgf1l4607z3bpg3yclgl-python3.9-artiq-7.8173.ff97675/lib/python3.9/site-packages/artiq/firmware/libboard_misoc/riscv32/boot.rs:23
0x40028f40 /nix/store/ky5j6106mjc6qgf1l4607z3bpg3yclgl-python3.9-artiq-7.8173.ff97675/lib/python3.9/site-packages/artiq/firmware/runtime/main.rs:257
0x40028ee8 /nix/store/ky5j6106mjc6qgf1l4607z3bpg3yclgl-python3.9-artiq-7.8173.ff97675/lib/python3.9/site-packages/artiq/firmware/liblogger_artiq/lib.rs:67
0x4003c414 /nix/store/ky5j6106mjc6qgf1l4607z3bpg3yclgl-python3.9-artiq-7.8173.ff97675/lib/python3.9/site-packages/artiq/firmware/runtime/main.rs:256
```

lriesebos, 2024-09-11:
This is an old one, but wanted to add more info for people running into this. this error seems to be caused by two `core.reset()` calls after each other without any waits or delays between them. Observed mostly for systems with DRTIO. I suspect that the second `reset()` call interferes with the first one. this happens mostly during handoff of experiments where the previous experiments called reset at the end, and the new experiment immediately starts with a reset. calling `wait_until_mu()` after the first reset seems to solve it for most cases, but it might just be a timing issue in how a second reset can interfere with the first reset.

yuyichao, 2024-10-23:
I'm seeing a similar crash. However, running 1000 core.reset in a tight loop does not seem to trigger this crash. (Not reliably so anyway...)

lriesebos, 2024-10-23:
> I'm seeing a similar crash. However, running 1000 core.reset in a tight loop does not seem to trigger this crash. (Not reliably so anyway...)

I assume you also use DRTIO? If so, is it easier to reproduce this error by waiting for a bit between `core.reset()` calls? since this seems to be a timing bug, it might be that the timing of the two calls must be in a certain range. for our crate, the latency between the crates is about 279 ns, so that is an order of magnitude that could maybe give a clue.

yuyichao, 2024-10-23:
> I assume you also use DRTIO

I assume this refers to having a master and a satellite box and if so the hardware has it. (I assume you can see it from https://gitlab.com/duke-artiq/dax-gateware-json/-/blob/master/gold_master.json?ref_type=heads)

I believe the experiment where I run the reset in a loop doesn't use it. In fact, very few of our experiments uses the satellite box. The one that does indeed does seem to have a somewhat higher probability of crashing on artiq 6.

lriesebos, 2024-10-23:
> I assume this refers to having a master and a satellite box and if so the hardware has it. (I assume you can see it from [...]

ah ok yes I was referring to that. so then we seem to agree that DRTIO (i.e. having a master and at least one satellite crate) has a higher probability of crashing.

regardless, do I understand you can also reproduce this on a standalone crate with the same error trance? I have not seen that before and that might change the diagnosis of this problem a bit.

so we mitigate the issue by just making sure that every `core.reset()` call is followed by a `wait_until_mu(now_mu())` call, even in handoff situations. that is normally the case anyway, but there was a bug somewhere in the code where that did not happen. additionally, our original idle kernel called `core.reset()`, but we changed that to `core.break_realtime()` to not have potential conflicts with that call. since then I don't think I have run into this issue again, but I have not confirmed that in the logs.

also, just updated the snippets accordingly. https://gitlab.com/duke-artiq/dax/-/snippets/1966949
with a proper startup kernel, that reset was never necessary to start with, but more as a safeguard. but if it can cause this issue, using `core.break_realtime()` works fine for us and probably most users of idle kernels.

yuyichao, 2024-10-23:
> ah ok yes I was referring to that. so then we seem to agree that DRTIO (i.e. having a master and at least one satellite crate) has a higher probability of crashing.

Given that we see this crash (at least the backtrace seems identical) and that we have DRTIO I would agree. I was wondering why few other experiments here see the crashes we've run into and I guess that might've been part of the reason....

> regardless, do I understand you can also reproduce this on a standalone crate with the same error trance? I have not seen that before and that might change the diagnosis of this problem a bit.

What do you mean by standalone crate? The hardware I've experienced any panic on have always been with DRTIO. The crashes I've got on artiq-6 have happened at random places previously but since we've upgraded to artiq-7, the two crashes have idential backtrace and both of them happened on experiment start/stop.

> so we mitigate the issue by just making sure that every core.reset() call is followed by a wait_until_mu(now_mu()) call, even in handoff situations. that is normally the case anyway, but there was a [...]

The second one of this crash happened after I changed every one of the reset I can find (including the two in pmt_monitor from dax.client) with the following,

```python
    core.wait_until_mu(core.get_rtio_counter_mu() + 100_000)
    core.reset()
    core.wait_until_mu(now_mu() + 50_000)
    # Some users expect enough slack after the reset returns
    at_mu(core.get_rtio_counter_mu() + 125_000)
```

So I guess if it was caused by the timing it's not as simple as happening too quickly.....

I'll update our idle kernel.... It's difficult to intentially trigger this issue though so we can only wait to see if this fixes things....

yuyichao, 2024-10-23:
> What do you mean by standalone crate? The hardware I've experienced any panic on have always been with DRTIO. The crashes I've got on artiq-6 have happened at random places previously but since [...]

Actually, I take that back. The backtrace for the second crash is different and I've also just got a third one which happened right after my 10000 reset experiment finished.

```
[2024-10-23 17:01:19] [  3955.343057s]  INFO(runtime::kern_hwreq): resetting RTIO
[2024-10-23 17:01:19] [  3955.391293s]  INFO(runtime::kern_hwreq): resetting RTIO
[2024-10-23 17:01:19] [  3955.439691s]  INFO(runtime::kern_hwreq): resetting RTIO
[2024-10-23 17:01:19] [  3955.633285s]  INFO(runtime::session): no connection, starting idle kernel
[2024-10-23 17:01:19] [  3955.716930s]  INFO(runtime::kern_hwreq): resetting RTIO
[2024-10-23 17:01:19] [  3955.771289s]  INFO(runtime::session): new connection from 10.236.89.1:38166
[2024-10-23 17:01:19] panic at runtime/rtio_mgt.rs:51:29: called `Result::unwrap()` on an `Err` value: Interrupted
[2024-10-23 17:01:19] backtrace for software version 7.8208.38c72fd;gold_master:
[2024-10-23 17:01:19] 0x4003cae0
[2024-10-23 17:01:19] 0x40008b90
[2024-10-23 17:01:19] 0x400084b4
[2024-10-23 17:01:19] 0x40026940
[2024-10-23 17:01:19] 0x40026768
[2024-10-23 17:01:19] 0x40021460
[2024-10-23 17:01:19] 0x400236f4
[2024-10-23 17:01:19] 0x4002d9b4
[2024-10-23 17:01:19] 0x4000ba10
[2024-10-23 17:01:19] 0x4000ba00
[2024-10-23 17:01:19] 0x400384e0
[2024-10-23 17:01:19] 0x40005ac8
[2024-10-23 17:01:19] 0x40002454
[2024-10-23 17:01:19] 0x400005c4
[2024-10-23 17:01:19] 0x4003bc60
```

Backtrace is
```
0x4003cae0: unwind_backtrace::backtrace at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/libunwind_backtrace/lib.rs:42
 (inlined by) rust_begin_unwind at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/main.rs:360
0x40008b90: core::panicking::panic_fmt at /nix/store/lp3wfd7jqiyljq5q43zkglcval0dyhv5-rust-1.51.0-nightly-2021-01-28-c0b64d97b/lib/rustlib/src/rust/library/core/src/panicking.rs:92
0x400084b4: core::result::unwrap_failed at /nix/store/lp3wfd7jqiyljq5q43zkglcval0dyhv5-rust-1.51.0-nightly-2021-01-28-c0b64d97b/lib/rustlib/src/rust/library/core/src/result.rs:1284
0x40026940: core::result::Result<T,E>::unwrap at /nix/store/lp3wfd7jqiyljq5q43zkglcval0dyhv5-rust-1.51.0-nightly-2021-01-28-c0b64d97b/lib/rustlib/src/rust/library/core/src/result.rs:1037
 (inlined by) runtime::rtio_mgt::drtio::recv_aux_timeout at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/rtio_mgt.rs:51
0x40026768: runtime::rtio_mgt::drtio::aux_transact at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/rtio_mgt.rs:59
0x40021460: runtime::rtio_mgt::drtio::reset at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/rtio_mgt.rs:316
 (inlined by) runtime::rtio_mgt::reset at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/rtio_mgt.rs:383
 (inlined by) runtime::kern_hwreq::process_kern_hwreq at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/kern_hwreq.rs:245
 (inlined by) runtime::session::process_kern_message::{{closure}} at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/session.rs:350
 (inlined by) runtime::session::kern_recv_notrace at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/session.rs:167
 (inlined by) runtime::session::process_kern_message at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/session.rs:334
0x400236f4: runtime::session::flash_kernel_worker at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/session.rs:548
0x4002d9b4: runtime::session::thread::{{closure}} at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/session.rs:666
 (inlined by) runtime::sched::Thread::new::{{closure}} at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/sched.rs:68
 (inlined by) fringe::generator::Generator<Input,Output,Stack>::unsafe_new::generator_wrapper at /tmp/nix-build-artiq-board-kasli-gold_master.drv-0/cargo-vendor-dir/fringe-1.2.1/src/generator.rs:123
0x4000ba10: fringe::arch::imp::init::trampoline_2 at /tmp/nix-build-artiq-board-kasli-gold_master.drv-0/cargo-vendor-dir/fringe-1.2.1/src/arch/riscv32.rs:91
0x4000ba00: fringe::arch::imp::init::trampoline_1 at /tmp/nix-build-artiq-board-kasli-gold_master.drv-0/cargo-vendor-dir/fringe-1.2.1/src/arch/riscv32.rs:56
0x400384e0: fringe::arch::imp::swap at /tmp/nix-build-artiq-board-kasli-gold_master.drv-0/cargo-vendor-dir/fringe-1.2.1/src/arch/riscv32.rs:210
[... cut ...]
0x400005c4: logger_artiq::BufferLogger::register at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/liblogger_artiq/lib.rs:67
0x4003bc60: ?? at /nix/store/hnfph2qjhw430yyb47d44x8dwy30h1n6-python3.9-artiq-7.8208.38c72fd/lib/python3.9/site-packages/artiq/firmware/runtime/main.rs:256
```

which is now very similar to the crash I've also got on artiq-6 https://github.com/m-labs/artiq/issues/2530#issuecomment-2433409033 (it went past the lock and crashed when reciving https://github.com/m-labs/artiq/blob/38c72fdab41679c300c235d960e87b9e06bea5b4/artiq/firmware/runtime/rtio_mgt.rs#L51)

lriesebos, 2024-10-23:
> What do you mean by standalone crate? The hardware I've experienced any panic on have always been with DRTIO. The crashes I've got on artiq-6 have happened at random places previously but since [...]

ah ok I read that wrong. so I think we agree that DRTIO is a prerequisite then for this issue.

> The second one of this crash happened after I changed every one of the reset I can find (including the two in pmt_monitor from dax.client)...

yeah, pmt monitor was also the first time we noticed this. I pushed https://gitlab.com/duke-artiq/dax/-/merge_requests/223 recently that fixed that. if you see any other occurrences in DAX where this handoff issue could be triggered, let me know.

regardless, it is a difficult to debug issue. all possible handoffs need to be taken into account. as mentioned earlier, if all kernels end with `wait_until_mu(now_mu())`, that seems to mitigate the issue for us. that should includes handoffs within experiments, between experiments, and any handoffs with the idle kernel. for us, no further timing juggling was needed.

> Actually, I take that back. The backtrace for the second crash is different and I've also just got a third one which happened right after my 10000 reset experiment finished.

lets not get more different errors involved in this ticket and focus on the handoff one for DRTIO that involves the reset sequence.

yuyichao, 2024-10-23:
> lets not get more different errors involved in this ticket and focus on the handoff one for DRTIO that involves the reset sequence.

Looking at the code, it seems that both were triggered by [`Io::suspend` returning a `Error::Interrupted`](https://github.com/m-labs/artiq/blob/4bf2331d6a03c5e84702fe22e666c41a043f603e/artiq/firmware/runtime/sched.rs#L238-L259). So I think the two are probably the same issue, just that whatever causing that "Interrupted" scheduling state happened at different part of this call chain.

yuyichao, 2024-10-23:
> yeah, pmt monitor was also the first time we noticed this. I pushed https://gitlab.com/duke-artiq/dax/-/merge_requests/223 recently that fixed that. if you see any other occurrences in DAX where this handoff issue could be triggered, let me know.

Not knowing what exactly would trigger this, I added, in our pmt monitor class inherited from this one, wait for the reset in the device_setup as well and as well as a wait in the device_cleanup function to achieve what you did in that MR.

yuyichao, 2024-10-23:
I've updated the idle kernel and I'll see if we still see any issues during daily operation.

However, looking at the code, it seems that if `respawn` was called (I assume to abort a kernel run) when the kernel was in the middle of some operator (reset included), a lot of the operations in rtio_mgr would simply panic. Inspired by this, I just found that the issue (or at least a version of it) is extremely easy to trigger by simply deleting an experiment during a reset, while some other non-trivial experiment (in this case pmt_monitor) was running in the background. So to be more precise, (and this is with an updated idle kernel that does not have reset)

1. Run pmt monitor, I assume a kernel doing some dummy reset/rpc would work too but I haven't tested it yet.
2. Start a experiment with 1000 resets in a loop
3. After the reset storm is started, use the "Delete" button on the UI to abort the reset experiment (so that artiq would switch back to the long-running experiment)

2 out of 2 times I've seen the panic at the `lock().unwrap()` call identical to the backtrace above. The call chain higher up is slightly different but I assume that's expected given the manual abort involved.

I know that the Delete feature is probably not recommended, but I also do feel like it shouldn't reliably cause a firmware panic....

yuyichao, 2024-10-23:
Here are two vanilla artiq experiments that I was able to use to reliably trigger the issue. I have not tested if all the components are required to trigger it.

To reproduce,
1. Run RPCExp with priority -1
2. After it starts printing `async_rpc`, run ResetExp with priority 0
3. After the second experiment starts running, "Delete" the second experiment

So far I was able to reproduce a panic 3 out of 3 times. Running ResetExp without running the RPCExp does not reproduce the issue (did not try with a idle kernel with reset in it).

```python
class RPCExp(EnvExperiment):
    kernel_invariants = {'core', 'scheduler'}

    def build(self):
        self.setattr_device('core')
        self.setattr_device('scheduler')

    @rpc(flags={'async'})
    def async_rpc(self, a):
        print('async_rpc')

    @kernel
    def run_kernel(self):
        self.core.reset()
        while not self.scheduler.check_pause():
            delay_mu(1000)
            self.core.wait_until_mu(now_mu())
            self.async_rpc([1, 2, 3, 4, 5])
        # self.core.reset()

    def run(self):
        # Initialize system
        while True:
            print('Run kernel')
            self.run_kernel()
            print('Exit from kernel')
            self.core.comm.close()
            print('Close comm')
            try:
                self.scheduler.pause()
            except TerminationRequested:
                return

class ResetExp(EnvExperiment):
    kernel_invariants = {'core'}

    def build(self):
        self.setattr_device('core')

    @kernel
    def run(self):
        for i in range(1000):
            self.core.reset()
```

yuyichao, 2024-10-23:
And the relation with DRTIO would simply be that with DRTIO the reset (RTIOInit) request contains asynchronous code to comunicate with the satellite box. This might be racing with the task switch causing the interrupted error and the failure to handle such error triggers the panic.

yuyichao, 2024-10-24:
And it seems that this should be fixed by https://github.com/m-labs/artiq/commit/6640bf0e829f72d719a4740644e9c389052d7b46 (At least rtio_mgt has proper error handling now, didn't check higher up in the call chain)

yuyichao, 2024-10-24:
Is there any plan to backport https://github.com/m-labs/artiq/commit/6640bf0e829f72d719a4740644e9c389052d7b46 (only the parts that applies) to artiq 7?

Spaqin, 2024-10-25:
> Is there any plan to backport [6640bf0](https://github.com/m-labs/artiq/commit/6640bf0e829f72d719a4740644e9c389052d7b46) (only the parts that applies) to artiq 7?

The panics was definitely a bigger problem with subkernels, when operations in firmware would take longer and the chance of getting interrupted was much higher. On top of that the errors within the firmware needed a cleanup. As such, I haven't even considered backporting to ARTIQ-7. Generally, after a release, to keep it stable, we don't refactor the code, but we still include bug fixes.

However, I would strongly encourage you to contribute :) At very least, catching and handling SchedError::Interrupted in a better way than unwrapping should be implemented.

dnadlinger, 2025-10-02:
A similar issue was fixed in 42d3c3b4b2943f24cc9792350a391b273cd19686 by just ignoring the `Interrupted` error case. Generally, with "non-graceful" termination/experiment replacement, any scheduler calls must consider the possibility of the task getting cancelled. The issue still exists in `Core.reset()` on latest master:

https://github.com/m-labs/artiq/blob/18d1cb939c8dc621172db872c02818efacd10c2a/artiq/firmware/runtime/rtio_mgt.rs#L491