m-labs/artiq #2144: Cannot move Applets GUI Window in WSL2

andrewvh4, 2023-07-21:
# Bug Report

## One-Line Summary

Cannot move Applets GUI Window in WSL2

## Issue Details
WSL2 now natively supports GUI Apps: https://learn.microsoft.com/en-us/windows/wsl/tutorials/gui-apps
However, applets cannot be moved or re-sized. They remain in the top-left corner. 

[image]

### Steps to Reproduce

1. Create an Ubuntu distribution in WSL2: https://learn.microsoft.com/en-us/windows/wsl/install
2. Install Artiq using Nix with the same procedure as an Ubuntu machine
3. Launch the artiq_session and run an experiment that generates an applet.

### Expected Behavior

We should be able to move and re-size the applet as is possible in native Linux systems.

### Actual (undesired) Behavior

The applet is stuck in the top left corner of the screen and cannot be moved or re-sized:
The following message appears when launching an applet:
`print:qt.qpa.xcb: internal error:  void QXcbWindow::setNetWmStateOnUnmappedWindow() called on mapped window`

[image]

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: artiq_client --version
* Hardware involved: Kasli 2.0

charlesbaynham, 2023-07-21:
+1 - this also happens on Windows 11 which introduced extra support for WSL GUIs.

charlesbaynham, 2023-07-21:
FWIW I think the `setNetWmStateOnUnmappedWindow` error is a red herring. We also get that from our Ubuntu systems which don't have this problem.

thomasfire, 2023-07-26:
Does this persist on ARTIQ-7 and ARTIQ-8 (beta)?
The ARTIQ-6 is unlikely to be fixed anyway

andrewvh4, 2023-07-26:
@thomasfire I just tested this with Artiq-7 and saw the same behavior. I don't have a system set up with Artiq-8

andrewvh4, 2023-08-18:
Are there any updates on this issue? Figuring this out would be very helpful for allowing remote operation across different platforms.

andrewvh4, 2023-11-09:
At the time of writing, I was having issues with conda-forge failing due to an invalid OpenSSL certificate. I re-tested this week, and that seems to have resolved and I am able to use the Conda installation method. 

In the future, I do think WSL support would be valuable. But I know that takes more resources to achieve.

charlesbaynham, 2024-02-02:
I'd be happy to work on this at some point in the future, though I can't guarantee when I'll find time.

If WSL can be fully supported then I think your support workload might reduce since Nix could then be the only installation method required. This issue is the only blocker for that AFAIK. I don't know how well WSL interfaces with USB devices however, we only use WSL for `artiq_client` connections.

andrewvh4, 2024-02-02:
You are correct that WSL has issues with USB devices. However, that's not a huge issue since it's pretty straightforward to [set up a Raspberry Pi for remote flashing/logging](https://gitlab.com/duke-artiq/dax-comtools/-/wikis/Raspberry-Pi-set-up).

charlesbaynham, 2024-02-03:
> It's also pretty straightforward to install Linux or BSD on your PC.

It's not quite so easy to convince all your collaborators and their diverse IT departments to do so!

evilmav, 2024-06-27:
Possibly related: https://github.com/microsoft/wslg/issues/1153

thatschatt, 2024-10-09:
Would folk be interested in a flag for the dashboard that disables floating dockable windows? From a very quick test this works around the WSL issue, and could also be of general interest for people who don't like things floating.

andrewvh4, 2024-10-09:
That sounds like a good solution to me that wouldn't interfere with the existing implementation. 
Would you still be able to move around the windows within the dashboard? They would just always be docked to the main GUI?

thatschatt, 2024-10-24:
Yep - they just snap back if they don't have a home. This is just a hack of turning off every `DockWidgetFloatable` flag I could find. When I have time I'll make a PR to put this as a flag.

This is Ubuntu in WSL2, under Windows 11.

https://github.com/user-attachments/assets/06228bac-413e-418e-95ba-e5d7693d22f9

lriesebos, 2024-10-24:
> Yep - they just snap back if they don't have a home. This is just a hack of turning off every `DockWidgetFloatable` flag I could find. When I have time I'll make a PR to put this as a flag.
> 
> This is Ubuntu in WSL2, under Windows 11.
> dashboard_wsl.mp4

that's awesome!
just wondering, new applets that appear for the first time are floating. where do these appear now?

charlesbaynham, 2024-10-24:
That is very awesome 😄 we'll definitely use this, thank you!