m-labs/artiq #2160: Placer fails when implementing artiq on Kasli v2 with Fastino log2_width=5

marmeladapk, 2023-08-21:
# Bug Report

## One-Line Summary

Placer fails when implementing artiq on Kasli v2 with Fastino log2_width=5 on artiq-7 and on master.

## Issue Details

Artiq release-7, ff97675; also happens on master branch 68dd0e0. 

Gateware definition:
```
{
    "target": "kasli",
    "variant": "fastino_wide",
    "hw_rev": "v2.0",
    "base": "standalone",
    "rtio_frequency": 125e6,
    "peripherals": [
    {
        "type": "fastino",
        "hw_rev": "v1.2.1",
        "log2_width": 5,
        "ports": [0]
    }
    ]
}
```

### Actual (undesired) Behavior

Release-7:

ERROR: [DRC UTLZ-1] Resource utilization: LUT as Distributed RAM over-utilized in Top Level Design (This design requires more LUT as Distributed RAM cells than are available in the target device. This design requires 30372 of such cell types but only 19000 compatible sites are available in the target device. (...)

ERROR: [DRC UTLZ-1] Resource utilization: LUT as Memory over-utilized in Top Level Design (This design requires more LUT as Memory cells than are available in the target device. This design requires 31431 of such cell types but only 19000 compatible sites are available in the target device. (...)

ERROR: [DRC UTLZ-1] Resource utilization: RAMD64E over-utilized in Top Level Design (This design requires more RAMD64E cells than are available in the target device. This design requires 30168 of such cell types but only 19000 compatible sites are available in the target device. (...)

ERROR: [DRC UTLZ-1] Resource utilization: Slice LUTs over-utilized in Top Level Design (This design requires more Slice LUTs cells than are available in the target device. This design requires 84115 of such cell types but only 63400 compatible sites are available in the target device. (...)

Master:

ERROR: [DRC UTLZ-1] Resource utilization: LUT as Distributed RAM over-utilized in Top Level Design (This design requires more LUT as Distributed RAM cells than are available in the target device. This design requires 27908 of such cell types but only 19000 compatible sites are available in the target device. (...)

ERROR: [DRC UTLZ-1] Resource utilization: LUT as Memory over-utilized in Top Level Design (This design requires more LUT as Memory cells than are available in the target device. This design requires 28967 of such cell types but only 19000 compatible sites are available in the target device. (...)

ERROR: [DRC UTLZ-1] Resource utilization: RAMD64E over-utilized in Top Level Design (This design requires more RAMD64E cells than are available in the target device. This design requires 27704 of such cell types but only 19000 compatible sites are available in the target device. (...)

ERROR: [DRC UTLZ-1] Resource utilization: Slice LUTs over-utilized in Top Level Design (This design requires more Slice LUTs cells than are available in the target device. This design requires 80192 of such cell types but only 63400 compatible sites are available in the target device. (...)

### Your System (omit irrelevant parts)

* ARTIQ version: Artiq release-7, ff97675 and master branch: 68dd0e0
* nix environment

marmeladapk, 2023-08-28:
In release-6 I can fit 10 Fastinos with log2_width=5 on Kasli v2. In release-7 I can fit 12 Fastinos on Kasli SoC.

occheung, 2023-08-30:
There are 3 main gateware changes from release-6 to release-7 that I am aware of:

1. mor1kx -> vexriscv. I can fit 8 Fastinos on Kasli v2 using somewhere around bb61f2d (after CPU change is merged in)
2. adding FPU in kernel CPU
3. 32-bits datapath width to 64-bits datapath width

There aren't a good ARTIQ commit to separate the impact of 2 and 3, so I restore the 32-bits datapath to Kasli v2 on release-7:
- Generate a new vexriscv core that keeps the FPU, with a 32-bits I-cache port instead of 64-bits.
- Replace the 64-bits datapath in misoc with 32-bits, and down convert the 64-bits D-cache bus to 32-bits. The CSR ignores the `sel` signals of the wishbone bus, so there are some hacks to the conversion. Memtest will fail otherwise.
- Fix the `csr_offset` in the compiler such that timestamps are handled correctly.

And I can fit 4 Fastinos without timing violations, 5 otherwise.

Kasli v1.1 does not have the FPU nor the 64-bits datapath. It can fit 8 Fastinos, and may have to lose 1 to 2 Fastinos to pass timing.

The BRAM is exhaused in all these builds, so maybe downsizing the caches can help by releasing a bit more LUTs from the DRAMs.