m-labs/artiq #2174: dashboard.datasets.CreateEditDialog misrepresents dataset types

jfniedermeyer, 2023-08-31:
# Bug Report

## One-Line Summary

Applet dashboard.datasets.CreateEditDialog misrepresents dataset types and is inconsistent with dashboard.datasets.DatasetsDock

## Issue Details

Applet dashboard.datasets.CreateEditDialog misrepresents dataset types and is inconsistent with dashboard.datasets.DatasetsDock. Int is recast as float and this is reproducible. Float is occasionally recast as float64 wihtout being decoded, which makes them not human readable. I have not been able to find a way to always reproduce a float being recast as float64. I suspect this has to do with implementation of metadata in the dataset, but I have not been able to find a consistent workaround by catching different types. 

It is important to note that the datatype in the underlying dataset_db.mdb is correct; if you create a new dataset as an int, it will be written as an int there. 

### Steps to Reproduce

1. Run artiq_master and artiq_dashboard (either in a populated repo or a clean one)
2. Right click the Dataset applet and select "New dataset..." or hit CTRL+N
3. Enter any name and set the value to an integer
4. Hit Ok
5. Double click on the new dataset and observe its representation

### Expected Behavior

Int should be represented by an int in both applets (and the underlying dataset)

### Actual (undesired) Behavior

Int is cast to float in dashboard.datasets.CreateEditDialog, but appears as an int in dashboard.datasets.DatasetsDock

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: ARTIQ v8.8482+9556ca5.beta
* Conda list: [spec-file.txt](https://github.com/m-labs/artiq/files/12489623/spec-file.txt)
* Hardware involved: Host PC

SimonRenblad, 2023-09-01:
Pushed a PR to resolve the reproducible bug. Please note that if a scaling factor other than 1 is used (either through units or explicitly) the dashboard will cast to float in our current implementation.

In terms of the float to float64 cast, it should only occur if the scale param is set to a float64 type AFAIK. Please let us know if you are able to reproduce this bug so we may address it. Will do more investigation into it myself, as well.

Thank you!

jfniedermeyer, 2023-09-01:
Thanks all.

Re: float64. I'm out of the lab until Wednesday, but I'll try and get my colleague to post a screenshot. It seems to mainly be an issue for datasets that were converted from .pyon to the original .mdb format (sans metadata), and then to the new format. New datasets in an empty repo haven't shown this behavior yet. 

Other than that, I haven't found a pattern, but I'll let you know. For example, 315e6 Hz was float, but 426e6 Hz and 90e6 Hz were float64. The CreateEditDialog would show the raw pyon decoded string of "npscalar(<many characters>)", but the DatasetsDock would decode it fine.  It's also worth noting that we have empty metadata for now until we update the rest of our codebase, so it's using the default_scale for everything.

nlysne, 2023-09-03:
Alright, finally found an entry that was consistently getting displayed as a float64 bytestring despite being (nominally) an int in the dataset.
[image]
Reentering the key's value as '315000000' and saving it causes the key to follow the main unexpected behavior above.
[image]