m-labs/artiq #2226: Noise coupling between the card inserts!!!

MohammadAbuzahra, 2023-09-28:
# Support Question

## One-Line Summery
some noise coupling occurs between card inserts

## Description
I am trying to use the Fastino card as an analog input for my circuit.  In addition, I am using the "DIO-SMA" card to generate the digital signals needed to control my circuit. However, unfortunately, the Fastino output is very noisy and does not match the documentation. I started debugging my setup to find the noise sources.

This support question is focusing on a single type of noise that I am finding difficulty to solve.

Whenever I connect any connector to the digital card pins I see spikes on my Fastino output signals. The state of the digital pin does not seem to mater on the coupled noise. First picture shows the output digital signal (yellow) while in high state. Second picture shows the output digital signal (yellow) after power-up (should be in high z state). But in both cases, the Fastino output (blue) shows 100mV periodic noise spikes. For both cases, the oscilloscope channels are AC coupled.

[image]
[image]

Pictures of the setup are below. The first picture shows that the digital card output is connected only to an SMA-BNC adapter. The second picture shows that a BNC cable is connected to the adapter. Nonetheless, same noise spikes are observed when an SMA cable is used.

[image]
[image]

However, when I connect nothing to the digital output pin, these particular noise spikes seems to disappear. The picture below shows the Fastino output (blue).

[image]

I know there are still noisy signals couples to the Fastino output seen from the picture above, however solving the problem one step at a time. Figuring out the source of these spikes would be the first step.

## Additional Information
Hardware:
* Kasli-2
* DIO-SMA (v1.2, port 0)
* 2x DIO-MCX (v1.2, ports 1,2,3,4)
* Fastino (v1.2, port 5)
* DIO-RJ45 (ports 6 and 7)
* Sampler (v2.3, ports 8 and 9)

Software:
* artiq 7.8161.8a2ea57

jordens, 2023-09-29:
@gkasprow 

A couple things: The LEDs indicate that your digital output is not high-z.
Do provide adequate airflow, leave sufficient space above and below the crate and/or installed forced airflow cooling.

gkasprow, 2023-09-29:
Isn't it DIO dc/dc converter common mode noise entering via ground loop?

gkasprow, 2023-09-29:
Could be also external noise source being pickup by the ground loop.

MohammadAbuzahra, 2023-10-05:
> Could be also external noise source being pickup by the ground loop.

I also see the noise when only the SMA/BNC adapter is connected and kept floating without connecting it to my measuring setup. In that case there should be no ground loop since the isolated ground of the DIO and the lab ground are not connected to each other, but I still see the same noise but with smaller magnitude.  Or would I be mistaken to assume so.

MohammadAbuzahra, 2023-10-05:
> @gkasprow
> 
> A couple things: The LEDs indicate that your digital output is not high-z. Do provide adequate airflow, leave sufficient space above and below the crate and/or installed forced airflow cooling.

I will give it a try, hopefully it helps

gkasprow, 2023-10-05:
There is always a ground loop via PE connection of your scope and PE connection of the chassis power supply. What power supply are you using?

MohammadAbuzahra, 2023-10-06:
> There is always a ground loop via PE connection of your scope and PE connection of the chassis power supply. What power supply are you using?

I am using R&S HMP4040 power supplies and Tektronix MSO56B scope. The sinara box is powered via the wall adapter.

gkasprow, 2023-10-09:
Does this wall adapter have the correct PE connection?
Can you connect the SInara box to the bench DC supply instead?

MohammadAbuzahra, 2023-10-10:
That is what I am trying to do next, power it via a bench DC supply. I am adjusting my setup