m-labs/artiq #2244: kasli-soc led0 nonfunctional

jbqubit, 2023-10-10:
# Bug Report

## One-Line Summary

led0 does not flash .

## Issue Details

### Steps to Reproduce

The [build script](https://git.m-labs.hk/M-Labs/artiq-zynq/src/branch/release-7/src/gateware/kasli_soc.py) automatically adds two LEDs: led0 and led1. 
```
$ gateware/kasli_soc.py -g ../build/gateware variant.json
DIO (EEM0) starting at RTIO channel 0x000000
USER LED at RTIO channel 0x000008
USER LED at RTIO channel 0x000009
...
```
Add them to the device_db.py.
```$ cat device_db.py
...
device_db["led0"] = {
    "type": "local",
    "module": "artiq.coredevice.ttl",
    "class": "TTLOut",
    "arguments": {"channel": 0x000008},
}
device_db["led1"] = {
    "type": "local",
    "module": "artiq.coredevice.ttl",
    "class": "TTLOut",
    "arguments": {"channel": 0x00009},
}
...
```

Observe that  ```led1``` causes the LED marked "L1" on the kasli-soc faceplate to flash. 
Observe that  ```led0``` causes no LED to flash. 

------
Other related things.
* There are no hooks for setting the LED marked "ERR" on the kasli-soc faceplate.
* What do the ```self.csr_devices.append("virtual_leds")``` map to on kasli-soc? I'll update the sinara wiki. 

### Your System (omit irrelevant parts)

* ARTIQ version: v7.8180.21c6f57
* Hardware is Kasli-SOC v1.1.1.
* Running in nix development environment for kasli-soc release-7.
* Boot is via run_local.sh

jbqubit, 2023-10-10:
It looks like the kasli-soc ERR led is controlled by the [firmware](https://git.m-labs.hk/M-Labs/zynq-rs/src/commit/c15b54f92b3d4e125ae47a0dce7abe4b2bc9e054/experiments/src/main.rs#L190). I can't yet trace out under what circumstances it is illuminated. It would be helpful if this new feature were documented.

Spaqin, 2023-10-11:
``led0`` is also on the Kasli-SoC board, below the power LEDs next to the SFP cage, not on the front panel.

jbqubit, 2023-10-11:
Confirmed that led0 works fine. :)