m-labs/artiq #2245: artiq_ddb_template.py does not add leds to most variants

jbqubit, 2023-10-10:
```artiq/frontend/artiq_ddb_template.py``` does not automatically add led0 and led1 for all kasli variants. Currently they are only added for [```standalone```](https://github.com/m-labs/artiq/blob/96941d7c04969aed295381e8bee3ca093f84cde8/artiq/frontend/artiq_ddb_template.py#L750). This makes getting started harder for new users and debugging of existing systems harder.

Spaqin, 2023-10-11:
Small note, on Kasli 2.0 LED RTIO devices are generated only for standalone; it's not just ``artiq_ddb_template`` that skips them.

jbqubit, 2023-10-11:
> That's because those LEDs are used to indicate presence of a DRTIO link on the SFP port next to each LED when DRTIO is enabled, and are not user-controllable then.

The current implementation reflects this shared use of the LEDs on Kasli v1. It's not (yet) noted on the Kasli wiki. Kasli v2 and kasli-soc now have dedicated user LEDs. Since the .JSON includes a [hardware version field](https://github.com/m-labs/artiq/blob/f7abc156cbc0a371452148d11436e5f241798092/artiq/coredevice/coredevice_generic.schema.json#L25) the behavior of the build scripts as well as ```artiq_ddb_template.py``` can take this into account and to the right thing. Here's what I think that would look like.

* __kasli v1__ : The gateware and ```artiq_ddb_template.py``` support user led0 and led1 in standalone but not master or satellite. 
* __kasli v2 and kasli-soc__: The gateware and ```artiq_ddb_template.py``` support user led0 and led1 for all build variants.

SimonRenblad, 2023-10-18:
> * **kasli v2 and kasli-soc**: The gateware and `artiq_ddb_template.py` support user led0, led1 and led2 for all build variants.

In kasli v2, `user_led0` is not on the front panel, which is most likely why it is not supported (v1.1 also has 3 user leds, none of them on the front panel). Kasli-SoC also does not have a third user led.

As @Spaqin mentioned, an update to the `kasli_generic` gateware would be needed as well as the update to `artiq_ddb_template`. Working on testing this ATM. As for Kasli-SoC I have yet to look at it in detail.

jbqubit, 2023-10-18:
Glad you're working on this @SimonRenblad! The User LEDs are helpful whether or not they are visible from the front panel. 

[image]

(above) kasli v2 has three user LEDs. 

----------------

[image]

[image]

(above) kasli-soc has three user LEDs. They are on two different pages of the schematic.

SimonRenblad, 2023-10-19:
> kasli-soc has three user LEDs. They are on two different pages of the schematic.

Thank you for the catch! From what I could gather, `user_led2` is not made available through migen yet, so I will look into fixing that.

> The User LEDs are helpful whether or not they are visible from the front panel.

I will amend the draft PR to reflect this and also add support for the user leds that are available in kasli v1.1.

SimonRenblad, 2024-02-05:
> AFAICT this is fixed?

Should be fixed. With the exception of the Kasli-SoC L2 LED, which we omitted due to it being controlled by the ps7, all other LEDs should be added in #2262.

jbqubit, 2025-01-21:
The recent Issue I created ([link]([url](https://git.m-labs.hk/M-Labs/artiq-zynq/issues/353))) indeed relates to LD2 on Kasli-SoC. What is ps7? AFAICT this is just another FPGA IO pin. 

[image]

[image]

dnadlinger, 2025-01-21:
ps7 is the processing system, i.e. the ARM cores.