m-labs/artiq #2246: idle kernel does not run at boot

jbqubit, 2023-10-10:
## One-Line Summary

The idle kernel does not run automatically after booting after startup kernel has exited. 

## Issue Details

### Steps to Reproduce

* Both idle kernel and startup kernel are on flash.
* Reboot. 

```
$ artiq_coremgmt log
[     0.000067s]  INFO(runtime): NAR3/Zynq7000 starting...
[     0.005240s]  INFO(runtime): gateware ident: brittonlab-kaslisoc-fun
[     0.016762s]  INFO(libboard_zynq::i2c): PCA9548 detected
[     0.176426s]  INFO(runtime::rtio_clocking): using 100MHz reference to make 125MHz RTIO clock with PLL
[     0.549549s]  INFO(libboard_artiq::si5324): waiting for Si5324 lock...
[     2.930830s]  INFO(libboard_artiq::si5324):   ...locked
[     2.939995s]  INFO(runtime::rtio_clocking): RTIO PLL locked
[     2.950751s]  INFO(libboard_zynq::i2c): PCA9548 detected
[     2.987599s]  INFO(runtime::comms): network addresses: MAC=fc-0f-e7-07-7b-df IPv4=192.168.1.45 IPv6-LL=fe80::fe0f:e7ff:fe07:7bdf IPv6: no configured address
[     3.005658s]  INFO(libboard_artiq::drtio_routing): could not read routing table from configuration, using default
[     3.015832s]  INFO(libboard_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; 3: 3 0; 4: 4 0; }
[     3.032774s]  INFO(runtime::comms): Loading startup kernel...
[     3.038523s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[     3.045228s]  INFO(runtime::comms): Starting startup kernel...
[     3.051049s]  INFO(runtime::kernel::core1): kernel starting
[     6.926713s]  INFO(runtime::kernel::core1): kernel finished
[     6.932277s]  INFO(runtime::comms): Startup kernel finished!
[     7.437087s]  INFO(libboard_zynq::eth): eth: got Link { speed: S1000, duplex: Full }
[    91.935033s]  INFO(runtime::mgmt): received connection
```

* Observe (above) that the idle kernel did not run after the startup kernel finished. 
* Run some other kernel that ends quickly.

```
[   181.878810s]  INFO(runtime::kernel::core1): kernel starting
[   185.011007s]  INFO(runtime::mgmt): received connection
[   185.754475s]  INFO(runtime::kernel::core1): kernel finished
[   185.760036s]  INFO(runtime::comms): peer closed connection
[   185.765510s]  INFO(runtime::comms): Loading idle kernel
[   185.770751s]  INFO(runtime::comms): Running idle kernel
[   185.775961s]  INFO(runtime::kernel::core1): kernel starting
[   188.540284s]  INFO(runtime::mgmt): received connection
```

* Observe (above) that the idle kernel is now running. 

### Expected Behavior

Idle kernel should run if no other kernel is running. 

### Your System

* Operating System:
* ARTIQ version: v7.8180.21c6f57
* Kasli-soc

jbqubit, 2023-10-10:
This seems to be related to https://github.com/m-labs/artiq/issues/2041.

SimonRenblad, 2023-11-01:
I have proposed a PR to address this issue here: https://git.m-labs.hk/M-Labs/artiq-zynq/pulls/276.

Spaqin, 2023-11-07:
Correct, the idle kernel starts immediately after booting, and after startup kernel on RISC-V.