m-labs/artiq #2250: kasli-soc remote reset

jbqubit, 2023-10-11:
The existing [script](https://git.m-labs.hk/M-Labs/artiq-zynq/src/branch/master/local_run.sh) for loading firmware.bin and top.bit onto kasli-soc over JTAG+ethernet works fine from a cold start. However, once the uP is running firmware.bin it no longer works with ```artiq_netboot```. A natural solution would be to reset kasli-soc over USB-JTAG however it looks like that's not supported by openocd. The result is that reflashing requires cycling the power. This isn't great as some systems may be remote or difficult to physically access.

```
Error: timed out while waiting for target halted
TARGET: zynq.cpu.0 - Not halted
```

I tried several openocd scripts with variants on [```reset```](https://openocd.org/doc-release/html/General-Commands.html#Breakpoint-and-Watchpoint-commands) but they fail with the same error (above).

jbqubit, 2023-10-11:
What does "pulse POR" mean?

jbqubit, 2023-10-11:
I've looked at all the scripts and see no reference to POR or any other openocd commands.

jbqubit, 2023-10-12:
Thanks, but your script doesn't work on my system. Perhaps our ftdi URLs differ? I'll check... Yours is ```ftdi://ftdi:4232h/0```. Here's what it looks like on my system. 

```
$ ftdi_urls.py 
Available interfaces:
  ftdi://ftdi:4232:1:54/1      (Quad RS232-HS)
  ftdi://ftdi:4232:1:54/2      (Quad RS232-HS)
  ftdi://ftdi:4232:1:54/3      (Quad RS232-HS)
  ftdi://ftdi:4232:1:54/4      (Quad RS232-HS)
```

None of these nor ```ftdi://ftdi:4232:1:54/0``` permit flashing. 

```
(base) britton@brittonlabbuild:~/m-labs/artiq-zynq$ ./flash.sh 
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Zynq CPU1.
Info : clock speed 1000 kHz
Info : JTAG tap: zynq.tap tap/device found: 0x1372c093 (mfg: 0x049 (Xilinx), part: 0x372c, ver: 0x1)
Info : JTAG tap: zynq.dap tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x4)
Info : zynq.cpu.0: hardware has 6 breakpoints, 4 watchpoints
Info : zynq.cpu.1: hardware has 6 breakpoints, 4 watchpoints
Info : zynq.cpu.1 rev 0, partnum c09, arch f, variant 3, implementor 41
Info : zynq.cpu.1: MPIDR level2 0, cluster 0, core 1, multi core, no SMT
Info : starting gdb server for zynq.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: zynq.tap tap/device found: 0x1372c093 (mfg: 0x049 (Xilinx), part: 0x372c, ver: 0x1)
Info : JTAG tap: zynq.dap tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd), part: 0xba00, ver: 0x4)
Warn : zynq.cpu.0: ran after reset and before halt ...
Info : zynq.cpu.1 rev 0, partnum c09, arch f, variant 3, implementor 41
Error: timeout waiting for DSCR bit change
Error: Error waiting for halt
```

Spaqin, 2023-10-13:
Is a jumper installed on your Kasli-SoC - near the FPGA there's 4 pins, 2 of them are marked "PS_POR_B"?

If it's missing the POR script will do nothing (and also report nothing) and you still won't be able to flash.

jbqubit, 2023-10-13:
Confirmed that this works! Thanks @sbourdeauducq and @Spaqin. 

So, what do the two jumpers do on the PCB?
* PS_POR_B
* PS_SRST_B