m-labs/artiq #2271: I2CError on some configurations Urukul

thomasfire, 2023-11-01:
# Bug Report

## One-Line Summary

I2CError on some configurations' Urukul

## Issue Details

### Steps to Reproduce

1. Build release-7 with following configuation:
```json
{
  "target": "kasli",
  "min_artiq_version": "7.0",
  "variant": "test",
  "hw_rev": "v2.0",
  "base": "satellite",
  "peripherals": [
    {
      "type": "urukul",
      "dds": "ad9910",
      "ports": [0, 1],
      "synchronization": true,
      "clk_sel": 2
    },
    {
      "type": "phaser",
      "ports": [2]
    },
    {
      "type": "grabber",
      "ports": [3, 4]
    },
    {
      "type": "zotino",
      "ports": [5]
    }
  ]
}
```

2. Test Urukuls with artiq_sinara_tester

### Expected Behavior

Tests pass as usual.

### Actual (undesired) Behavior

During calibration of inter-device synchronization:
```
*** Testing Urukul DDSes.
  File "/nix/store/8bzpg145r6vbqssfyva0hw8zcwpv3hiw-python3.9-artiq-7.8185.cc81464/bin/.artiq_sinara_tester-wrapped", line 9, in <module>
    sys.exit(main())
  File "/nix/store/8bzpg145r6vbqssfyva0hw8zcwpv3hiw-python3.9-artiq-7.8185.cc81464/lib/python3.9/site-packages/artiq/frontend/artiq_sinara_tester.py", line 788, in main
    experiment.run(tests)
  File "/nix/store/8bzpg145r6vbqssfyva0hw8zcwpv3hiw-python3.9-artiq-7.8185.cc81464/lib/python3.9/site-packages/artiq/frontend/artiq_sinara_tester.py", line 739, in run
    getattr(self, f"test_{name}")()
  File "/nix/store/8bzpg145r6vbqssfyva0hw8zcwpv3hiw-python3.9-artiq-7.8185.cc81464/lib/python3.9/site-packages/artiq/frontend/artiq_sinara_tester.py", line 299, in test_urukuls
    sync_delay_seed, io_update_delay = self.calibrate_urukul(channel_dev)
  File "/nix/store/8bzpg145r6vbqssfyva0hw8zcwpv3hiw-python3.9-artiq-7.8185.cc81464/lib/python3.9/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/nix/store/8bzpg145r6vbqssfyva0hw8zcwpv3hiw-python3.9-artiq-7.8185.cc81464/lib/python3.9/site-packages/artiq/coredevice/core.py", line 140, in run
    self._run_compiled(kernel_library, embedding_map, symbolizer, demangler)
  File "/nix/store/8bzpg145r6vbqssfyva0hw8zcwpv3hiw-python3.9-artiq-7.8185.cc81464/lib/python3.9/site-packages/artiq/coredevice/core.py", line 130, in _run_compiled
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/nix/store/8bzpg145r6vbqssfyva0hw8zcwpv3hiw-python3.9-artiq-7.8185.cc81464/lib/python3.9/site-packages/artiq/coredevice/comm_kernel.py", line 716, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "/nix/store/8bzpg145r6vbqssfyva0hw8zcwpv3hiw-python3.9-artiq-7.8185.cc81464/lib/python3.9/site-packages/artiq/coredevice/comm_kernel.py", line 698, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.I2CError: failed to ack bus address
```

If you move Urukul ports to 8,9 it will work fine.

### Your System (omit irrelevant parts)

* Operating System: NixOS
* ARTIQ version: 7

Spaqin, 2023-11-01:
Any messages about timing not being met in the build log?

thomasfire, 2023-11-02:
> Any messages about timing not being met in the build log?

None I could find
[build.log](https://github.com/m-labs/artiq/files/13233941/build.log)

bodokaiser, 2024-04-29:
@thomasfire did you solve the problem?

bodokaiser, 2024-04-29:
@Spaqin According to [this forum post](https://forum.m-labs.hk/d/622-eeprom-aliasing-when-using-drtio), the problem arises as there is aliasing for thee `urukulX_eeproms` when generating the `device_db.py`.