m-labs/artiq #2284: Dashboard does not close on Windows

lriesebos, 2023-11-18:
# Bug Report

## One-Line Summary

On Windows, the dashboard does not close.

## Issue Details

### Steps to Reproduce

1. Install ARTIQ 7 on Windows 10 using Conda. Using M-Labs and conda-forge as sources
2. Open the ARTIQ dashboard and connect to a remote master. Now try to close the window.

### Expected Behavior

Window closes.

### Actual (undesired) Behavior

Verbose log output shown here, only the parts from the moment the window close button is pressed:

```
...
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._EventPoller:Starting (proactor: <_IocpProactor overlapped#=14 result#=0>)...
DEBUG:dashboard:qasync._windows._EventWorker:Thread started
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._EventPoller:Starting (proactor: <_IocpProactor overlapped#=14 result#=0>)...
DEBUG:dashboard:qasync._windows._EventWorker:Thread started
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._EventPoller:Starting (proactor: <_IocpProactor overlapped#=12 result#=0>)...
DEBUG:dashboard:qasync._windows._EventWorker:Thread started
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._EventPoller:Starting (proactor: <_IocpProactor overlapped#=12 result#=0>)...
DEBUG:dashboard:qasync._windows._EventWorker:Thread started
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._EventPoller:Starting (proactor: <_IocpProactor overlapped#=11 result#=0>)...
DEBUG:dashboard:qasync._windows._EventWorker:Thread started
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._EventPoller:Starting (proactor: <_IocpProactor overlapped#=10 result#=0>)...
DEBUG:dashboard:qasync._windows._EventWorker:Thread started
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._EventPoller:Starting (proactor: <_IocpProactor overlapped#=9 result#=0>)...
DEBUG:dashboard:qasync._windows._EventWorker:Thread started
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._EventPoller:Starting (proactor: <_IocpProactor overlapped#=8 result#=0>)...
DEBUG:dashboard:qasync._windows._EventWorker:Thread started
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._EventPoller:Starting (proactor: <_IocpProactor overlapped#=7 result#=0>)...
DEBUG:dashboard:qasync._windows._EventWorker:Thread started
DEBUG:dashboard:qasync._windows._EventPoller:Stopping worker thread...
DEBUG:dashboard:qasync._windows._EventWorker:Exiting thread
DEBUG:dashboard:qasync._windows._IocpProactor:Closing
DEBUG:dashboard:asyncio:<_IocpProactor overlapped#=2 result#=0> is running after closing for 1.0 seconds
DEBUG:dashboard:asyncio:<_IocpProactor overlapped#=2 result#=0> is running after closing for 2.0 seconds
DEBUG:dashboard:asyncio:<_IocpProactor overlapped#=2 result#=0> is running after closing for 3.0 seconds
...
```

the last line keeps repeating (tested up to 2 minutes).

### Your System (omit irrelevant parts)

* Operating System: Win 10 (ugh)
* ARTIQ version: 7.8176.6fbfa12
* [conda env export explicit](https://github.com/m-labs/artiq/files/13401174/artiq-dashboard-explicit.txt)

dnadlinger, 2023-11-18:
This is a `qasync` bug, see https://github.com/CabbageDevelopment/qasync/commit/5f6580ee3b865dd6a69f52a081c988ea65f4ff4e. It should go away with qasync 0.25 or 0.26. They have just managed to release a completely broken 0.27.0, so don't update to that (yet).

Do we need to update some dependencies for this somewhere, or provide a newer qasync on the M-Labs channel?

lriesebos, 2023-11-18:
ah good catch, appreciate that. I can work around that, but another option is to set the qasync dependency for artiq to <0.27 for example.

lriesebos, 2023-11-18:
so that would be here https://git.m-labs.hk/M-Labs/nix-scripts/src/branch/master/artiq-fast/conda/artiq.nix#L17 , though actually I don't know why it currently resolves to 0.24.2

lriesebos, 2023-11-20:
should be fixed, see https://github.com/conda-forge/qasync-feedstock/pull/14#event-11008550551

closed 2023-11-20 by lriesebos