m-labs/artiq #2305: mirny/ADF5356 driver doesn't have 52-bits precision

thomasfire, 2024-01-11:
According to [datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/ADF5356.pdf), the chip has 52-bits precision available, however set_frequency function doesn't allow sub-Hz values.

Customer shared their way to use 38 bits:
```py
   @portable
    def calculate_pll(self,f_vco: TFloat, f_pfd: TInt64):
        """
        Calculate fractional-N PLL parameters such that

        ``f_vco`` = ``f_pfd`` * (``n`` + (``frac1`` + ``frac2``/``mod2``) / ``mod1``)

        where
            ``mod1 = 2**24`` and ``mod2 <= 2**28``

        :param f_vco: target VCO frequency
        :param f_pfd: PFD frequency
        :return: ``(n, frac1, (frac2_msb, frac2_lsb), (mod2_msb, mod2_lsb))``
        """
        f_pfd = int64(f_pfd)
        f_vco = float(f_vco)

        # integral part
        n, r = int32(f_vco // f_pfd), f_vco % f_pfd

        # main fractional part
        r *= ADF5356_MODULUS1
        frac1, frac2 = int32(r // f_pfd), int64(r % f_pfd)

        # auxiliary fractional part
        mod2 = f_pfd

        ### ADD THIS TO FIX THE PROBLEM TEMPORARILY, restricts mod2 to within 14 bits (will only write LSB) ###
        while mod2 > (ADF5356_MAX_MODULUS2 >> 14):
            mod2 >>= 1
            frac2 >>= 1
        ### END EDIT ###

        gcd_div = gcd(frac2, mod2)
        mod2 //= gcd_div
        frac2 //= gcd_div
       
        return n, frac1, split_msb_lsb_28b(frac2), split_msb_lsb_28b(mod2)
```

ldaokang, 2024-01-11:
Thanks for opening this issue for this problem. To provide more context, we are currently on Artiq 7 and measuring the frequency output of the Mirny around 5 GHz on a frequency counter. We noticed that there was a Hz-level frequency error when operating at f = 5GHz, which is much larger than the specified 52-bit precision of the chip.

[image]

Frequency was set using the `artiq.coredevice.adf5356.set_frequency` command. In this test, I am using a f_ref = 125 MHz, and get f_PFD = 125 MHz. At 5 GHz, from the fractional-N PLL formula, we get n = 40, frac1 = frac2 = 0, mod2 = 1 (mod1 = 2^24, fixed). As the set frequency deviates from the round 5GHz, only frac2 and mod2 change (frac1 only changes every 125MHz / 2^24 = 7.45 Hz). 

frac2 and mod2 are 28-bit integers nominally, which are each separated into a 14-bit MSB and 14-bit LSB and written to registers 2 and 13 on the ADF5356 chip. From the measured error, I guessed that the MSB on either frac2 or mod2 was wrongly bit-shifted by 1. After messing around a bit, it was only when I coerced them both to 14-bit (such that only the LSB registry was written) did that remove the frequency error (up to the new resolution of 38-bit).

[image]

I'm not sure why writing the MSB register for frac2 and mod2 (REG13) is problematic.

(Also note that the native `artiq.coredevice.adf5356.set_frequency` command rounds the set frequency to the nearest Hz, but it seems like it can be set to a float and that works fine as well)

More details:
Our frequency counter was referenced to 10 MHz from our maser, and the Mirny is clocked externally to a 125 MHz DDS which is also referenced to the 10 MHz maser. The device_db entry:
```
device_db["mirny2_cpld"] = {
    "type": "local",
    "module": "artiq.coredevice.mirny",
    "class": "Mirny",
    "arguments": {
        "spi_device": "spi_mirny2",
        "refclk": 125000000.0,
        "clk_sel": "sma"
    },
}
```

kjarnold, 2025-08-21:
I found the bug in the ADF5356 driver code. In the ADF5356.sync() function reg[13] (the msb of mod2 and frac2) is overwritten for the autocalibration step but then not set back to the correct value.

With the following modifications (changed lines commented by ##) to ADF5356.set_frequency(), ADF5356.sync() and calculate_pll(), the frequency is set correctly to 52-bit precision. We verified this on hardware (4457 Mirny + Almazny v1.1 with ADF5356).

```
    @kernel
    def set_frequency(self, f):
        """
        Output given frequency on output A.

        :param f: 53.125 MHz <= f <= 6800 MHz
        """
        freq = f ##allow f as float, previously cast to int64 which rounded to the nearet 1 Hz

        if freq > ADF5356_MAX_VCO_FREQ:
            raise ValueError("Requested too high frequency")

        # select minimal output divider
        rf_div_sel = 0
        while freq < ADF5356_MIN_VCO_FREQ:
            freq *= 2 ##changed for float
            rf_div_sel += 1

        if (1 << rf_div_sel) > 64:
            raise ValueError("Requested too low frequency")

        # choose reference divider that maximizes PFD frequency
        self.regs[4] = ADF5356_REG4_R_COUNTER_UPDATE(
            self.regs[4], self._compute_reference_counter()
        )
        f_pfd = self.f_pfd()

        # choose prescaler
        if freq > int64(6e9):
            self.regs[0] |= ADF5356_REG0_PRESCALER(1)  # 8/9
            n_min, n_max = 75, 65535

            # adjust reference divider to be able to match n_min constraint
            while n_min * f_pfd > freq:
                r = ADF5356_REG4_R_COUNTER_GET(self.regs[4])
                self.regs[4] = ADF5356_REG4_R_COUNTER_UPDATE(self.regs[4], r + 1)
                f_pfd = self.f_pfd()
        else:
            self.regs[0] &= ~ADF5356_REG0_PRESCALER(1)  # 4/5
            n_min, n_max = 23, 32767

        # calculate PLL parameters
        n, frac1, (frac2_msb, frac2_lsb), (mod2_msb, mod2_lsb) = calculate_pll(
            freq, f_pfd
        )

        if not (n_min <= n <= n_max):
            raise ValueError("Invalid INT value")

        # configure PLL
        self.regs[0] = ADF5356_REG0_INT_VALUE_UPDATE(self.regs[0], n)
        self.regs[1] = ADF5356_REG1_MAIN_FRAC_VALUE_UPDATE(self.regs[1], frac1)
        self.regs[2] = ADF5356_REG2_AUX_FRAC_LSB_VALUE_UPDATE(self.regs[2], frac2_lsb)
        self.regs[2] = ADF5356_REG2_AUX_MOD_LSB_VALUE_UPDATE(self.regs[2], mod2_lsb)
        self.regs[13] = ADF5356_REG13_AUX_FRAC_MSB_VALUE_UPDATE(
            self.regs[13], frac2_msb
        )
        self.regs[13] = ADF5356_REG13_AUX_MOD_MSB_VALUE_UPDATE(self.regs[13], mod2_msb)

        self.regs[6] = ADF5356_REG6_RF_DIVIDER_SELECT_UPDATE(self.regs[6], rf_div_sel)
        self.regs[6] = ADF5356_REG6_CP_BLEED_CURRENT_UPDATE(
            self.regs[6], int32(floor(24 * f_pfd / (61.44 * MHz)))
        )
        self.regs[9] = ADF5356_REG9_VCO_BAND_DIVISION_UPDATE(
            self.regs[9], int32(ceil(f_pfd / 160e3))
        )

        # commit
        self.sync()
```

```
 @kernel
    def sync(self):
        """
        Write all registers to the device. Attempts to lock the PLL.
        """
        f_pfd = self.f_pfd()
        delay(200 * us)         # Slack

        if f_pfd <= 75.0 * MHz:
            for i in range(13, 0, -1):
                self.write(self.regs[i])
            delay(200 * us)
            self.write(self.regs[0] | ADF5356_REG0_AUTOCAL(1))
        else:
            # AUTOCAL AT HALF PFD FREQUENCY

            # calculate PLL at f_pfd/2
            n, frac1, (frac2_msb, frac2_lsb), (mod2_msb, mod2_lsb) = calculate_pll(
                self.f_vco(), f_pfd >> 1
            )
            delay(200 * us)     # Slack

            self.write(
                13
                | ADF5356_REG13_AUX_FRAC_MSB_VALUE(frac2_msb)
                | ADF5356_REG13_AUX_MOD_MSB_VALUE(mod2_msb)
            )

            for i in range(12, 4, -1):
                self.write(self.regs[i])

            self.write(
                ADF5356_REG4_R_COUNTER_UPDATE(self.regs[4], 2 * self.ref_counter())
            )

            self.write(self.regs[3])
            self.write(
                2
                | ADF5356_REG2_AUX_MOD_LSB_VALUE(mod2_lsb)
                | ADF5356_REG2_AUX_FRAC_LSB_VALUE(frac2_lsb)
            )
            self.write(1 | ADF5356_REG1_MAIN_FRAC_VALUE(frac1))

            delay(200 * us)
            self.write(ADF5356_REG0_INT_VALUE(n) | ADF5356_REG0_AUTOCAL(1))

            # RELOCK AT WANTED PFD FREQUENCY

            for i in [13, 4, 2, 1]:  ##13 added to list of regs to be set
                self.write(self.regs[i])

            # force-disable autocal
            self.write(self.regs[0] & ~ADF5356_REG0_AUTOCAL(1))
```

```
@portable
def calculate_pll(f_vco: TFloat, f_pfd: TInt64):  ##allow output frequency to be float
    """
    Calculate fractional-N PLL parameters such that

    ``f_vco = f_pfd * (n + (frac1 + frac2/mod2) / mod1)``

    where
        
    ``mod1 = 2**24`` and ``mod2 <= 2**28``

    :param f_vco: target VCO frequency
    :param f_pfd: PFD frequency
    :return: (``n``, ``frac1``, ``(frac2_msb, frac2_lsb)``, ``(mod2_msb, mod2_lsb)``)
    """
    f_pfd = int64(f_pfd)
    f_vco = float(f_vco) ##changed typecast

    # integral part
    n, r = int32(f_vco // f_pfd), f_vco % f_pfd

    # main fractional part
    r *= ADF5356_MODULUS1
    frac1, frac2 = int32(r // f_pfd), int64(r % f_pfd). ##cast frac2 result as int64

    # auxiliary fractional part
    mod2 = f_pfd

    while mod2 > ADF5356_MAX_MODULUS2:
        mod2 >>= 1
        frac2 >>= 1

    gcd_div = gcd(frac2, mod2)
    mod2 //= gcd_div
    frac2 //= gcd_div

    return n, frac1, split_msb_lsb_28b(frac2), split_msb_lsb_28b(mod2)
```