m-labs/artiq #2309: Impossibility of using spare Urukul alongside with SUServo

thomasfire, 2024-01-19:
# Bug Report

## One-Line Summary

Combining Urukul with and without SUServo in the same configuration result in compiler error when trying to use both.

## Issue Details

### Steps to Reproduce

```from artiq.experiment import *

class DDSTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul2_ch2")
        self.setattr_device("urukul2_cpld")

        self.setattr_device("suservo0")
        self.setattr_device("urukul0_cpld")
        self.setattr_device("suservo0_ch0")

    @kernel
    def run(self):
        self.core.reset()

        ################### Standalone Urukul ###################
        self.urukul2_cpld.init()
        self.urukul2_ch2.init()
        self.urukul2_ch2.sw.on()
        self.urukul2_ch2.set_att(0.0)
        self.urukul2_ch2.set(frequency=100*MHz, phase=0.0, amplitude=1.0)
        #####################################################

        ################### SUServo Urukul ###################
        self.suservo0.init()
        self.suservo0.set_config(enable=1)

        self.suservo0.set_pgia_mu(0, 1)
        self.urukul0_cpld.set_att(0, 0.0)

        self.suservo0_ch0.set_iir(profile=0, adc=0, kp=0.05, ki=-10000.0, g=0.0)
        self.suservo0_ch0.set_dds(profile=0, frequency=70 * MHz, offset=-0.15)
        self.suservo0_ch0.set(en_out=1, en_iir=1, profile=0)
        #####################################################

```

### Expected Behavior

Every part of the code works fine without errors.

### Actual (undesired) Behavior

[image]

### Your System (omit irrelevant parts)

* ARTIQ version: 7
* JSON:
```json
{
    "target": "kasli_soc",
    "min_artiq_version": "7.0",
    "variant": "test",
    "hw_rev": "v1.0",
    "base": "standalone",
    "core_addr": "192.168.1.75",
    "peripherals": [
...
        {
            "type": "suservo",
            "sampler_ports": [8, 9],
            "urukul0_ports": [5, 6],
            "urukul1_ports": [7, 10],
            "clk_sel": 2
        },
        {
            "type": "urukul",
            "dds": "ad9910",
            "ports": [3, 4],
            "clk_sel": 2,
            "synchronization": true
        }
...
    ]
}

```

dnadlinger, 2024-01-19:
To work around this, you can make a subclass of `CPLD` (empty body) that is used in the device db for the suservo Urukuls. This is what we do in Oxford internally.

Of course, we should implement a workaround like that upstream as well (including integration with the JSON machinery which we are not using yet).

thomasfire, 2024-09-11:
Just posting more detailed workaround.

1. Add following to the `device_db.py` file:
  ```py
  from artiq.coredevice.urukul import CPLD
  
  class CPLD2(CPLD):
      pass
  ````

2. Change affected `urukulX_cpld` entries in the `device_db.py`: 
  ```py
  device_db["urukulX_cpld"] = {
      "type": "local",
      "module": "device_db", # switch module to look for to this device_db file
      "class": "CPLD2", # use child class
      "arguments": {
          .... # remain as it was
      }
  }
  ```
  For example, if `urukul0` is standalone, and `urukul1` and `urukul2` are in SUServo, then either only `urukul0` or both `urukul1` and `urukul2`  need this change from p.2.

thomasfire, 2024-12-17:
Update for those on NixOS with ARTIQ installed as a system package: you'll need to add the target directory with the dummy module to the module search path with either way:
1. In `device_db.py`: 
    ```py
    import os
    import sys
    sys.path.append(os.getcwd()) # or any other path
   ```
2. Or by adding it with environment during artiq_run/artiq_compile: `PYTHONPATH=./ artiq_run ...`