m-labs/artiq #2316: LLVM Linker (ld.lld) Invocation Fails on Compilation with MSYS2

nlysne, 2024-01-24:
# Bug Report

## One-Line Summary

When attempting to compile and run an experiment on ARTIQ 8 Beta, the compilation fails with the worker exception `Exception: ld.lld invocation failed`.

## Issue Details

I've been hesitating to submit this issue as I'm still on the fence about whether or not this is a local configuration issue, but I've run out of avenues to try to diagnose what's going on.

### Steps to Reproduce

1. Install MSYS2.
2. Use `pacman` to install `mingw-w64-x86_64-artiq` per instructions on the manual-beta page.
3. Launch instances of `artiq_master` and `artiq_dashboard` in MINGW64.
4. Submit an experiment via the dashboard to be compiled and run on the core.

### Expected Behavior

Experiment object file is compiled, assembled, and loaded onto the core for running by the scheduler. 

### Actual (undesired) Behavior

worker returns an exception and quits out before finishing compilation.
```
root:Terminating with exception (Exception: ld.lld invocation failed: )
Traceback (most recent call last):
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/master/worker_impl.py", line 340, in main
    exp_inst.run()
  File "C:/Users/nkl/QIP/code/artiq/triangle_trap-dev/artiq_scan_framework/scans/scan.py", line 709, in run
    self._run_scan_core(resume)
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/coredevice/core.py", line 149, in run
    self.compile(function, args, kwargs, set_result)
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/coredevice/core.py", line 124, in compile
    library = target.compile_and_link([module])
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/targets.py", line 205, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/targets.py", line 188, in link
    with RunTool([self.tool_ld, "-shared", "--eh-frame-hdr"] +
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/targets.py", line 37, in __enter__
    raise Exception("{} invocation failed: {}".
Exception: ld.lld invocation failed:
artiq.master.scheduler:got worker exception in run stage, deleting RID 759
```

### Your System (omit irrelevant parts)

* Operating System: Windows 10
* ARTIQ version: 8.8593+77c6553.beta-1
* Hardware involved: Xilinx Zync ZC706 (`EK-Z7-ZC706-G`)
<details>
<summary>List of Managed Packages</summary>

```
$ pacman -Q
base 2022.06-1
bash 5.2.026-1
bash-completion 2.11-3
brotli 1.1.0-1
bsdtar 3.7.2-1
bzip2 1.0.8-4
ca-certificates 20230311-1
coreutils 8.32-5
curl 8.5.0-1
dash 0.5.12-1
db 5.3.28-5
file 5.45-1
filesystem 2023.02.07-1
findutils 4.9.0-3
gawk 5.3.0-1
gcc-libs 13.2.0-2
gdbm 1.23-1
getent 2.18.90-4
gettext 0.22.4-1
gmp 6.3.0-1
gnupg 2.4.3-2
grep 1~3.0-6
gzip 1.13-1
heimdal-libs 7.8.0-4
inetutils 2.5-2
info 7.1-2
less 643-1
libargp 20110921-4
libasprintf 0.22.4-1
libassuan 2.5.6-1
libbz2 1.0.8-4
libcurl 8.5.0-1
libdb 5.3.28-5
libedit 20221030_3.1-1
libexpat 2.5.0-1
libffi 3.4.4-1
libgcrypt 1.10.3-1
libgdbm 1.23-1
libgettextpo 0.22.4-1
libgnutls 3.8.2-1
libgpg-error 1.47-1
libgpgme 1.23.2-1
libhogweed 3.9.1-1
libiconv 1.17-1
libidn2 2.3.4-2
libintl 0.22.4-1
libksba 1.6.5-1
liblz4 1.9.4-1
liblzma 5.4.5-1
libnettle 3.9.1-1
libnghttp2 1.58.0-1
libnpth 1.6-2
libopenssl 3.2.0-1
libp11-kit 0.25.3-1
libpcre 8.45-4
libpcre2_8 10.42-1
libpsl 0.21.2-1
libreadline 8.2.007-1
libsqlite 3.44.2-1
libssh2 1.11.0-1
libtasn1 4.19.0-1
libunistring 1.1-2
libutil-linux 2.35.2-4
libxcrypt 4.4.36-1
libxml2 2.12.3-1
libxslt 1.1.39-1
libzstd 1.5.5-1
mingw-w64-x86_64-artiq 8.8593+77c6553.beta-1
mingw-w64-x86_64-artiq-comtools 1.2-1
mingw-w64-x86_64-brotli 1.1.0-1
mingw-w64-x86_64-bzip2 1.0.8-3
mingw-w64-x86_64-c-ares 1.25.0-1
mingw-w64-x86_64-ca-certificates 20230311-1
mingw-w64-x86_64-curl 8.5.0-1
mingw-w64-x86_64-dbus 1.14.10-1
mingw-w64-x86_64-double-conversion 3.3.0-1
mingw-w64-x86_64-expat 2.5.0-1
mingw-w64-x86_64-fontconfig 2.14.2-1
mingw-w64-x86_64-freetype 2.13.2-1
mingw-w64-x86_64-gcc-libgfortran 13.2.0-3
mingw-w64-x86_64-gcc-libs 13.2.0-3
mingw-w64-x86_64-gettext 0.22.4-3
mingw-w64-x86_64-glib2 2.78.4-1
mingw-w64-x86_64-graphite2 1.3.14-3
mingw-w64-x86_64-harfbuzz 8.3.0-1
mingw-w64-x86_64-hdf5 1.14.3-1
mingw-w64-x86_64-http-parser 2.9.4-2
mingw-w64-x86_64-icu 74.2-1
mingw-w64-x86_64-libaec 1.1.2-2
mingw-w64-x86_64-libffi 3.4.4-1
mingw-w64-x86_64-libgit2 1.7.1-1
mingw-w64-x86_64-libiconv 1.17-3
mingw-w64-x86_64-libidn2 2.3.4-1
mingw-w64-x86_64-libjpeg-turbo 3.0.1-1
mingw-w64-x86_64-libpng 1.6.40-1
mingw-w64-x86_64-libpsl 0.21.5-1
mingw-w64-x86_64-libssh2 1.11.0-2
mingw-w64-x86_64-libsystre 1.0.1-5
mingw-w64-x86_64-libtasn1 4.19.0-1
mingw-w64-x86_64-libtre-git r177.07e66d0-1
mingw-w64-x86_64-libunistring 1.1-1
mingw-w64-x86_64-libwinpthread-git 11.0.0.r551.g86a5e0f41-1
mingw-w64-x86_64-libxml2 2.12.4-1
mingw-w64-x86_64-lld 17.0.6-7
mingw-w64-x86_64-llvm-libs 17.0.6-7
mingw-w64-x86_64-lmdb 0.9.31-1
mingw-w64-x86_64-md4c 0.5.1-1
mingw-w64-x86_64-mpdecimal 2.5.1-1
mingw-w64-x86_64-ncurses 6.4.20231217-1
mingw-w64-x86_64-nghttp2 1.59.0-1
mingw-w64-x86_64-openblas 0.3.26-1
mingw-w64-x86_64-openssl 3.2.0-1
mingw-w64-x86_64-p11-kit 0.25.3-1
mingw-w64-x86_64-pcre2 10.42-1
mingw-w64-x86_64-pyqt5-sip 12.13.0-1
mingw-w64-x86_64-python 3.11.7-1
mingw-w64-x86_64-python-aiohttp 3.9.1-1
mingw-w64-x86_64-python-aiosignal 1.3.1-2
mingw-w64-x86_64-python-attrs 23.2.0-1
mingw-w64-x86_64-python-cffi 1.16.0-1
mingw-w64-x86_64-python-colorama 0.4.6-2
mingw-w64-x86_64-python-dateutil 2.8.2-4
mingw-w64-x86_64-python-frozenlist 1.4.1-1
mingw-w64-x86_64-python-h5py 3.10.0-1
mingw-w64-x86_64-python-idna 3.6-1
mingw-w64-x86_64-python-levenshtein 0.23.0-1
mingw-w64-x86_64-python-llvmlite 0.41.1-2
mingw-w64-x86_64-python-lmdb 1.4.1-3
mingw-w64-x86_64-python-multidict 6.0.4-3
mingw-w64-x86_64-python-numpy 1.26.3-1
mingw-w64-x86_64-python-ply 3.11-4
mingw-w64-x86_64-python-prettytable 3.9.0-1
mingw-w64-x86_64-python-pybase64 1.3.2-1
mingw-w64-x86_64-python-pycparser 2.21-3
mingw-w64-x86_64-python-pygit2 1.13.3-1
mingw-w64-x86_64-python-pyqt5 5.15.10-1
mingw-w64-x86_64-python-pyqtgraph 0.13.3-2
mingw-w64-x86_64-python-pythonparser 1.4-1
mingw-w64-x86_64-python-qasync 0.27.1-1
mingw-w64-x86_64-python-rapidfuzz 3.6.1-1
mingw-w64-x86_64-python-regex 2023.12.25-1
mingw-w64-x86_64-python-scipy 1.11.4-1
mingw-w64-x86_64-python-sipyco 1.6-1
mingw-w64-x86_64-python-six 1.16.0-4
mingw-w64-x86_64-python-tqdm 4.66.1-1
mingw-w64-x86_64-python-wcwidth 0.2.13-1
mingw-w64-x86_64-python-yarl 1.9.4-1
mingw-w64-x86_64-qt5-base 5.15.12+kde+r147-2
mingw-w64-x86_64-qt5-svg 5.15.12+kde+r6-1
mingw-w64-x86_64-readline 8.2.010-1
mingw-w64-x86_64-sqlite3 3.45.0-1
mingw-w64-x86_64-tcl 8.6.13-1
mingw-w64-x86_64-termcap 1.3.1-7
mingw-w64-x86_64-tk 8.6.12-2
mingw-w64-x86_64-tzdata 2023d-1
mingw-w64-x86_64-vulkan-headers 1.3.275.0-1
mingw-w64-x86_64-vulkan-loader 1.3.275.0-1
mingw-w64-x86_64-wineditline 2.206-1
mingw-w64-x86_64-xz 5.4.5-1
mingw-w64-x86_64-zlib 1.3.1-1
mingw-w64-x86_64-zstd 1.5.5-1
mintty 1~3.7.0-1
mpfr 4.2.1-1
msys2-keyring 1~20231013-1
msys2-launcher 1.5-1
msys2-runtime 3.4.10-4
nano 7.2-1
ncurses 6.4-2
nettle 3.9.1-1
openssl 3.2.0-1
p11-kit 0.25.3-1
pacman 6.0.2-13
pacman-contrib 1.10.1-1
pacman-mirrors 20231208-1
perl 5.38.2-1
pinentry 1.2.1-1
rebase 4.5.0-4
sed 4.9-1
tar 1.35-2
time 1.9-3
tzcode 2023d-1
util-linux 2.35.2-4
wget 1.21.4-1
which 2.21-4
xz 5.4.5-1
zlib 1.3-1
zstd 1.5.5-1
```
</details>

nlysne, 2024-01-24:
I was giving this another shot after a thought this afternoon. Sadly I didn't get it working, but I did get another more verbose error that might help? Nothing has changed appreciably between the state of the system reported above and this output:
<details>
<summary>Exception: ld.lld invocation failed output:</summary>

```root:Terminating with exception (Exception: ld.lld invocation failed: PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.)
Traceback (most recent call last):
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/master/worker_impl.py", line 340, in main
    exp_inst.run()
  File "C:/Users/nkl/QIP/code/artiq/triangle_trap-dev/artiq_scan_framework/scans/scan.py", line 709, in run
    self._run_scan_core(resume)
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/coredevice/core.py", line 149, in run
    self.compile(function, args, kwargs, set_result)
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/coredevice/core.py", line 124, in compile
    library = target.compile_and_link([module])
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/targets.py", line 205, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/targets.py", line 188, in link
    with RunTool([self.tool_ld, "-shared", "--eh-frame-hdr"] +
  File "C:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/targets.py", line 37, in __enter__
    raise Exception("{} invocation failed: {}".
Exception: ld.lld invocation failed: PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.	Program arguments: ld.lld -shared --eh-frame-hdr -m armelf_linux_eabi --target2=rel -TC:/Users/nkl/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/kernel.ld C:/Users/nkl/msys64/tmp/tmpo2an3kft -x -o C:/Users/nkl/msys64/tmp/tmpni4zcz8l
Exception Code: 0xC0000005
0x00007FFE42F969E2, C:\WINDOWS\SYSTEM32\ntdll.dll(0x00007FFE42F70000) + 0x269E2 byte(s), RtlGetCurrentServiceSessionId() + 0x2192 byte(s)
0x00007FFE42F95B74, C:\WINDOWS\SYSTEM32\ntdll.dll(0x00007FFE42F70000) + 0x25B74 byte(s), RtlGetCurrentServiceSessionId() + 0x1324 byte(s)
0x00007FFE42F947B1, C:\WINDOWS\SYSTEM32\ntdll.dll(0x00007FFE42F70000) + 0x247B1 byte(s), RtlFreeHeap() + 0x51 byte(s)
0x00007FFE424C9C9C, C:\WINDOWS\System32\msvcrt.dll(0x00007FFE424B0000) + 0x19C9C byte(s), free() + 0x1C byte(s)
0x00007FFE424C9366, C:\WINDOWS\System32\msvcrt.dll(0x00007FFE424B0000) + 0x19366 byte(s), _aligned_free() + 0x16 byte(s)
0x00007FFDB86621EA, C:\Users\nkl\msys64\mingw64\bin\libstdc++-6.dll(0x00007FFDB8530000) + 0x1321EA byte(s), _ZdlPvSt11align_val_t() + 0xA byte(s)
0x00007FFD98816890, C:\Users\nkl\msys64\mingw64\bin\libLLVM-17.dll(0x00007FFD970D0000) + 0x1746890 byte(s), _ZN4llvm14lowertypetests20isJumpTableCanonicalEPNS_8FunctionE() + 0x18C0 byte(s)
0x00007FFD986DE1C9, C:\Users\nkl\msys64\mingw64\bin\libLLVM-17.dll(0x00007FFD970D0000) + 0x160E1C9 byte(s), _ZN4llvm13llvm_shutdownEv() + 0x39 byte(s)
0x00007FF6CC73C6FF, C:\Users\nkl\msys64\mingw64\bin\ld.lld.exe(0x00007FF6CC500000) + 0x23C6FF byte(s)
0x00007FF6CC5A59B2, C:\Users\nkl\msys64\mingw64\bin\ld.lld.exe(0x00007FF6CC500000) + 0xA59B2 byte(s)
0x00007FF6CC50AF1B, C:\Users\nkl\msys64\mingw64\bin\ld.lld.exe(0x00007FF6CC500000) + 0xAF1B byte(s)
0x00007FF6CC894969, C:\Users\nkl\msys64\mingw64\bin\ld.lld.exe(0x00007FF6CC500000) + 0x394969 byte(s)
0x00007FF6CC5012EE, C:\Users\nkl\msys64\mingw64\bin\ld.lld.exe(0x00007FF6CC500000) + 0x12EE byte(s)
0x00007FF6CC501406, C:\Users\nkl\msys64\mingw64\bin\ld.lld.exe(0x00007FF6CC500000) + 0x1406 byte(s)
0x00007FFE42577344, C:\WINDOWS\System32\KERNEL32.DLL(0x00007FFE42560000) + 0x17344 byte(s), BaseThreadInitThunk() + 0x14 byte(s)
0x00007FFE42FC26B1, C:\WINDOWS\SYSTEM32\ntdll.dll(0x00007FFE42F70000) + 0x526B1 byte(s), RtlUserThreadStart() + 0x21 byte(s)
```
</details>

nlysne, 2024-02-05:
> Does it crash then?
It did not. As far as I can tell, the output file was generated successfully.
> Why (do I believe it is a problem specifically related to ARTIQ)?
As I can explicitly run the linker with the same files in the MSYS2 environment without issue, and I can compile the code successfully on a conda-managed environment using the same version of ARTIQ.

thomasfire, 2024-05-02:
It does not fail on ld.lld now, but fails on llvm tools when I tried to use artiq_run, because there is no such dependency https://git.m-labs.hk/M-Labs/artiq-extrapkg/src/commit/82601c365c8949865254efa221a3be516e4dae91/flake.nix#L445 .
Will be added with https://git.m-labs.hk/M-Labs/artiq-extrapkg/pulls/12

thomasfire, 2024-05-13:
> @thomasfire What is the status of this Issue?

Can be closed since llvm tools is a dependency now and shipped with installer.

kaolpr, 2024-05-13:
We've experienced this fail today on a fresh MSYS2 installation and ARTIQ installed according to https://m-labs.hk/artiq/manual-beta/installing.html#installing-via-msys2-windows Will this installation path be available or will be deprecated in favor of the installer?

thomasfire, 2024-05-14:
> Yes. @thomasfire what did you test?

`artiq_compile` compiles file after following https://m-labs.hk/artiq/manual-beta/installing.html#installing-via-msys2-windows

martafafara, 2024-05-14:
I tried to use above instruction on my script. I received following result:

$ artiq_compile sinGen.py
Traceback (most recent call last):
  File "C:\msys64\mingw64\bin\artiq_compile-script.py", line 33, in <module>
    sys.exit(load_entry_point('artiq==8.8779+193962f.beta', 'console_scripts', 'artiq_compile')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.11/site-packages/artiq/frontend/artiq_compile.py", line 67, in main
    core.compile(exp.run, [exp_inst], {},
  File "C:/msys64/mingw64/lib/python3.11/site-packages/artiq/coredevice/core.py", line 140, in compile
    library = target.compile_and_link([module])
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/targets.py", line 205, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/targets.py", line 188, in link
    with RunTool([self.tool_ld, "-shared", "--eh-frame-hdr"] +
  File "C:/msys64/mingw64/lib/python3.11/site-packages/artiq/compiler/targets.py", line 37, in __enter__
    raise Exception("{} invocation failed: {}".
Exception: ld.lld invocation failed:

thomasfire, 2024-05-16:
I see, seems this bug is floating and reproduces only a few times, while others may be successful

thomasfire, 2024-05-16:
> I don't think so. Pacman is quite deterministic.

Pacman yes, but the problem is compilation process is not

[image]

thomasfire, 2024-05-16:
Also happens to be different on different windows versions, for example on (tiny) windows 11 it failed only once after reboot, which **feels** like it got timed out because of loading something...
[image]

However on windows 10 it fails once in a few times

thomasfire, 2024-05-16:
> What happens if you run lld manually multiple times with the same inputs?

Yeah, it is different 
[image]

kaolpr, 2024-05-16:
In @martafafara case it also happens to work from time to time, but time after boot does not seem to be relevant.

nlysne, 2024-05-16:
FWIW in our case it would never succeed when compiling through the dashboard (if it matters, as we never tried `artiq_compile` directly), but it worked the few times we tried to directly invoke `ld.lld` with the same inputs. We did not do more than a few repeated tests of the latter.

thomasfire, 2024-05-17:
I just checked lld from clang64 repo on the same file, and it seems like only the mingw64 version occasionally fails. The lld package in conda is built with MSVC2019, which may explain the differences in behavior.
Also the LLVM itself doesn't really support mingw on windows https://llvm.org/docs/GettingStarted.html#hardware , so I doubt  its LLVM's fault.

martafafara, 2024-05-17:
I found your results really optimistic. In my case I receive more` ld.lld `errors then successful compilations. Running any script requires at least few tries, so limiting it to "occasionally" will be really helpful.

thomasfire, 2024-05-17:
@martafafara We are experimenting with clang64 version of ARTIQ, would you mind checking if it works on your OS, since it seems you have it being reproduced more often? The [installer](https://files.m-labs.hk/s/sAonsbgsfn5fgMX) is built from https://git.m-labs.hk/esavkin/artiq-extrapkg/src/branch/2316-move-to-clang . I tried to run `artiq_run` a few times and `artiq_compile` a ~20 times each on win10 and win11 VM with latest updates, and no segfault happened. 
Just as a reminder, it is a standalone installer, and artiq should work out of the box, though installation size and time increased quite significantly.

martafafara, 2024-05-17:
Unfortunately I'm rather the user of Artiq than developer, so I need help of more experienced colleagues. The operation will be done asap

thomasfire, 2024-05-20:
Packages are moved to clang64 environment, they can be installed with `pacman -S mingw-w64-clang-x86_64-artiq` now,  or with _a few clicks_ offline installer https://nixbld.m-labs.hk/job/artiq/extra-beta/msys2-offline-installer/latest . No developer experience required 😊

thomasfire, 2024-05-20:
> Using ARTIQ requires you to do some programming e.g. to write kernels. It is misleading to advertise ARTIQ as a "no developer experience required" control system in any way.

Partially true, but programming is not exactly developing, and developing is not only programming. Based on my experience with analytics'/data scientists' code, there are major differences in the ways we work. In similar way, based on helpdesk, using ARTIQ is far from developing it, and developing is far from using it.

martafafara, 2024-05-20:
Of coarse some knowledge and programming skills are required, but from my point of view Artiq is just a tool. Complicated, but still only the tool that I'm using. The knowledge how it is exactly constructed is for sure fascinating but not necessary. 

I followed the online installer and as for now for 5 scripts runs I've received no errors at all. What a relief... Thank you!