m-labs/artiq #2399: Fastino  garbage output with long sequences

marmeladapk, 2024-04-26:
# Bug Report

## One-Line Summary

Fastino generates garbage output with long transmission sequences.

## Issue Details

### Steps to Reproduce

1. Generate gateware from this json:
```
{
    "target": "kasli",
    "variant": "test",
    "hw_rev": "v2.0",
    "base": "standalone",
    "rtio_frequency": 125e6,
    "peripherals": [
    {
        "type": "fastino",
        "hw_rev": "v1.2.1",
        "ports": [10]
    }
    ]
}
```
3. Run this experiment:
```
from artiq.experiment import *
from numpy import sin, pi

class SystemExample(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("core_dma")
        self.fastino = [self.get_device("fastino" + str(i)) for i in range(1)]

        # self.s = [1.2*sin(i/7) for i in range(32)]
        self.s = [self.fastino[0].voltage_to_mu(1.2*sin(2*pi*i/32)) for i in range(32)]
        print(self.s)

    @kernel
    def run(self):
        self.core.reset()
        self.init()
        delay(10 * ms)
        # This function never returns
        # self.fastino_sine(amplitude=1.2*V)
        self.record_waveform()
        self.play_waveform()

    @kernel
    def init(self):
        for fastino in self.fastino:
            fastino.init()
        self.core.break_realtime()

    @kernel
    def fastino_sine(self, amplitude):
        i = 0
        while True:
            for j in range(8):
                self.fastino[0].set_dac_mu(j, self.s[i%32])
                delay(39200 * ns)
            i += 1
            delay(200*us)

    @kernel
    def record_waveform(self):
        with self.core_dma.record("fastino0"):
            for i in range(100):
                for j in range(32):
                    self.fastino[0].set_dac(j, sin(i / 5)*1.5)
                    delay(392*ns)
                delay(100*us)

    @kernel
    def play_waveform(self):
        dma_handle = self.core_dma.get_handle("fastino0")
        self.core.break_realtime()
        for _ in range(200):
            self.core_dma.playback_handle(dma_handle)
```

### Expected Behavior

Sine output on Fastino.

### Actual (undesired) Behavior

Malformed output.

[image]
[image]
[image]
[image]
[image]

The same behaviour happens with DMA and was tested with other Fastino and Kasli. This happens both in master and standalone variants. Notably, this happens after a long period of time when the signal is generated on <=4  channels.

Error LED on Fastino does not light up.

### Your System (omit irrelevant parts)

* ARTIQ version: release-7, db79100; master baa5834
* Version of the gateware and runtime loaded in the core device: same, respective for release-7 and master
* Hardware involved: Kasli v2.0.2, Fastino v1.2.1

marmeladapk, 2024-04-26:
Ok, I tested this on 4th Fastino and this one is able to sustain such a sequence on all channels for an extended period of time (tens of minutes) before the output becomes garbage.

nkrackow, 2024-05-14:
Hi, (un-)fortunately I was not able to reproduce this. I tried with two Fastino 1.2.1, different EEM ports on a Kasli-SoC and with/without `log2_width = 5`. The output was always a stable sine on all channels (only ch0 for `log2_width = 5` ofc).

The last blue trace looks suspiciously like the runaway of a CIC if the poles and zeros didn't perfectly cancel out, which happens if you somehow get an extra bit into an accumulator.

Did the red LED (indicating communication problems with Kasli) on Fastino ever turn on?

marmeladapk, 2024-05-14:
Red LED was off in all cases. It seems that it was the case of a quirky hw, maybe one that barely passed tests and only in longer sequences exhibited problems. I think we should wait for Fastino v1.3 and see if supply sequencing helps with intermittent issues like this.

nkrackow, 2024-05-14:
While trying to reproduce your issue I also ran into some problems with the communication with Kasli. Sometimes I could turn on the red LED just by touching the negative output of a Fastino channel. Longer DMA sequences or your experiment never worked. It seems to have fixed itself after the solar storm.. :shrug: