m-labs/artiq #2413: Dashboard target name error

vmsch, 2024-05-23:
# Bug Report

## One-Line Summary

When running artiq_dashboard -s x.x.x.x with the msys2 installation, an "IncompatibleServer" error is raised

## Issue Details

### Steps to Reproduce

Full error message:

```
$ artiq_dashboard -s 192.168.1.110

fmayer@pf-2404a MINGW64 ~
$ Traceback (most recent call last):
  File "C:\msys64\mingw64\bin\artiq_dashboard-script.pyw", line 33, in <module>
    sys.exit(load_entry_point('artiq==8.8779+193962f.beta', 'gui_scripts', 'artiq_dashboard')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.11/site-packages/artiq/frontend/artiq_dashboard.py", line 145, in main
    loop.run_until_complete(client.connect_rpc(
  File "C:/msys64/mingw64/lib/python3.11/site-packages/qasync/__init__.py", line 409, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.11/site-packages/sipyco/pc_rpc.py", line 223, in connect_rpc
    await self.select_rpc_target(target_name)
  File "C:/msys64/mingw64/lib/python3.11/site-packages/sipyco/pc_rpc.py", line 232, in select_rpc_target
    target_name = _validate_target_name(target_name, self.__target_names)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.11/site-packages/sipyco/pc_rpc.py", line 52, in _validate_target_name
    raise IncompatibleServer(
sipyco.pc_rpc.IncompatibleServer: valid target name(s): master_dataset_db master_device_db master_experiment_db master_management master_schedule
```

This error appears to occur when in the artiq_dashboard main() an AsyncioClient is created, without an argument for the target name. 
l. 144 in https://github.com/m-labs/artiq/blob/master/artiq/frontend/artiq_dashboard.py
This then calls 
```
if target_name is not None:
                await self.select_rpc_target(target_name)
```
Which probably since the commit f5bf2ba in sipyco, pc_rpc leads to this error message, as target_name is no longer None, but AutoTarget:
https://github.com/m-labs/sipyco/commit/f5bf2ba875340a31a135aea14ad184575ca800ac

The same error occurs when opening a Client without a target_name, but disappears, when a target name is given:

```

>>> cl = Client("192.168.1.110",3251,target_name="master_schedule")
>>> cl = Client("192.168.1.110",3251)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:/msys64/mingw64/lib/python3.11/site-packages/sipyco/pc_rpc.py", line 121, in __init__
    self.select_rpc_target(target_name)
  File "C:/msys64/mingw64/lib/python3.11/site-packages/sipyco/pc_rpc.py", line 129, in select_rpc_target
    target_name = _validate_target_name(target_name, self.__target_names)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/mingw64/lib/python3.11/site-packages/sipyco/pc_rpc.py", line 47, in _validate_target_name
    raise ValueError("Server has multiple targets: " +
ValueError: Server has multiple targets: master_dataset_db master_device_db master_experiment_db master_management master_schedule
```

### Expected Behavior

Dashboard opens 

### Actual (undesired) Behavior

Value error occurs, dashboard does not open

### Your System (omit irrelevant parts)

* Operating System: Windows, MSYS2
* ARTIQ version: artiq==8.8779+193962f.beta

installed with Msys following the artiq 8 installation instructions in the manual

fsagbuya, 2024-05-24:
Hi @vmsch,

Thank you for reporting this issue. May I ask if  you made any modifications to the `artiq_dashboard` code? Additionally, can you provide more details on reproducing the error, particularly how you ran `artiq_master`? I attempted to reproduce the `IncompatibleServer` error on my MSYS2 setup with a similar ARTIQ version `8.8779+193962f.beta` and did not encounter the issue. The primary difference might be that I performed a fresh installation of ARTIQ, [now using clang](https://github.com/m-labs/artiq/commit/2b31d38084416a9ef473f28e21e266ced04e6610). As a potential solution, could you try installing it again or using the new offline [msys2-installer with ARTIQ already available](https://m-labs.hk/artiq/manual-beta/installing.html#installing-via-msys2-windows)?

vmsch, 2024-05-24:
Hi Florian,

We didn’t make any changes to the artiq_dashboard code, and have installed artiq with the latest installation instructions.
I just tried to reinstall artiq with both the offline installer and clang, and both times still got the same error message (all on the windows PC, PC1).
The master is running on a different computer (PC2), a linux machine, where we have a nix installation, and version '8.8674+dcf1bba.beta’.

Maybe that version difference is the issue? Is there a way to install an older version with msys2? We’re reluctant to update our master, because it usually introduces a lot of bugs, so would rather downgrade the version of the gui we’re using.

Our command for starting the master on PC2 is simply
artiq_master --bind 192.168.1.110 -r ~/scratch/repo/ --log-file log/log

From a different linux computer (PC3), which also has artiq version '8.8674+dcf1bba.beta' via nix installed (ie the same as the master PC2), we can successfully open the dashboard remotely connecting to that master (on PC2), without error messages

vmsch, 2024-05-24:
Yes, I think that would be good, and maybe also a possibility with msys to install a specific version.
We use the beta version because we want to use some features introduced quite early in artiq8, but don't really want to have to update the master everytime we have a new PC that wants to connect via the dashboard, I think a stable release every 6 months would be great.

vmsch, 2024-05-24:
Thank you, that worked!

I had to re-install the non-clang msys version to be compatible to the older package type I needed
[](https://nixbld.m-labs.hk/build/157311)
but then it worked and I can start the dashboard without upgrading the master.
Maybe you could add a sentence on how to install older/specfic versions with msys2 in the manual, as most people are new to msys?

Thanks again!

vmsch, 2024-05-24:
Ah ok. We're using the msys installations only for dashboards, everything else is on linux machines, and we'll update everything to the next stable release once it is out, so that should be fine hopefully.