m-labs/artiq #2414: Almazny v1.2 driver doesn't work

thomasfire, 2024-05-24:
# Bug Report

## One-Line Summary

`set()` function in AlmaznyChannel doesn't change the actual state of the Almazny v1.2.

## Issue Details

### Steps to Reproduce

1. Disassemble and flash master firmware from https://github.com/quartiq/mirny to Mirny via JTAG
2. Assemble back
3. Run the script example below

### Expected Behavior

Almazny outputs signal of doubled frequency of Mirny, Almazny LED is up.

### Actual (undesired) Behavior

Mirny outputs signal, Almazny doesn't output anything.

### Your System (omit irrelevant parts)

Hardware:

```json
{
    "target": "kasli",
    "min_artiq_version": "7.0",
    "variant": "test",
    "hw_rev": "v2.0",
    "base": "standalone",
    "core_addr": "192.168.1.42",
    "peripherals": [        
         {
            "type": "mirny",
            "almazny": true,
            "almazny_hw_rev": "v1.1",
            "ports": [0]
        },
        {
            "type": "mirny",
            "almazny": true,
            "almazny_hw_rev": "v1.2",
            "ports": [1]
        }
    ]
}
```

The experiment:

```py
from artiq.experiment import *

class TestAlmazny(EnvExperiment):
    def build(self):
        self.core = self.get_device("core")
        self.almazny = self.get_device("mirny1_almazny0")
        self.ttl_mirny1_sw0 = self.get_device("ttl_mirny1_sw0")
        self.mirny_cpld = self.get_device("mirny1_cpld")
        self.mirny = self.get_device("mirny1_ch0")

    @kernel
    def setup_mirny(self, channel, frequency):
        self.core.break_realtime()
        channel.init()

        channel.set_att(11.5*dB)
        channel.sw.on()
        self.core.break_realtime()

        channel.set_frequency(frequency*MHz)
        delay(5*ms)

    @kernel
    def run(self):
        self.core.reset()
        self.core.break_realtime()
        self.mirny_cpld.init()
        self.setup_mirny(self.mirny, 2000)
        self.almazny.set(20, True, True)
        self.ttl_mirny1_sw0.on()

```

Spaqin, 2024-05-31:
That must've been a mistake on my part, mix-up of the firmwares. I generated the v0.3 CPLD firmware again and it works, with the caveat - only for the first channel. Needs further investigation into the firmware, as the driver is really simple, probably fine as it works with the first channel.

Spaqin, 2024-06-03:
That was a firmware issue (see https://github.com/quartiq/mirny/pull/7) - the driver is actually fine.

closed 2024-06-03 by Spaqin