m-labs/artiq #2453: Minor issues/suggestions on `flake.nix` and `.gitignore`

LEXUGE, 2024-06-20:
# ARTIQ Feature Request

## Problem this request addresses
Several minor issues/suggestions on `flake.nix` and `.gitignore`
- [ ] add `result` produced by `nix` to `.gitignore`, and refactor `.gitignore`.
- [ ] format `flake.nix` by [nixpkgs-fmt](https://github.com/nix-community/nixpkgs-fmt), a formatter used by nixpkgs.
- [ ] use [flake-utils](https://github.com/numtide/flake-utils) to avoid boilerplates like `openocd-bscanspi-f` and manually writing out `pkgs-aarch64` and `pkgs`
- [ ] use [pre-commit-hooks](https://github.com/cachix/git-hooks.nix) to automatically check formatting and lints with `flake8`, `nixpkgs-fmt`, and `rustfmt` before commit and with `nix flake check`.

## Additional context

I locally tested `flake8` lint, it seems the python codebase has a lot of lint errors.

LEXUGE, 2024-06-20:
You can install by cargo.

bloxx12, 2024-12-01:
> * [ ]  format `flake.nix` by [nixpkgs-fmt](https://github.com/nix-community/nixpkgs-fmt), a formatter used by nixpkgs.

archived by now, [nixfmt](https://github.com/NixOS/nixfmt) is the new replacement.
> * [ ]  use [flake-utils](https://github.com/numtide/flake-utils) to avoid boilerplates like `openocd-bscanspi-f` and manually writing out `pkgs-aarch64` and `pkgs`

I'd *heavily* advise against this, the same thing can be done using a few lines of simple nix code. flake-utils just adds an extra abstraction layer.  Furthermore, `artiq` only outputs stuff for two different architectures, so you'd be defining the systems in a list anyways, and at this point it'd be much easier to just do that without an abstraction library.