m-labs/artiq #2498: Coremgmt support for satellites

Spaqin, 2024-07-17:
There's no other interface with satellites than an USB cable for now. With ``artiq_coremgmt``, we should also be able to:

- set config keys, both string- (``sed_spread_enable``) and file-based (``boot`` on Zynq),
- reboot,
- would be also nice to access logs (setting log levels optional IMO).

On artiq_coremgmt side, it would just require the user to specify the satellite number, e.g.:

```
artiq_coremgmt -s 1 reboot
```

evilmav, 2024-11-28:
Any chance this could be backported on the release-8 branch? This feature is thoroughly missed on the SoC satellites.

jordens, 2024-11-28:
As you say: it's a feature. The rist of breakage is too high and it would introduce an incompatibility.