m-labs/artiq #2499: Urukuls in 1-EEM and 2-EEM modes cannot be used together

thomasfire, 2024-07-18:
# Bug Report

## One-Line Summary

Urukuls in 1-EEM and 2-EEM modes cannot be used together.

## Issue Details

### Steps to Reproduce

Device configuration:
```json
{
    "target": "kasli",
    "min_artiq_version": "7.0",
    "variant": "test",
    "hw_rev": "v2.0",
    "base": "standalone",
    "core_addr": "...",
    "peripherals": [
        {
            "type": "urukul",
            "dds": "ad9910",
            "ports": [9, 10],
            "clk_sel": 2
        },
        {
            "type": "urukul",
            "dds": "ad9910",
            "ports": [11],
            "clk_sel": 2
        }
   ]
}
```

Experiment code:

```py
from artiq.experiment import *

class ERROR_urukul0_and1(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("urukul0_ch0")
        self.setattr_device("urukul0_ch1")
        self.setattr_device("urukul0_ch2")
        self.setattr_device("urukul0_ch3")
        self.setattr_device("urukul1_ch0")
        self.setattr_device("urukul1_ch1")
        self.setattr_device("urukul1_ch2")
        self.setattr_device("urukul1_ch3")

    @kernel
    def run(self):
        self.core.reset()
        self.urukul0_ch3.cpld.init()
        self.urukul0_ch3.init()

        self.urukul1_ch3.cpld.init() # deleting these 2 lines makes 
        self.urukul1_ch3.init() # it compiling fine

        self.urukul0_ch3.sw.off()
```

### Expected Behavior

Code compiles and runs without errors.

### Actual (undesired) Behavior

```
artiq_compile test.py
<synthesized>:1:1-1:60: warning: object <artiq.coredevice.ad9910.AD9910 object at 0x7ffed9d0e0d0> of type <instance artiq.coredevice.ad9910.AD9910> does not declare attribute(s) 'sw' as kernel invariant, but other objects of the same type do; the invariant annotation on other objects will be ignored
`<artiq.coredevice.ad9910.AD9910 object at 0x7ffed9d0e0d0>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
test.py:16:1: note: expanded from here
def run(self):
^
test.py:21:9-21:25: note: while inferring a type for an attribute 'urukul1_ch3' of a host object
        self.urukul1_ch3.cpld.init()
        ^^^^^^^^^^^^^^^^
<synthesized>:1:1-1:60: error: host object does not have an attribute 'sw'
`<artiq.coredevice.ad9910.AD9910 object at 0x7ffed9d0e0d0>`
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
test.py:16:1: note: expanded from here
def run(self):
^
test.py:24:9-24:28: note: attribute accessed here
        self.urukul0_ch3.sw.off()
        ^^^^^^^^^^^^^^^^^^^
```

### Versions

ARTIQ stable and master.

### Workaround

We may add an SW device to the 1-EEM Urukul, that points to the mock device with some wrong RTIO channel. In device_db:

```py

device_db["ttl_urukul1_sw_mock"] = {
    "type": "local",
    "module": "artiq.coredevice.ttl",
    "class": "TTLOut",
    "arguments": {"channel": 0x0FFFFF}
}

device_db["urukul1_ch0"] = {
    "type": "local",
    "module": "artiq.coredevice.ad9910",
    "class": "AD9910",
    "arguments": {
        "pll_n": 32,
        "pll_en": 1,
        "chip_select": 4,
        "cpld_device": "urukul1_cpld",
        "sw_device": "ttl_urukul1_sw_mock"
    }
}

# ... other channels
```

dnadlinger, 2024-07-18:
It may be worth to document the slightly neater workaround (make a trivial subclass of AD9910 that's used for one variant, so the compiler doesn't attempt to unify them) and add it to the JSON-driven pipeline as well. As far as the regular user is concerned, NAC3 may well be vapourware, in the sense that it's not an alternative right now, and its not clear when it will be.

dnadlinger, 2024-07-18:
?!