m-labs/artiq #2512: Installation with MSYS fails

jsm8989, 2024-07-23:
# Bug Report

## One-Line Summary

Installing artiq via MSYS as in the (stable) manual fails due to unrecognised scipy dependency.

## Issue Details

### Steps to Reproduce

1. Install MSYS from msys2.org
2. Following the exact Windows (MSYS) installation instructions in manual for 8.8912+cd12600
3. The step `pacman -S mingw-w64-clang-x86_64-artiq` fails

### Expected Behavior

Install artiq

### Actual (undesired) Behavior

* Unable to install package due to unrecognised scipy dependency. I am new to MSYS but I'm assuming the dependency should be [`mingw-w64-x86_64-python-scipy`](https://packages.msys2.org/package/mingw-w64-x86_64-python-scipy) and not `mingw-w64-clang-x86_64-python-scipy` as in the error message.
* $ pacman -S mingw-w64-clang-x86_64-artiq
resolving dependencies...
warning: cannot resolve "mingw-w64-clang-x86_64-python-scipy", a dependency of "mingw-w64-clang-x86_64-artiq"
:: The following package cannot be upgraded due to unresolvable dependencies:
      mingw-w64-clang-x86_64-artiq

:: Do you want to skip the above package for this upgrade? [Y/n]

### Your System (omit irrelevant parts)

* Operating System: Windows 11, MSYS
* ARTIQ version: latest as of 23.07.2024 (8.8912+cd12600)

thomasfire, 2024-07-24:
>  should be [`mingw-w64-x86_64-python-scipy`](https://packages.msys2.org/package/mingw-w64-x86_64-python-scipy)

Well, it wouldn't fix the problem, because the artiq is clang64 package, and packages across platforms are not interchangeable.

For now you can either install the artiq from offline installer or install the previous scipy package from file: https://mirror.msys2.org/mingw/clang64/mingw-w64-clang-x86_64-python-scipy-1.12.0-1-any.pkg.tar.zst .

Here is what happened: https://github.com/msys2/MINGW-packages/pull/21416

jfniedermeyer, 2024-09-24:
>install the previous scipy package from file: https://mirror.msys2.org/mingw/clang64/mingw-w64-clang-x86_64-python-scipy-1.12.0-1-any.pkg.tar.zst.

Until there is a fix, would it be possible to put this in the install documentation? We're trying to upgrade to stable from a Conda-based beta version. A lot of our code was broken until I did this and it might be good for others to know.

thomasfire, 2024-09-25:
> Until there is a fix, would it be possible to put this in the install documentation?

There is some risk that this hotfix will also break, for example if/when dependencies of scipy become incompatible with the old scipy.