m-labs/artiq #2514: Dashboard cannot connect with remote moninj and coreanalyzer proxy

bodokaiser, 2024-07-28:
# Bug Report

## One-Line Summary

Dashboard cannot connect with remote moninj and coreanalyzer proxy

## Issue Details

### Steps to Reproduce

1. Have `artiq_master`, `aqctl_moninj_proxy`, and  `aqctl_coreanalyzer_proxy` run on a remote computer, e.g., `ctrl`.
2. Start the dashboard from a different computer in the network of `ctrl` via `artiq_dashboard -s ctrl`

### Expected Behavior

Dashboard successfully connects to moninj and coreanalyzer proxy.

### Actual (undesired) Behavior

The terminal output of `artiq_dashboard -s ctrl` as well as the logging table of the dashboard GUI constantly print error messages, e.g.:

```
ConnectionRefusedError: [Errno 111] Connect call failed ('::1', 1385, 0, 0)
ERROR:dashboard:artiq.dashboard.moninj:failed to connect to moninj. Is aqctl_moninj_proxy running?
Traceback (most recent call last):
  File "/nix/store/nhxybj21xn8ipaairklzl0jg2jg7fyi7-python3.11-artiq-8.0+554d7bd/lib/python3.11/site-packages/artiq/dashboard/moninj.py", line 725, in mi_connector
    await new_mi_connection.connect(self.mi_addr, self.mi_port)
  File "/nix/store/nhxybj21xn8ipaairklzl0jg2jg7fyi7-python3.11-artiq-8.0+554d7bd/lib/python3.11/site-packages/artiq/coredevice/comm_moninj.py", line 32, in connect
    self._reader, self._writer = await async_open_connection(
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/vv837g1qs47j1hgsgvvyb4cyf82g5zjk-python3.11-sipyco-1.7/lib/python3.11/site-packages/sipyco/keepalive.py", line 80, in async_open_connection
    reader, writer = await asyncio.open_connection(host, port, *args, **kwargs)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/streams.py", line 48, in open_connection
    transport, _ = await loop.create_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/base_events.py", line 1086, in create_connection
    raise exceptions[0]
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/base_events.py", line 1070, in create_connection
    sock = await self._connect_sock(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/base_events.py", line 974, in _connect_sock
    await self.sock_connect(sock, address)
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/selector_events.py", line 638, in sock_connect
    return await fut
           ^^^^^^^^^
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/selector_events.py", line 678, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('::1', 1383, 0, 0)
```

### Your System (omit irrelevant parts)

* ARTIQ version: `ARTIQ v8.0+554d7bd`

### Background

I think [https://github.com/m-labs/artiq/blob/master/artiq/dashboard/moninj.py#L427](moninj.py) should use the server address from the command line instead of the host from the device_db (?) (or localhost). Same with the coreanalyzer

bodokaiser, 2024-07-28:
I updated the `device_db.py` to

```python
# Autogenerated for the main variant
core_addr = "192.168.0.1"

ctrl_addr = "10.163.100.12"

device_db = {
    "core": {
        "type": "local",
        "module": "artiq.coredevice.core",
        "class": "Core",
        "arguments": {
            "host": core_addr,
            "ref_period": 1e-09,
            "analyzer_proxy": "core_analyzer",
            "target": "rv32g",
            "satellite_cpu_targets": {},
        },
    },
    "core_log": {
        "type": "controller",
        "host": "::1",
        "port": 1068,
        "command": "aqctl_corelog -p {port} --bind {bind} " + core_addr,
    },
    "core_moninj": {
        "type": "controller",
        "host": ctrl_addr,
        "port_proxy": 1383,
        "port": 1384,
        "command": "aqctl_moninj_proxy --port-proxy {port_proxy} --port-control {port} --bind {bind} "
        + core_addr,
    },
    "core_analyzer": {
        "type": "controller",
        "host": ctrl_addr,
        "port_proxy": 1385,
        "port": 1386,
        "command": "aqctl_coreanalyzer_proxy --port-proxy {port_proxy} --port-control {port} --bind {bind} "
        + core_addr,
    },
    "core_cache": {
        "type": "local",
        "module": "artiq.coredevice.cache",
        "class": "CoreCache",
    },
    "core_dma": {"type": "local", "module": "artiq.coredevice.dma", "class": "CoreDMA"},
    "i2c_switch0": {
        "type": "local",
        "module": "artiq.coredevice.i2c",
        "class": "I2CSwitch",
        "arguments": {"address": 0xE0},
    },
    "i2c_switch1": {
        "type": "local",
        "module": "artiq.coredevice.i2c",
        "class": "I2CSwitch",
        "arguments": {"address": 0xE2},
    },
}

// ... peripherals
```

but now get a timeout and name resolution errors:

```
The above exception was the direct cause of the following exception:
[... 198 lines cut ...]
    reader, writer = await asyncio.open_connection(host, port, *args, **kwargs)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/streams.py", line 48, in open_connection
    transport, _ = await loop.create_connection(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/base_events.py", line 1046, in create_connection
    infos = await self._ensure_resolved(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/base_events.py", line 1420, in _ensure_resolved
    return await loop.getaddrinfo(host, port, family=family, type=type,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/base_events.py", line 868, in getaddrinfo
    return await self.run_in_executor(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/nix/store/nhxybj21xn8ipaairklzl0jg2jg7fyi7-python3.11-artiq-8.0+554d7bd/lib/python3.11/site-packages/artiq/dashboard/waveform.py", line 78, in _reconnect
    await asyncio.wait_for(new_receiver.connect(self.addr, self.port_proxy),
  File "/nix/store/4rf5qybw37b4lh1g0xczlv14sqdbmnpm-python3-3.11.9/lib/python3.11/asyncio/tasks.py", line 502, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
```

bodokaiser, 2024-07-28:
Problem solved. For future reference:

Introduce a new variable to overwrite the `host` key in the `core_moninj` and `core_analyzer` devices in the `device_db.py`.

Potential pitfalls:

Even `artiq_dashboard` runs local, it uses the `device_db.py` from the computer running `artiq_master`, which might be out of sync, if you haven't commited and pushed changes to the repository used by `artiq_master`.
(In case you are using a git hook, you need to add `artiq_client scan-devices`.
Furthermore, the computer running `artiq_master` is not able to resolve DNS lookups but that's unrelated to ARTIQ.

closed 2024-07-28 by bodokaiser

harrykendell, 2025-01-12:
@bodokaiser what exactly do you mean by 
> Introduce a new variable to overwrite the host key in the core_moninj and core_analyzer devices in the device_db.py.

I'm having the same problem and can't work it out

bodokaiser, 2025-01-12:
If you can post or write me your device_db.py with the new host name, I can modify it for you.

harrykendell, 2025-01-12:
That would be great, thank you.

```
# Autogenerated for the BU2402001 variant
core_addr = "192.168.0.2"
server_addr = "137.222.69.28"

# standard core components
device_db = {
    "core": {
        "type": "local",
        "module": "artiq.coredevice.core",
        "class": "Core",
        "arguments": {
            "host": core_addr,
            "ref_period": 1e-09,
            "target": "rv32g",
            "analyzer_proxy": "core_analyzer",
        },
    },
    "core_log": {
        "type": "controller",
        "host": server_addr,
        "port": 1068,
        "command": "aqctl_corelog -p {port} --bind {bind} " + core_addr,
    },
    "core_analyzer": {
        "type": "controller",
        "host": "::1",
        "port_proxy": 1385,
        "port": 1386,
        "command": "aqctl_coreanalyzer_proxy --port-proxy {port_proxy} --port-control {port} --bind {bind} "
        + core_addr,
    },
    "core_moninj": {
        "type": "controller",
        "host": "::1",
        "port_proxy": 1383,
        "port": 1384,
        "command": "aqctl_moninj_proxy --port-proxy {port_proxy} --port-control {port} --bind {bind} "
        + core_addr,
    },
    "core_cache": {
        "type": "local",
        "module": "artiq.coredevice.cache",
        "class": "CoreCache",
    },
    # allows for direct memory access of RTIO events to play back faster than the cpu can
    "core_dma": {"type": "local", "module": "artiq.coredevice.dma", "class": "CoreDMA"},
    #     I2C switches
    "i2c_switch0": {
        "type": "local",
        "module": "artiq.coredevice.i2c",
        "class": "I2CSwitch",
        "arguments": {"address": 0xE0},
    },
    "i2c_switch1": {
        "type": "local",
        "module": "artiq.coredevice.i2c",
        "class": "I2CSwitch",
        "arguments": {"address": 0xE2},
    },
}

# DAC = Digital to Analog Converter
# 32 channels 16bit DAC
device_db["fastino"] = {
    "type": "local",
    "module": "artiq.coredevice.fastino",
    "class": "Fastino",
[... 217 lines cut ...]
        }
    )

# Aliases
aliases = {
    # 780 MOT
    "suservo_aom_LOCK": "suservo_ch0",
    "suservo_aom_MOT": "suservo_ch1",
    "suservo_aom_IMG": "suservo_ch2",
    "suservo_aom_PUMP": "suservo_ch3",
    "mirny_eom_repump": "mirny_ch0",
    "almazny_eom_repump": "almazny_ch0",
    "shutter_aom_2DMOT": "ttl0",
    "shutter_aom_3DMOT": "ttl1",
    # 852 Lattice
    "suservo_aom_LATX": "suservo_ch4",
    "suservo_aom_LATY": "suservo_ch5",
    # 1064 Cross Dipole Trap
    "suservo_aom_CDT1": "suservo_ch6",
    "suservo_aom_CDT2": "suservo_ch7",
}

device_db |= aliases

```

bodokaiser, 2025-01-12:
I replaced the `host` property in `core_analyzer` and `core_moninj` with `server_addr`, e.g.:

```python
# Autogenerated for the BU2402001 variant
core_addr = "192.168.0.2"
server_addr = "137.222.69.28"

# standard core components
device_db = {
    "core": {
        "type": "local",
        "module": "artiq.coredevice.core",
        "class": "Core",
        "arguments": {
            "host": core_addr,
            "ref_period": 1e-09,
            "target": "rv32g",
            "analyzer_proxy": "core_analyzer",
        },
    },
    "core_log": {
        "type": "controller",
        "host": server_addr,
        "port": 1068,
        "command": "aqctl_corelog -p {port} --bind {bind} " + core_addr,
    },
    "core_analyzer": {
        "type": "controller",
        "host": server_addr,
        "port_proxy": 1385,
        "port": 1386,
        "command": "aqctl_coreanalyzer_proxy --port-proxy {port_proxy} --port-control {port} --bind {bind} "
        + core_addr,
    },
    "core_moninj": {
        "type": "controller",
        "host": server_addr,
        "port_proxy": 1383,
        "port": 1384,
        "command": "aqctl_moninj_proxy --port-proxy {port_proxy} --port-control {port} --bind {bind} "
        + core_addr,
    },
    "core_cache": {
        "type": "local",
        "module": "artiq.coredevice.cache",
        "class": "CoreCache",
    },
    # allows for direct memory access of RTIO events to play back faster than the cpu can
    "core_dma": {"type": "local", "module": "artiq.coredevice.dma", "class": "CoreDMA"},
    #     I2C switches
    "i2c_switch0": {
        "type": "local",
        "module": "artiq.coredevice.i2c",
        "class": "I2CSwitch",
        "arguments": {"address": 0xE0},
    },
    "i2c_switch1": {
        "type": "local",
        "module": "artiq.coredevice.i2c",
        "class": "I2CSwitch",
        "arguments": {"address": 0xE2},
    },
}

# ...
```

Let me know if that works for you :)

harrykendell, 2025-01-12:
This doesn't work sadly. I had actually tried that before but thought you may mean something else.

The issue seems to stem from moninj and analyzer silently failing to start inside artiq_session if trying to bind to anything but "::1"

"host": "::1" starts the processes on local addresses as expected:

`127.0.0.1:1383'
'::1:1383'

but anything else means they don't start.

They do however start fine and bind to the correct addresses if I manually run the command with another binding address.
`aqctl_moninj_proxy --port-proxy 1383 --port-control 1384 --bind 137.222.69.28 192.168.0.2`

`137.222.69.28:1383'
'127.0.0.1:1383'
'::1:1383'

bodokaiser, 2025-01-12:
Ah yes, I think you need to use a `*` as `--bind` parameter, which is quite counter-intuitive.

We managed to register these services using systemd with the following config:

```ini
[Unit]
Description=ARTIQ coreanalyzer
After=sssh.target
Wants=sssh.target

[Service]
User=gitlab-runner
WorkingDirectory=/scratch-local/gitlab-runner/builds/8De2XtsNn/0/rydyb/control/sequences
ExecStart=/nix/var/nix/profiles/default/bin/nix develop --command aqctl_coreanalyzer_proxy --bind=* 192.168.0.1
#Restart=always
#RestartSec=120

[Install]
WantedBy=multi-user.target
```

Does that go in the right direction?

harrykendell, 2025-01-12:
That was my backup solution, as I'd prefer to find where exactly moninj and analyzer are launched from in artiq_session and back out a solution in the device_db settings if possible. Just to avoid having extra setup if this ever moves to another system.