m-labs/artiq #2560: Improve scheduling by interleaving setup and teardown of an experiment

bodokaiser, 2024-08-28:
# ARTIQ Feature Request

## Problem this request addresses

Scheduling of experiment runs is not efficient, i.e., we "waste" about 1 second between experiment runs.

## Describe the solution you'd like

Currently, the [scheduler worker](https://github.com/m-labs/artiq/blob/master/artiq/master/worker_impl.py#L286-L387) runs an experiment run's setup and teardown steps, e.g., `exp.prepare` and `write_results`, sequentially. 

Can we have a worker thread for each stage? Could we also add a precompilation step?

architeuthidae, 2024-08-28:
> Currently, the [scheduler worker](https://github.com/m-labs/artiq/blob/master/artiq/master/worker_impl.py#L286-L387) runs an experiment run's setup and teardown steps, e.g., exp.prepare and write_results, sequentially.

It could hardly be the other way around. On the other hand, experiment stages are already pipelined (and there can be multiple pipelines at any one time.) See [the manual](https://m-labs.hk/artiq/manual/management_system.html#experiment-scheduling). There is also already [support for precompilation](https://m-labs.hk/artiq/manual/core_drivers_reference.html#artiq.coredevice.core.Core.precompile).

architeuthidae, 2024-08-28:
Oh, race condition. Excuse me.

bodokaiser, 2024-08-28:
> What do you expect?
> 
> The experiment needs to be prepared before it can run. Results need to have been obtained before they can be written.
> 

Yes, but I can prepare for the next experiment while writing the results for the previous one.

> If you have two _different_ experiments then those steps should already run in parallel.

Why not extend it for the same experiments (with different argument overwrites)?

> Precompilation is already supported, please read the manual.

Yes, but how would I tell the scheduler to exploit `precompile`? Is it as simple as

```python
class Experiment(EnvExperiment):

  def prepare(self):
    self._run = self.precompile(self.run)

  def run(self):
    self._run()
```

?

> > i.e., we "waste" about 1 second between experiment runs.
> 
> What operating system is this? Each experiment runs in a different worker process. How much of that second is just creating a process? Though even then, it does not seem worthwhile to optimize since this only affects the first experiment in a sequence, then the rest gets pipelined.

We are using Linux amd64. Yes, you are right, a starting point is get a better understanding of the actual bottlenecks. I can provide an analysis in a few weeks (also testing the new NAC3 compiler).

dnadlinger, 2024-08-28:
> Yes, but I can prepare for the next experiment while writing the results for the previous one.

This should already be the case. The next-to-be-run experiment is already `prepare`d while the currently active one is still `running`.

dnadlinger, 2024-11-27:
> This makes me wonder, why this is not done in the first place. I can only imagine a case where a result of a previous experiment influences how a consecutive experiment is executed.

That's exactly it. In fact, this is often the natural case, as the "previous experiment" might be a calibration experiment that updates some dataset with a new π time, frequency, etc. that the new experiment ought to pull in. One can of course opportunistically precompile experiments and then check afterwards whether any dataset/controller/… dependencies changed. It is possible to write a framework that achieves this as well, but baseline ARTIQ is a bit limited here, as you can't start to execute `run()`(where those datasets would be read in, etc.) without being in the `running` phase, as the experiment might already start accessing hardware, etc.