m-labs/artiq #2578: Kasli-SoC build instructions fail to build

jbqubit, 2024-09-24:
The updated ARTIQ developer build [documentation](https://m-labs.hk/artiq/manual/building_developing.html#building-artiq) provides several methods for building for Kasli-SoC. Both methods fail for various reasons. Tests are conducted on a system that successfully builds for kasli. 

# nix makeArtiqZynqPackage
```
$ vivado
****** Vivado v2022.2 (64-bit)
  **** SW Build 3671981 on Fri Oct 14 04:59:54 MDT 2022
  **** IP Build 3669848 on Fri Oct 14 08:30:02 MDT 2022
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

$ nix build --print-build-logs --impure --expr 'let fl = builtins.getFlake "git+https://git.m-labs.hk/m-labs/artiq-zynq?ref=release-8"; in (fl.makeArtiqZynqPackage {target="kasli_soc"; variant="master"; json=/home/britton/experiment/cryo0/cryo0.json;}).kasli_soc-master-sd' 

kasli_soc-master-gateware> DIO (EEM0) starting at RTIO channel 0x000000
kasli_soc-master-gateware> DIO (EEM1) starting at RTIO channel 0x00000c
kasli_soc-master-gateware> Urukul (EEM2) starting at RTIO channel 0x000014
kasli_soc-master-gateware> Urukul (EEM4) starting at RTIO channel 0x00001a
kasli_soc-master-gateware> Sampler (EEM6) starting at RTIO channel 0x000020
kasli_soc-master-gateware> Zotino (EEM10) starting at RTIO channel 0x000023
kasli_soc-master-gateware> USER LED at RTIO channel 0x000026
kasli_soc-master-gateware> USER LED at RTIO channel 0x000027
kasli_soc-master-gateware> /etc/profile: line 37: /opt/Xilinx/Vivado/2022.2/settings64.sh: No such file or directory
kasli_soc-master-gateware> Traceback (most recent call last):
kasli_soc-master-gateware>   File "/nix/store/a2iy41gccss952qhh8vnaivxjla29alq-gateware/kasli_soc.py", line 672, in <module>
kasli_soc-master-gateware>     main()
kasli_soc-master-gateware>   File "/nix/store/a2iy41gccss952qhh8vnaivxjla29alq-gateware/kasli_soc.py", line 668, in main
kasli_soc-master-gateware>     soc.build(build_dir=args.g)
kasli_soc-master-gateware>   File "/nix/store/vbv516vz6qi4m4wgavbdb1m1qwq7qnc4-python3-3.11.9-env/lib/python3.11/site-packages/migen_axi/integration/soc_core.py", line 169, in build
kasli_soc-master-gateware>     self.platform.build(self, *args, **kwargs)
kasli_soc-master-gateware>   File "/nix/store/vbv516vz6qi4m4wgavbdb1m1qwq7qnc4-python3-3.11.9-env/lib/python3.11/site-packages/migen/build/xilinx/platform.py", line 59, in build
kasli_soc-master-gateware>     return self.toolchain.build(self, *args, **kwargs)
kasli_soc-master-gateware>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
kasli_soc-master-gateware>   File "/nix/store/vbv516vz6qi4m4wgavbdb1m1qwq7qnc4-python3-3.11.9-env/lib/python3.11/site-packages/migen/build/xilinx/vivado.py", line 224, in build
kasli_soc-master-gateware>     _run_vivado(build_name)
kasli_soc-master-gateware>   File "/nix/store/vbv516vz6qi4m4wgavbdb1m1qwq7qnc4-python3-3.11.9-env/lib/python3.11/site-packages/migen/build/xilinx/vivado.py", line 69, in _run_vivado
kasli_soc-master-gateware>     raise OSError("Subprocess failed")
kasli_soc-master-gateware> OSError: Subprocess failed
kasli_soc-master-gateware> 
error: builder for '/nix/store/bs3sksq2xlx8iw9dgjldfn224aq1r6fq-kasli_soc-master-gateware.drv' failed with exit code 1
error: 1 dependencies of derivation '/nix/store/a3s5lylm1rqlvb9jam38bfi3fgf7sb5v-kasli_soc-master-sd.drv' failed to build
```

# Zynq incremental build

```
$ git status
On branch release-8
Your branch is up to date with 'origin/release-8'.

$ pwd
~/experiment/artiq-zynq/src

$ rm -rf ~/experiment/artiq-zynq/build

$ gateware/kasli_soc.py -g ../build/gateware /home/britton/experiment/cryo0/cryo0.json

$ make TARGET=kasli_soc GWARGS="/home/britton/experiment/cryo0/cryo0.json" runtime
mkdir -p ../build
python gateware/kasli_soc.py -r ../build/pl.rs -c ../build/rustc-cfg -m ../build/mem.rs /home/britton/experiment/cryo0/cryo0.json
DIO (EEM0) starting at RTIO channel 0x000000
DIO (EEM1) starting at RTIO channel 0x00000c
Urukul (EEM2) starting at RTIO channel 0x000014
Urukul (EEM4) starting at RTIO channel 0x00001a
Sampler (EEM6) starting at RTIO channel 0x000020
Zotino (EEM10) starting at RTIO channel 0x000023
USER LED at RTIO channel 0x000026
USER LED at RTIO channel 0x000027
cd runtime && \
XBUILD_SYSROOT_PATH=`pwd`/../../build/sysroot \
cargo xbuild --release \
	--target-dir ../../build/firmware \
[... 151 lines cut ...]
            "pll_n": 40
        },
        {
            "type": "urukul",
            "hw_rev": "v1.3",
            "ports": [4, 5],
            "clk_sel": 1,
            "refclk": 100e6,
            "pll_vco": 5,
            "pll_n": 40
        },
	{
		"type": "sampler",
		"hw_rev": "v1.2",
		"ports": [6,7]
	}, 
        {
            "type": "zotino",
            "hw_rev": "v1.1",
            "ports": [10]
        }
   ]
}

```

SimonRenblad, 2024-09-24:
> /opt/Xilinx/Vivado/2022.2/settings64.sh: No such file or directory

Did you run `nix develop git+https://git.m-labs.hk/m-labs/artiq-zynq\?ref=release-8` before running this command? I copied the provided JSON and was able to build it using both methods on my machine, as long as I am in the devShell. For the incremental method, I also run the `make` command with the `-B` flag to ensure that the manifests get updated correctly. 

> gateware/kasli_soc.py -g ../build/gateware /home/britton/experiment/cryo0/cryo0.json

Did this command not output anything, or was it cut for length? Normally this would take some time for the gateware to build...

Here is an additional link that may be helpful: https://git.m-labs.hk/sinara-hw/assembly/src/branch/master/src/sw_sup/setup_build_pc.md#installation. AFAIK it is not linked in the manual but there are some forum posts that discuss this.

jbqubit, 2024-09-24:
Thanks for the quick advice.

I made the following edits to my system. 
1) In ```~/.local/share/nix/trusted-settings.json``` add the following

```
{
   "extra-sandbox-paths":{
      "/opt":true
   },
   "extra-substituters":{
      "https://nixbld.m-labs.hk":true
   },
   "extra-trusted-public-keys":{
      "nixbld.m-labs.hk-1:5aSRVA5b320xbNvu30tqxVPXpld73bhtOeH6uAjRyHc=":true
   }
}

```

2) In ```~/.config/nix/nix.conf``` add ```extra-sandbox-paths = /opt```

And both build methods now work.

If I revert (2) I see that the makeArtiqZynqPackage error recurs. 

I'm running Ubuntu 2022.04 LTS.

architeuthidae, 2024-09-24:
ACK. Will see about adding the additional Nix configuration tips to the manual as soon as I've gotten the chance to try and recreate the original problem (just have to get a Vivado install going again first...)