m-labs/artiq #2583: 100MHz sysclk setting still generates 125MHz bootstrap clock

Spaqin, 2024-10-01:
This creates issues with UART tuning word mismatch, for one. SDRAM could also potentially be affected (?).

For a quick investigation, tried outputting 100MHz in [RtioSysCrg](https://github.com/m-labs/misoc/blob/master/misoc/targets/kc705.py#L103), that causes kc705 not to boot with SDRAM Initialization Fail. May require some further tuning.

Spaqin, 2024-10-02:
Kasli 2.0 works fine with 100MHz bootstrap (PR incoming), but kc705 does not boot - requires further investigation. At very least it proves that it's kc705-specific part that causes issues, rather than the bulk of ARTIQ gateware.

dhslichter, 2024-10-02:
@sbourdeauducq the NIST QC2 variant ran for many years with a KC705 core device clocked externally at 100 MHz, without issues. My impression was that everything was clocked at 100 MHz in this instance, but https://forum.m-labs.hk/d/795-100-mhz-satellite-kc705-variants/4 seems to suggest that even in that old version there were some things on 100 MHz and others on 125 MHz. Is this true? This issue is blocking our ability to use KC705 as a DRTIO satellite for development work.

Spaqin, 2024-10-04:
m-labs/misoc#149 covers the fix for the issue of having bootstrap clock on wrong frequency on Kasli and kc705.

However, I cannot find any way to get kc705 actually running with 100MHz. Tried adjusting memory timings in multiple ways, and deriving a 500MHz clock just for ddram phy. Always fails at read leveling.

<details>
<summary> Boot log </summary>
```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2024 M-Labs Limited

Bootloader CRC passed
Gateware ident 9.0+unknown.beta;nist_qc2_satellite
Initializing SDRAM...
Write leveling scan:
Module 7:
00000111111111111100000000000000
Module 6:
00000111111111111100000000000000
Module 5:
00000001111111111110000000100000
Module 4:
00000000111111111111100000100000
Module 3:
00000000000001111111111111111111
Module 2:
10000000000001111111111111111111
Module 1:
11000000000000111111111111111111
Module 0:
11100000000000111111111111111111
Write leveling: 15* 14* 13* 13 8 7 4 5 done
Read bitslip: 7 6 5 4 3 2 
Read leveling scan:
Module 7:
00000000000000000000000000000000
Module 6:
00000000000000000000000000000000
Module 5:
00000000000000000000000000000000
Module 4:
00000000000000000000000000000000
Module 3:
00000000000000000000000000000000
Module 2:
00000000000000000000000000000000
Module 1:
00000000000000000000000000000000
Module 0:
00000000000000000000000000000000
Read leveling: Zero window: 7: 0-0 (0)
SDRAM initialization failed
Halting.
No boot medium.
```

</details>

Spaqin, 2024-10-09:
fixed with m-labs/misoc#149

closed 2024-10-09 by Spaqin