m-labs/artiq #2585: Web UI

jordens, 2024-10-02:
# ARTIQ Feature: Web interface

## Problem this request addresses

The ARTIQ GUI components (dashboard and browser) use Qt. The interface employs a (dated) MDI style UI design. ARTIQ GUI applets run into process separation imposed by e.g. Wayland. While the GUI operation is mostly decoupled from the (headless) components they talk to (master, moninj proxy, etc), their codebases are still integrated. The installation of ARTIQ with support for just the GUI components is possible and supported but it still runs into the same installation woes and hurdles that the installation of the main ARTIQ components hit. While the ARTIQ installation experience has changed and arguably improved over the years, we're considering adding a different user interface to ARTIQ.

## Describe the solution you'd like

A Web interface that offers comparable functionality to the ARTIQ Qt applications.

A web application allows controlling an ARTIQ system from any computer with zero download and installation hurdles. There may be issues with browser compatibility though. Another issue may be design differences for touch screens or keyboard/mouse. A web UI can hopefully integrate better with network security requirements imposed by IT departments. It may also play nicer with containerized (headless) ARTIQ installations which then don't need graphical access and UI libraries/desktop environments.
There should be no functional conflict between the web application and the Qt dashboard.
The scope of the MVP will be less than what the Qt apps can currently do.

A draft prioritization of features (along the lines of the docks):

* Schedule viewer/editor
* Log
* UI state persistence
* Experiment applets
* Dataset viewer/editor
* Moninj
* Experiment Explorer
* Experiment windows (arguments)
* Analyzer waveform
* ...

## Additional context

* Web UI toolkits

dhslichter, 2024-10-10:
I like the idea of a lighter-weight, more portable GUI option in a browser window, which would be nice for various instances. One of the big challenges always with the ARTIQ GUI is managing the inevitable clutter from all of these features. Is there a vision around how clutter reduction/organization would be done? For example, would there be a main page from which e.g. new browser tabs are opened for scheduler, log, experiments, dataset viewer/editor, etc? Are you envisioning that all these features would be available (in some appropriately switchable way) within a single browser tab?

jordens, 2024-10-11:
We'll see. This is just a heads up that we might start working on this.