m-labs/artiq #2604: Attributes not synced at kernel exit if there is a pending exception

lriesebos, 2024-10-31:
# Bug Report

## One-Line Summary

No attribute syncing when kernel exits with an exception.

## Issue Details

### Steps to Reproduce

See the following example code:

```python

class MyExperiment(EnvExperiment):

    def build(self):
        self.setattr_device('core')
        self.attr_sync_var = 0

    def run(self):
        try:
            self._run()
        except RuntimeError:
            pass
        assert self.attr_sync_var == 1

    @kernel
    def _run(self):
        self.attr_sync_var += 1
        raise RuntimeError
```

### Expected Behavior

The assertion passes. The same as what would happen if the `RuntimeError` was not raised because the exception raised after the variable was modified.

### Actual (undesired) Behavior

The assertion fails. Because of the `RuntimeError`, the kernel exits with a "pending" exception. I expect that is the cause. The logs show that attribute synchronization (from the kernel to the host) is not performed. Hence, `attr_sync_var` is still 0. Exception propagation still works fine.

This behavior is at least seen in ARTIQ 6 and 7. I have not tested 8. If there is agreement that this behavior is incorrect then I would like backporting of this fix to be considered for versions down to ARTIQ 6 if possible but at least 7 (current old stable).

### Your System (omit irrelevant parts)

* Operating System: Ubuntu
* ARTIQ version: 7.8211

lriesebos, 2024-11-16:
@sbourdeauducq any opinions about this? intended behavior, or a bug?

jordens, 2024-11-16:
It's incorrect behavior but changing it in stable releases is not worth the breakage.