m-labs/artiq #2652: Artiq compiler reads wrong return types from `@rpc def return_many(): return ([4], [True],)`

dimitsev, 2025-01-14:
# Bug Report

## One-Line Summary

In the function below, the Artiq compiler thinks that both return types are `TList[TInt32]`:

```
def return_many() -> TTuple([TList(TInt32), TList(TBool)]):
    return ([4], [True],)
```

## Issue Details

[https://m-labs.hk/artiq/manual-beta/compiler.html#basic-artiq-python](https://m-labs.hk/artiq/manual-beta/compiler.html#basic-artiq-python) explicitly shows this tip, which we have tested with success:

```
def return_many() -> TTuple([TInt32, TFloat, TStr]):
    return (4, 12.34, "hello",)
```

Surprisingly, this also works for us:

```
def return_many() -> TTuple([TList(TInt32), TList(TFloat), TList(TStr)]):
    return ([4], [12.34], ["hello"],)
```

However, our function at the top with signature `-> TTuple([TList(TInt32), TList(TBool)])` does **not** work for us.

Maybe there is a problem that only affects `TBool` or `TList(TBool)`?

### Minimal self-contained examples

```
from artiq.experiment import EnvExperiment, kernel, rpc, host_only
from artiq.language.types import TNone, TBool, TInt32, TFloat, TStr, TList, TTuple

def return_many() -> TTuple([TList(TInt32), TList(TBool)]):
    return ([4], [True],)

def get_type(obj) -> TStr:
    return str(type(obj))

class TestSatellites(EnvExperiment):
    
    def build(self) -> TNone:
        self.setattr_device("core")
        (self.a, self.b) = return_many()
        print(" ----    @rpc-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
    
    @kernel
    def run(self) -> TNone:
        self.core.reset()
        print(" ---- @kernel-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
```

outputs

```
 ----    @rpc-environment ----
type(a) = <class 'int'>
type(b) = <class 'bool'>
 ---- @kernel-environment ----
type(a) = <class 'numpy.int32'>
type(b) = <class 'numpy.int32'>
```

whereas

```
from artiq.experiment import EnvExperiment, kernel, rpc, host_only
from artiq.language.types import TNone, TBool, TInt32, TFloat, TStr, TList, TTuple

def return_many() -> TTuple([TList(TInt32), TList(TFloat), TList(TStr)]):
    return ([4], [12.34], ["hello"],)

def get_type(obj) -> TStr:
    return str(type(obj))

class TestSatellites(EnvExperiment):
    
    def build(self) -> TNone:
        self.setattr_device("core")
        (self.a, self.b, self.c) = return_many()
        print(" ----    @rpc-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
        print("type(c) =", get_type(self.c[0]))
    
    @kernel
    def run(self) -> TNone:
        self.core.reset()
        print(" ---- @kernel-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
        print("type(c) =", get_type(self.c[0]))
```

outputs

```
 ----    @rpc-environment ----
type(a) = <class 'int'>
type(b) = <class 'float'>
type(c) = <class 'str'>
 ---- @kernel-environment ----
type(a) = <class 'numpy.int32'>
type(b) = <class 'float'>
type(c) = <class 'str'>
```

### Why this is a problem

```
from artiq.experiment import EnvExperiment, kernel, rpc, host_only
from artiq.language.types import TNone, TBool, TInt32, TFloat, TStr, TList, TTuple

def return_many() -> TTuple([TList(TInt32), TList(TBool)]):
    return ([4], [True],)

def get_type(obj) -> TStr:
    return str(type(obj))

@kernel
def do_something(
    a : TList(TInt32),
    b : TList(TBool),
):
    x = 1

class TestSatellites(EnvExperiment):
    
    def build(self) -> TNone:
        self.setattr_device("core")
        (self.a, self.b) = return_many()
        print(" ----    @rpc-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
    
    @kernel
    def run(self) -> TNone:
        self.core.reset()
        print(" ---- @kernel-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
        do_something(a=self.a, b=self.b)
```

fails with output

```
 ----    @rpc-environment ----
type(a) = <class 'int'>
type(b) = <class 'bool'>
minimal_example.py:32:34-32:40: error: cannot unify list(elt=bool) with list(elt=numpy.int32) for attribute 'b': numpy.int32 is incompatible with bool
        do_something(a=self.a, b=self.b)
                                 ^^^^^^
minimal_example.py:32:34-32:40: note: expression of type list(elt=bool)
        do_something(a=self.a, b=self.b)
                                 ^^^^^^
minimal_example.py:32:34-32:38: note: expression of type <instance artiq_run_minimal_example.TestSatellites {
                __objectid__: numpy.int32,
                core: <instance artiq.coredevice.core.Core>,
                a: list(elt=numpy.int32),
                b: list(elt=numpy.int32)
        }>
        do_something(a=self.a, b=self.b)
                                 ^^^^
```

whereas

```
from artiq.experiment import EnvExperiment, kernel, rpc, host_only
from artiq.language.types import TNone, TBool, TInt32, TFloat, TStr, TList, TTuple

def return_many() -> TTuple([TList(TInt32), TList(TFloat), TList(TStr)]):
    return ([4], [12.34], ["hello"],)

def get_type(obj) -> TStr:
    return str(type(obj))

@kernel
def do_something(
    a : TList(TInt32),
    b : TList(TFloat),
    c : TList(TStr),
):
    x = 1

class TestSatellites(EnvExperiment):
    
    def build(self) -> TNone:
        self.setattr_device("core")
        (self.a, self.b, self.c) = return_many()
        print(" ----    @rpc-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
        print("type(c) =", get_type(self.c[0]))
    
    @kernel
    def run(self) -> TNone:
        self.core.reset()
        print(" ---- @kernel-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
        print("type(c) =", get_type(self.c[0]))
        do_something(a=self.a, b=self.b, c=self.c)
```

succeeds with output

```
 ----    @rpc-environment ----
type(a) = <class 'int'>
type(b) = <class 'float'>
type(c) = <class 'str'>
 ---- @kernel-environment ----
type(a) = <class 'numpy.int32'>
type(b) = <class 'float'>
type(c) = <class 'str'>
```

### To make really really sure this is an Artiq bug and not my code's bug

```
from artiq.experiment import EnvExperiment, kernel, rpc, host_only
from artiq.language.types import TNone, TBool, TInt32, TFloat, TStr, TList, TTuple

def return_many() -> TTuple([TList(TInt32), TList(TBool)]):
    return ([4], [True],)

@kernel
def do_something(
    a : TList(TInt32),
    b : TList(TBool),
):
    x = 1

class TestSatellites(EnvExperiment):
    
    def build(self) -> TNone:
        self.setattr_device("core")
        (self.a, self.b) = return_many()
        print(" ----    @rpc-environment ----")
    
    @kernel
    def run(self) -> TNone:
        self.core.reset()
        print(" ---- @kernel-environment ----")
        do_something(a=self.a, b=self.b)
```

fails with output

```
 ----    @rpc-environment ----
minimal_example.py:25:34-25:40: error: cannot unify list(elt=bool) with list(elt=numpy.int32) for attribute 'b': numpy.int32 is incompatible with bool
        do_something(a=self.a, b=self.b)
                                 ^^^^^^
minimal_example.py:25:34-25:40: note: expression of type list(elt=bool)
        do_something(a=self.a, b=self.b)
                                 ^^^^^^
minimal_example.py:25:34-25:38: note: expression of type <instance artiq_run_minimal_example.TestSatellites {
                __objectid__: numpy.int32,
                core: <instance artiq.coredevice.core.Core>,
                a: list(elt=numpy.int32),
                b: list(elt=numpy.int32)
        }>
        do_something(a=self.a, b=self.b)
                                 ^^^^
```

### Your System

* Operating System: Ubuntu Desktop 22.04.5 LTS (with some GUI packages manually purged)
* Artiq version: latest commit at the time of writing (January 2025) on branch master with `git rev-parse HEAD` output `63db4af1fcd9c5cb305a3eee2c0f82bf3de95563` and `artiq_client --version` output `ARTIQ v9.0+unknown.beta`
* Version of the gateware and runtime loaded in the core device: built from [artiq-zynq's](https://git.m-labs.hk/m-labs/artiq-zynq) latest commit at the time of writing (January 2025) on branch `master` with `git rev-parse HEAD` output `a0281e498927d5c8467dd18f28aa497cb75150fd`
* Hardware involved: one Kasli SoC v1.1.1 master with 12 DIO_SMA v1.4; one Kasli SoC v1.1.1 satellite with 2 Fastino v1.2; one Kasli SoC satellite v1.1.1 satellite with 1 Clocker v.1.1 and 6 Urukul v1.5

dnadlinger, 2025-01-14:
Your problem, at least this example, does not seem to be related to RPC:
```
from artiq.experiment import EnvExperiment, kernel, rpc, host_only
from artiq.language.types import TNone, TBool, TInt32, TFloat, TStr, TList, TTuple

def return_many() -> TTuple([TList(TInt32), TList(TBool)]):
    return ([4], [True],)

def get_type(obj) -> TStr:
    return str(type(obj))

class TestSatellites(EnvExperiment):
    
    def build(self) -> TNone:
        self.setattr_device("core")
        (self.a, self.b) = return_many()
        print(" ----    @rpc-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
    
    @kernel
    def run(self) -> TNone:
        self.core.reset()
        print(" ---- @kernel-environment ----")
        print("type(a) =", get_type(self.a[0]))
        print("type(b) =", get_type(self.b[0]))
```

dimitsev, 2025-01-14:
Okay, but then the compiler simply interprets `[True]` as being of type `TList(TInt32)`? How else can this happen?

Sorry, I don't know how the compiler works in detail. All I know is that this is a bug and maybe somebody knowledgeable can fix it. More likely, however, they will just switch to [NAC3](https://git.m-labs.hk/M-Labs/nac3), which I support whole-heartedly

SimonRenblad, 2025-01-17:
@dimitsev thank you for raising an issue! I tracked down the issue to line 302 in "artiq/compiler/embedding.py":

```python
    def fast_quote_list(self, value):
        elts = [None] * len(value)
        is_T = False
        if len(value) > 0:
            v = value[0]
            is_T = True
            if isinstance(v, int):
                T = int
...
```
You can verify in a REPL that:

`isinstance(False, int)`

evaluates to `True`.

There are no checks for `bool` or `numpy._bool` before `ininstance(v, int)`, and as a consequence the list elements are marked as `Int`. My assuption is that the `IntMonomorphizer` then converts this to `Int32`.

I was able to, with a quick and dirty change to `fast_quote_list`, verify a fix for the issue for the example you provided.

dimitsev, 2025-01-17:
@SimonRenblad Wonderful! And thank you for looking into it. Will you push the fix?

SimonRenblad, 2025-01-17:
See [#2654](https://github.com/m-labs/artiq/pull/2654).