m-labs/artiq #2662: DRTIO: Link errors and Sequence errors

JammyL, 2025-01-29:
# Bug Report

## One-Line Summary

After an nondeterministic period of time in our experiments we encounter both link errors and sequence errors. These put two of our satellite devices in a bad state (the ones showing sequence errors) requiring a restart.
I have also opened a discussion [here](https://forum.m-labs.hk/d/894-drtio-link-errors-and-sequence-errors), but an issue feels more appropriate.

## Issue Details

After both reinitialisation of devices and a core.reset devices on the offending satellites are unresponsive. After a power cycle of the satellite or a reset of the master FPGA (artiq_flash start) the satellites come back online.

There is an added effect of experiments hanging. This can be patched by replacing the rtio_input_data calls for Sampler and SUServo with the timestamped alternative in the locations linked below.
https://github.com/m-labs/artiq/blob/c1f2ff371784ad47463969b98c455cf66abb2b6a/artiq/coredevice/suservo.py#L149
https://github.com/m-labs/artiq/blob/c1f2ff371784ad47463969b98c455cf66abb2b6a/artiq/coredevice/spi2.py#L241

The above behaviour leads me to believe that no `rtio_output` events are being triggered on these satellites, thus there is no input data to read, causing the experiments to hang.

### Steps to Reproduce

We are unsure. The behaviour is non-deterministic. The same experiment may produce an error after 10 minutes or 2 hours.

### Expected Behavior

Running experiments without any errors. This is usually the case for up to several hours.

### Actual (undesired) Behavior

After a non-deterministic amount of experiment run time (10mins to 2hours) we see the following errors.
These errors show up across experiments that have run perfectly well for a few years.

DEST #0
```
[  1542.604249s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] error(s) found (0x04):
[  1542.610091s] ERROR(runtime::rtio_mgt::drtio): [LINK#0] timeout attempting to get remote buffer space
[  1542.619944s]  WARN(runtime::rtio_mgt::drtio): [LINK#1] unsolicited aux packet: TSCAck
[  1542.627026s] ERROR(runtime::rtio_mgt::drtio): [LINK#1] error(s) found (0x04):
[  1542.634141s] ERROR(runtime::rtio_mgt::drtio): [LINK#1] timeout attempting to get remote buffer space
[  1542.645159s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0006:_1762eom
[  1542.655393s] ERROR(runtime::rtio_mgt::drtio): [DEST#2] RTIO sequence error involving channel 0x0003:_ground_dp
[  1542.867581s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1543.081299s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1545.149344s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0003:_650dp
[  1545.363290s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0008:suservo0
[  1555.827323s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0003:_650dp
[  1556.041276s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1556.255132s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1556.468229s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0004:_614dp
[  1556.682110s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0011:_493eom_11_ttl
[  1556.896094s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1557.109112s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1557.322117s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0002:_493sigma
[  1557.535115s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1557.748161s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1557.961308s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0003:_650dp
[  1558.175227s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0006:_1762eom
[  1558.389158s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1558.602279s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0005:_1762sp
[  1560.615320s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0003:_650dp
[  1560.829287s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0008:suservo0
```

DEST #1 
```
[  2784.146092s] ERROR(satman): received packet of an unknown type
[  2784.150586s] ERROR(satman): timeout attempting to get buffer space from CRI, destination=0xc0
[  2784.159099s] ERROR(satman): write underflow, channel=0, timestamp=1545247125414, counter=1545372549488, slack=-125424074
[  2784.173638s] ERROR(satman): received packet of an unknown type
[  2784.178139s] ERROR(satman): received truncated packet
[  2784.183159s] ERROR(satman): write underflow, channel=1, timestamp=437905711215, counter=1545400101
```

DEST #2 
```
[  2930.554069s] ERROR(satman): write underflow, channel=6, timestamp=1270499749083, counter=1545372648680, slack=-274872899597
[  2930.563936s]  INFO(satman): TSC loaded from uplink
[  2930.568706s] ERROR(satman): received packet of an unknown type
[  2930.574510s] ERROR(satman): received truncated packet
[  2930.579547s] ERROR(satman): timeout attempting to get buffer space from CRI, destination=0x57
[  2930.588077s] ERROR(satman): write underflow, channel=2, timestamp=1546005020011, counter=12363432125432, slack=-10817427105421
[  2943.699773s] ERROR(satman): write underflow, channel=3, timestamp=1558521121013, counter=12376550453472, slack=-10818029332459
[  3012.501525s] ERROR(satman): write underflow, channel=3, timestamp=1627322209389, counter=124453
```

DEST #3 
Shows nothing on the log - it isn't involved in this particular experiment.

### Your System (omit irrelevant parts)

Master process running on Ubuntu 20.04

ARTIQ version: 9.unknown.beta. Commit: c1f2ff371784ad47463969b98c455cf66abb2b6a
Gateware version: 9.unknown.beta. Commit: c1f2ff371784ad47463969b98c455cf66abb2b6a

* Hardware involved:
- Master Kasli v2.0 (DEST #0) 
- - DIO BNC v1.3
- - Urukul ad9910 v1.5
- - Fastino v1.1
- - Fastino v1.1
- Satellite kasli v2.0 (DEST #1)
  - SUServo
    - Urukul v.1.5
    - Urukul v.1.5
    - Sampler v.2.2
  - BNC IO rev1.1
- Satellite kasli v2.0.2 (DEST #2)
  - SUServo
    - Sampler v2.3
    - Urukul v1.5.2
    - Urukul v1.5.2
  - Sampler v2.3
- Satellite kasli v2.0 (DEST #3)
  - Urukul v1.5
  - Fastino v1.2
  - Fastino v1.2

sbourdeauducq, 2025-12-04:
Did you check the FPGA temperatures and/or tried adding a fan tray?

lukas-gerster, 2025-12-12:
Some more information from our side, as it may be the same issue. The link between satellite crate and master crate gets lost and cannot be restored. The channels are reported as unknown, and on restarting the connection we receive an `Urukul AD9910 AUX_DAC mismatch` on the DDS, that can only be fixed by power cycling the satellite. That would line up with incomplete transactions bricking the DDS chip, but doesn't give much information on why the link went down to begin with.

The system was running without issues for several months before the first of these errors occurred. Time between errors also varies between 3 minutes and 36 hours, though we do have exceptionally bad runs where it happens every half hour.

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2025 M-Labs Limited

Bootloader CRC passed
Gateware ident 8.9024+a1bf0e1;qudits
Initializing SDRAM...
Read leveling scan:
Module 1:
00000001111111111000000000000000
Module 0:
00000001111111111000000000000000
Read leveling: 11+-5 11+-5 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000012s]  INFO(runtime): ARTIQ runtime starting...
[     0.003901s]  INFO(runtime): software ident 8.9024+a1bf0e1;qudits
[     0.009989s]  INFO(runtime): gateware ident 8.9024+a1bf0e1;qudits
[     0.016100s]  INFO(runtime): log level set to INFO by default
[     0.021806s]  INFO(runtime): UART log level set to INFO by default
[     0.159310s]  INFO(runtime::rtio_clocking): using 10MHz reference to make 125MHz RTIO clock with PLL
[     0.422603s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     2.976153s]  INFO(board_artiq::si5324):   ...locked
[     2.979770s]  INFO(runtime::rtio_clocking): Switching sys clock, rebooting...

 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2025 M-Labs Limited

Bootloader CRC passed
Gateware ident 8.9024+a1bf0e1;qudits
Initializing SDRAM...
Read leveling scan:
Module 1:
00000001111111111000000000000000
Module 0:
00000001111111111100000000000000
Read leveling: 11+-5 12+-5 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000012s]  INFO(runtime): ARTIQ runtime starting...
[     0.003904s]  INFO(runtime): software ident 8.9024+a1bf0e1;qudits
[     0.009989s]  INFO(runtime): gateware ident 8.9024+a1bf0e1;qudits
[     0.016102s]  INFO(runtime): log level set to INFO by default
[     0.021805s]  INFO(runtime): UART log level set to INFO by default
[     0.173298s]  INFO(runtime::rtio_clocking): Clocking has already been set up.
[     0.204527s]  INFO(runtime): network addresses: MAC=68-27-19-9f-bc-16 IPv4=192.168.1.57/0 IPv6-LL=fe80::6a27:19ff:fe9f:bc16/10 IPv6=no configured address
[     0.218290s]  INFO(board_artiq::drtio_routing): could not read routing table from configuration, using default
[     0.227082s]  INFO(board_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; 3: 3 0; }
[     0.238215s]  WARN(runtime::rtio_mgt): error reading device map (key not found), device names will not be available in RTIO error messages
[     0.249413s]  INFO(runtime::rtio_mgt): SED spreading disabled by default
[     0.257500s]  INFO(runtime::mgmt): management interface active
[     0.268741s]  INFO(runtime::session): accepting network sessions
[     0.273500s]  INFO(runtime::session): running startup kernel
[     0.279348s]  INFO(runtime::session): no startup kernel found
[     0.284867s]  INFO(runtime::session): no connection, starting idle kernel
[     0.299433s]  INFO(runtime::session): no idle kernel found
[     0.303740s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link RX became up, pinging
[     0.699048s]  INFO(board_artiq::grabber): grabber0 locked: 39MHz
[     0.905300s]  INFO(board_artiq::grabber): grabber0 alignment success
[     1.110082s]  INFO(board_artiq::grabber): grabber0 frame size: 512x512
[     5.882051s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] remote replied after 27 packets
[     6.115388s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link initialization completed
[     6.121820s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[     6.129585s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[     6.135184s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 128
[    28.702231s]  INFO(runtime::mgmt): new connection from 192.168.88.42:40314
[    38.971994s]  INFO(runtime::session): new connection from 192.168.88.42:43802
[    39.100159s]  INFO(runtime::kern_hwreq): resetting RTIO
[    39.109724s]  INFO(kernel): Initialize TTL devices...
[    39.114402s]  INFO(kernel): Initialize Phaser devices...
[    39.176970s]  INFO(kernel): Initialize DDS CPLD devices...
[    39.185877s]  INFO(kernel): Initialize DDS devices...
[   109.005566s] ERROR(runtime::rtio_mgt::drtio): [DEST#1] RTIO sequence error involving channel 0x0011:unknown

```

**Master:**
Kasli v2.02
3x Urukul
1x Phaser
1x Grabber
2x DIO

**Satellite:**
Kasli v2.02
2x Phaser
2x Urukul

Link:
SFP Connectors  SFP-10G23-BX10

lukas-gerster, 2025-12-14:
Thanks for the info. We will look around for electromagnetic interference, we already replaced link cables and the satellite Kasli to no effect. I'll report our findings once we have some results.

Calling core reset unfortunately does not work for us due to the urukul board locking up, only rebooting the Kasli does not help there. 

Our trap drive is on an external function generator and isolated from sinara already, our ion loss seems to primarily stem from our automation detecting this as a molten ion crystal and trying to recover the crystal without any refreeze.