m-labs/artiq #2664: `artiq_ddb_template` does not handle AD9910 & AD9912 Urukuls together on the same Kasli / Kasli SoC

dimitsev, 2025-01-30:
# Bug Report

1. When at least one AD9910 Urukul and at least one AD9912 Urukul are present together on the same Kasli / Kasli SoC and
2. when `device_db.py` is generated in the usual manner with `artiq_ddb_template`,

then `self.setattr_device(f"urukul5_cpld"); self.urukul5_cpld.init()` for the AD9912 Urukuls will fail at run-time. (I think, at least.)

Fact is that the following work-around is suggested by knowledgeable people:

- Prepend `import urukul_workaround` to `device_db.py`. Create `urukul_workaround.py` in the same directory as `device.py` with contents:

```
from artiq.coredevice import urukul
class CPLD(urukul.CPLD):
    pass
```

- Find `device_db["urukul5_cpld"]` for all AD9912 Urukuls in `device_db.py` and change `"module": "artiq.coredevice.urukul"` to `"module": "urukul_workaround"`.

I don't want to execute the above steps manually every single time, so I have written short script that executes them automatically.

## Suggested bug fix

**EITHER** make `artiq_ddb_template` execute the above steps. The affected lines are:

- [artiq/frontend/artiq_ddb_template.py#L270](https://github.com/m-labs/artiq/blob/master/artiq/frontend/artiq_ddb_template.py#L270)
- [artiq/frontend/artiq_ddb_template.py#L540](https://github.com/m-labs/artiq/blob/master/artiq/frontend/artiq_ddb_template.py#L540)

**OR** fix whatever needs to be fixed under the hood to let AD9910 Urukuls and AD9912 Urukuls find their CPLD classes correctly.

Spaqin, 2025-01-30:
Related #2309 (simpler workaround is to put the dummy CPLD class in device_db itself)

dimitsev, 2025-01-30:
Honestly, I only know that Creotech gave us the workaround described above for one of our crates that contains 5x AD9910 Urukul and 1x AD9912 Urukul. I haven't tested anything myself yet.