m-labs/artiq #2680: Builtin "print" in kernel should be an async rpc top avoid RTIO underflows

systemofapwne, 2025-02-13:
# Bug Report

Builtin `print` in kernel should be an async rpc to avoid RTIO underflows

Short summary.

When calling `print` within a kernel, e.g. for debugging a certain part of kernel code, it can cause an RTIO underflow. Especially, when `print` is called close to timing critical calls to hardware.
The nature of the cause is, that `print` is not an async rpc so the kernel waits for the function to return. The amount of time for this to happen varries and mostly depends on network delays.

### Steps to Reproduce
Run this code with the `print` statement. It will throw an RTIO underflow.
Run the the code instead with my custom `kernel_print` method, that is an async RPC, it will work as expected

```python
from artiq.language import *
from artiq.experiment import *

@rpc(flags={"async"})
def kernel_print(*args, **kwargs): print(*args,**kwargs)

class RTIOUnderflow(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl6")    

    def prepare(self):
        pass

    @kernel
    def run(self) -> TNone:
        self.core.break_realtime()

        self.ttl6.off()
        print("Hello there")            # This builtin function causes an underflow
        kernel_print("Hello there")     # This custom function works
        self.ttl6.on()
```

### Expected Behavior

No RTIO underflow

### Actual (undesired) Behavior

RTIO underflow thrown with a negative slack of roughly the network delay (e.g. a few ms)

### Your System (omit irrelevant parts)

* Operating System: Arch Linux
* ARTIQ version: ARTIQ v9.9013+38e0d6b.beta
* Version of the gateware and runtime loaded in the core device: Same as ARTIQ environment above
* Hardware involved: Kasli with master & satellite crate

thatschatt, 2025-02-13:
Sure, but it wouldn't *hurt* to make print async would it? From what I remember the last time I checked, the difference (on a Kasli 2.0) async RPC takes hundreds of μs, versus several ms for synchronous. So in a lot of cases that would be fast enough to slip in a quick print() for debug or diagnostic purposes.

As far as I can tell, print is already just an RPC call to the standard python function, so this should just be a matter of putting an async rpc decorator on print() in the global namespace, right?

systemofapwne, 2025-02-13:
> Sure, but it wouldn't _hurt_ to make print async would it? From what I remember the last time I checked, the difference (on a Kasli 2.0) async RPC takes hundreds of μs, versus several ms for [...]

This is also what my opinion is, no matter if using `print` for debugging is good practice or not. 

> As far as I can tell, print is already just an RPC call to the standard python function, so this should just be a matter of putting an async rpc decorator on print() in the global namespace, right?

Yes, it absolutely is. We already tried to simply overwrite `print` with our own definition but the compiler explicitly replaces the symbol with an rcp call to the inbuild version:  https://github.com/m-labs/artiq/blob/77029874364ff8493081e8000ef964c9ba1ca965/artiq/compiler/prelude.py#L36