m-labs/artiq #2718: Core Reset Doesn't Clear Satellite FIFOs

connorgoham, 2025-04-07:
# Bug Report

## One-Line Summary

Running ``self.core.reset()`` on the master core device doesn't empty events already queued in the FIFO of a satellite device.

## Issue Details

### Steps to Reproduce

Minimal code to reproduce:

`
    from artiq.experiment import *
    
    class DRTIOCoreResetBug(EnvExperiment):
        """DRTIO Core Reset Minimal Bug Example"""
    
        def build(self):
            self.setattr_device("core")
            self.setattr_device("scheduler")
            self.master_ttl = self.get_device("ttl14")      # TTL device on our master Kasli
            self.satellite_ttl = self.get_device("ttl55")   # TTL device on our satellite Kasli
    
    
        @kernel
        def run(self):
    
            self.core.reset()   # initialize crate
    
            while not self.scheduler.check_pause():
                # trigger pulses
                self.core.break_realtime()
                with parallel:
                    self.master_ttl.pulse(100 * us)
                    self.satellite_ttl.pulse(100 * us)
                delay(100 * us)
                self.core.wait_until_mu(now_mu())    # ensure trigger pulses occur before core reset
    
                delay(1 * ms)
    
                # test pulses
                for _ in range(5):
                    with parallel:
                        self.master_ttl.pulse(100 * us)
                        self.satellite_ttl.pulse(100 * us)
                    delay(100 * us)
    
                self.core.reset()                       # test reset
    
                delay(5 * ms)
    
            self.core.wait_until_mu(now_mu())
`

### Expected Behavior

All events already queued in all device FIFOs are cleared upon calling ``self.core.reset()`` on the master device (i.e. not as part of a subkernel). In the case of the above code, the 5 TTL pulses scheduled on both the master & satellite kaslis never occur, only the preceeding trigger pulses do.

### Actual (undesired) Behavior

Events scheduled in the master FIFOs are cleared as expected, however events in the satellite FIFOs are not. In the case of the above example code, the satellite produces 5 TTL pulses out after the initial trigger pulse.

TTL scope traces:

[image]

Scope key:
- ch#1 (yellow): master TTL
- ch#3 (pink): satellite TTL

Core log upon boot:

> [     0.000067s]  INFO(runtime): NAR3/Zynq7000 starting...
[     0.005242s]  INFO(runtime): gateware ident: 8+b1ddb89e;brittonlab-legacy-trap
[     0.017639s]  INFO(libboard_zynq::i2c): PCA9548 detected
[     0.181447s]  INFO(runtime::rtio_clocking): using 100MHz reference to make 125MHz RTIO clock with PLL
[     0.554629s]  INFO(libboard_artiq::si5324): waiting for Si5324 lock...
[     2.738176s]  INFO(libboard_artiq::si5324):   ...locked
[     2.793995s]  INFO(runtime::rtio_clocking): SYS CLK switched successfully
[     2.805965s]  INFO(libboard_zynq::i2c): PCA9548 detected
[     2.843612s]  INFO(runtime::comms): network addresses: MAC=fc-0f-e7-07-7b-df IPv4=192.168.1.76 IPv6-LL=fe80::fe0f:e7ff:fe07:7bdf IPv6: no configured address
[     2.875396s]  INFO(libboard_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; }
[     2.887624s]  INFO(runtime::rtio_mgt): SED spreading disabled by default
[     2.907284s]  INFO(runtime::comms): Loading startup kernel...
[     2.913074s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[     2.919801s]  INFO(runtime::comms): Starting startup kernel...
[     2.925631s]  INFO(ksupport::kernel::core1): kernel starting
[    22.214959s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link RX became up, pinging
[    28.483947s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] remote replied after 31 packets
[    28.517541s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link initialization completed
[    28.525299s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[    28.531987s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 128
[    29.580728s]  INFO(ksupport::kernel::core1): kernel finished
[    29.586376s]  INFO(runtime::comms): Startup kernel finished!
[    29.592133s]  INFO(runtime::comms): Loading idle kernel
[    29.597319s]  INFO(runtime::comms): Running idle kernel
[    29.602552s]  INFO(ksupport::kernel::core1): kernel starting
[    30.092042s]  INFO(libboard_zynq::eth): eth: got Link { speed: S1000, duplex: Full }
[    37.320226s]  INFO(runtime::mgmt): received connection

### Your System (omit irrelevant parts)

* Operating System: Ubuntu 22.04.5 LTS
* ARTIQ version: ARTIQ v8.9006+17be223
* Version of the gateware and runtime loaded in the core device: gateware ident: 8+b1ddb89e
* Hardware involved: Kalsi SoC v1.1.1 for both master & satellite.

connorgoham, 2025-04-10:
I must be missing something... All I see output to the satellite UART log is boot related messages:

> rabi@legacy0:~/experiment/legacy-trap$ cat /dev/ttyUSB7

                 	__________   __
                	/ ___/__  /  / /
                	\__ \  / /  / /
               	___/ / / /__/ /___
              	/____/ /____/_____/

             	(C) 2020-2022 M-Labs

[ 	0.019994s]  INFO(szl): Simple Zynq Loader starting...
[ 	0.025200s] DEBUG(libboard_zynq::clocks::source): Set ARM_PLL to 2000000000 Hz
[ 	0.007040s] DEBUG(libboard_zynq::clocks::source): Set IO_PLL to 1000000000 Hz
[ 	0.016259s] DEBUG(libboard_zynq::clocks::source): Set DDR_PLL to 1066666666 Hz
[ 	0.023608s] DEBUG(libboard_zynq::ddr): DDR 3x/2x clocks: 533333328/355555552
[ 	0.030777s] DEBUG(libboard_zynq::ddr): DDR DCI clock: 10062892 Hz (divisors=2*53)
[ 	0.042000s] DEBUG(libboard_zynq::sdio): Reset SDIO!
[ 	0.046947s] DEBUG(libboard_zynq::sdio): Changing clock frequency to 400000
[ 	0.053895s]  INFO(szl): Card inserted. Mounting file system.
[ 	0.074543s] DEBUG(libboard_zynq::sdio): Changing clock frequency to 25000000
[ 	0.081676s] DEBUG(libboard_zynq::sdio::sd_card): Getting bus width
[ 	0.088279s] DEBUG(libboard_zynq::sdio::sd_card): 4 bit support
[ 	0.094175s] DEBUG(libboard_zynq::sdio::sd_card): Changing bus width
[ 	0.102001s] DEBUG(libboard_zynq::sdio): Set block size to 512
[ 	0.108509s] DEBUG(libconfig::sd_reader): Partition ID: C
[ 	0.114958s]  INFO(szl): Loading gateware
[ 	0.119284s] DEBUG(libconfig::bootgen): Partition header pointer = C80
[ 	0.126123s] DEBUG(libconfig::bootgen): Unencrypted length = B0889
[ 	0.132280s] DEBUG(libconfig::bootgen): Partition start address: B630
[ 	1.142381s] DEBUG(libboard_zynq::devc): Invalidate DCache for bitstream buffer
[ 	1.155456s] DEBUG(libboard_zynq::devc): Init preload FPGA
[ 	1.160921s] DEBUG(libboard_zynq::devc): Toggling PROG_B
[ 	1.189032s] DEBUG(libboard_zynq::devc): Waiting for done
[ 	1.194410s] DEBUG(libboard_zynq::devc): Init postload FPGA
[ 	1.199964s]  INFO(szl): Loading runtime
[ 	1.204208s] DEBUG(libconfig::bootgen): Partition header pointer = C80
[ 	1.211052s] DEBUG(libconfig::bootgen): Unencrypted length = B062
[ 	1.217126s] DEBUG(libconfig::bootgen): Unencrypted length = 23E1A
[ 	1.223287s] DEBUG(libconfig::bootgen): Partition start address: BBEC0
[ 	1.434548s]  INFO(szl): Preparing for runtime execution
[ 	1.440186s]  INFO(szl): executing payload
[ 	0.000067s]  INFO(satman): ARTIQ satellite manager starting...
[ 	0.006023s]  INFO(satman): gateware ident 8+b1ddb89e;brittonlab-shared-DRTIO
[ 	0.018252s]  INFO(libboard_zynq::i2c): PCA9548 detected
[ 	0.532068s]  INFO(libboard_artiq::si5324): waiting for Si5324 lock...
[ 	2.484470s]  INFO(libboard_artiq::si5324):   ...locked
[ 	2.589678s]  INFO(satman): Switching SYS clocks...
[ 	2.644464s]  INFO(satman): SYS CLK switched successfully
[ 	2.669859s]  INFO(satman): SED spreading disabled by default
[ 	2.675677s]  INFO(satman): uplink is up, switching to recovered clock
[ 	2.721676s]  INFO(libboard_artiq::si5324): waiting for Si5324 lock...
[ 	4.443388s]  INFO(libboard_artiq::si5324):   ...locked
[ 	8.361326s]  INFO(libboard_artiq::si5324::siphaser): calibration successful, lead: 300, width: 438 (351deg)
[ 	8.688303s]  INFO(satman): TSC loaded from uplink
[ 	8.703644s]  INFO(satman): rank: 1
[ 	8.707117s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; }

I see no additional messages when running the above bug example; neither a 

> resetting RTIO

nor 

> failed to issue RTIO reset

message.

Spaqin, 2025-04-10:
Embarrassing oversight indeed. RISC-V based Kaslis will schedule an RTIO reset in satellites, but Kasli-SoC will reset [itself only](https://git.m-labs.hk/M-Labs/artiq-zynq/src/branch/release-8/src/libksupport/src/rtio_csr.rs#L21-L25).

Spaqin, 2025-04-11:
Fixed in [artiq-zynq release-8](https://git.m-labs.hk/M-Labs/artiq-zynq/commit/44f7fa3049be0cefbff1d427e192708da463cfb3). Please update the firmware @connorgoham.

connorgoham, 2025-04-11:
Thanks, unfortunately it seems I am now unable to build the new firmware.

Running the [standard nix command](https://m-labs.hk/artiq/manual/building_developing.html#kasli-soc-or-zc706-artiq-on-zynq) for release-8 now yields an error for both master & satellite devices.

My (respective) commands:

`nix build --print-build-logs --impure --expr 'let fl = builtins.getFlake "git+https://git.m-labs.hk/m-labs/artiq-zynq?ref=release-8"; in (fl.makeArtiqZynqPackage {target="kasli_soc"; variant="master"; json=/home/rabi/experiment/legacy-trap/device_descriptions/master.json;}).kasli_soc-master-sd'`

`nix build --print-build-logs --impure --expr 'let fl = builtins.getFlake "git+https://git.m-labs.hk/m-labs/artiq-zynq?ref=release-8"; in (fl.makeArtiqZynqPackage {target="kasli_soc"; variant="satellite"; json=/home/rabi/experiment/legacy-trap/device_descriptions/satellite.json;}).kasli_soc-satellite-sd'`

Common error message to both above commands:

> firmware>	Compiling libasync v0.0.0 (/nix/store/53dsvdsnjvp26v3q630if2f67rmcjqfc-source/libasync)
firmware>	Compiling libboard_zynq v0.0.0 (/nix/store/53dsvdsnjvp26v3q630if2f67rmcjqfc-source/libboard_zynq)
firmware>	Compiling libconfig v0.1.0 (/nix/store/53dsvdsnjvp26v3q630if2f67rmcjqfc-source/libconfig)
firmware>	Compiling futures-macro v0.3.25
firmware>	Compiling num-derive v0.3.3
firmware>	Compiling async-recursion v0.3.2
firmware>	Compiling io v0.0.0 (/build/src/libio)
firmware>	Compiling dwarf v0.0.0 (/build/src/libdwarf)
firmware> error[E0433]: failed to resolve: use of undeclared crate or module `eh_artiq`
firmware>	--> libksupport/src/kernel/api.rs:246:35
firmware> 	|
firmware> 246 |     	api!(__nac3_personality = eh_artiq::artiq_personality),
firmware> 	|                               	^^^^^^^^ use of undeclared crate or module `eh_artiq`
firmware>
firmware> error[E0433]: failed to resolve: use of undeclared crate or module `eh_artiq`
firmware>	--> libksupport/src/kernel/api.rs:247:29
firmware> 	|
firmware> 247 |     	api!(__nac3_raise = eh_artiq::raise),
firmware> 	|                         	^^^^^^^^ use of undeclared crate or module `eh_artiq`
firmware>
firmware> error[E0433]: failed to resolve: use of undeclared crate or module `eh_artiq`
firmware>	--> libksupport/src/kernel/api.rs:248:30
firmware> 	|
firmware> 248 |     	api!(__nac3_resume = eh_artiq::resume),
firmware> 	|                          	^^^^^^^^ use of undeclared crate or module `eh_artiq`
firmware>
firmware> error[E0433]: failed to resolve: use of undeclared crate or module `eh_artiq`
firmware>	--> libksupport/src/kernel/api.rs:249:33
firmware> 	|
firmware> 249 |     	api!(__nac3_end_catch = eh_artiq::end_catch),
firmware> 	|                             	^^^^^^^^ use of undeclared crate or module `eh_artiq`
firmware>
firmware> error[E0433]: failed to resolve: use of undeclared crate or module `eh_artiq`
firmware>	--> libksupport/src/kernel/api.rs:252:36
firmware> 	|
firmware> 252 |     	api!(__artiq_personality = eh_artiq::artiq_personality),
firmware> 	|                                	^^^^^^^^ use of undeclared crate or module `eh_artiq`
firmware>
firmware> error[E0433]: failed to resolve: use of undeclared crate or module `eh_artiq`
firmware>	--> libksupport/src/kernel/api.rs:253:30
firmware> 	|
firmware> 253 |     	api!(__artiq_raise = eh_artiq::raise),
firmware> 	|                          	^^^^^^^^ use of undeclared crate or module `eh_artiq`
firmware>
firmware> error[E0433]: failed to resolve: use of undeclared crate or module `eh_artiq`
firmware>	--> libksupport/src/kernel/api.rs:254:31
firmware> 	|
firmware> 254 |     	api!(__artiq_resume = eh_artiq::resume),
firmware> 	|                           	^^^^^^^^ use of undeclared crate or module `eh_artiq`
firmware>
firmware> error[E0433]: failed to resolve: use of undeclared crate or module `eh_artiq`
firmware>	--> libksupport/src/kernel/api.rs:255:34
firmware> 	|
firmware> 255 |     	api!(__artiq_end_catch = eh_artiq::end_catch),
firmware> 	|                              	^^^^^^^^ use of undeclared crate or module `eh_artiq`
firmware>
firmware> error[E0433]: failed to resolve: use of undeclared crate or module `eh_artiq`
firmware>	--> libksupport/src/kernel/api.rs:328:39
firmware> 	|
firmware> 328 |     	api!(test_exception_id_sync = eh_artiq::test_exception_id_sync)
firmware> 	|                                   	^^^^^^^^ use of undeclared crate or module `eh_artiq`
firmware>
firmware> error: aborting due to 9 previous errors
firmware>
firmware> For more information about this error, try `rustc --explain E0433`.
firmware> error: could not compile `ksupport`
firmware>
firmware> To learn more, run the command again with --verbose.
firmware> warning: build failed, waiting for other jobs to finish...
firmware> error: build failed
firmware> make: *** [Makefile:24: ../build/firmware/armv7-none-eabihf/release/runtime] Error 101
error: builder for '/nix/store/fdwyb4m9pkcl5mmqwd3ncqz9vvkkpilw-firmware.drv' failed with exit code 2
error: 1 dependencies of derivation '/nix/store/3agxgr3ypfjhwap423q72k5vxzqjsrgw-kasli_soc-master-sd.drv' failed to build

I have confirmed that I can still successfully build firmware with the [previous commit](https://git.m-labs.hk/M-Labs/artiq-zynq/commit/fada5b22b065079984786896adbd6ab6456c7727) using the commands:

`nix build --print-build-logs --impure --expr 'let fl = builtins.getFlake "git+https://git.m-labs.hk/m-labs/artiq-zynq?ref=fada5b22b0"; in (fl.makeArtiqZynqPackage {target="kasli_soc"; variant="master"; json=/home/rabi/experiment/legacy-trap/device_descriptions/master.json;}).kasli_soc-master-sd'`

`nix build --print-build-logs --impure --expr 'let fl = builtins.getFlake "git+https://git.m-labs.hk/m-labs/artiq-zynq?ref=fada5b22b0"; in (fl.makeArtiqZynqPackage {target="kasli_soc"; variant="satellite"; json=/home/rabi/experiment/legacy-trap/device_descriptions/satellite.json;}).kasli_soc-satellite-sd'`

Spaqin, 2025-04-12:
My bad, minor errors sneaked in while merging the changes from master. [Builds now](https://nixbld.m-labs.hk/build/178419).