m-labs/artiq #2720: Calling Core Reset with Satellite Bricks Devices

connorgoham, 2025-04-14:
# Bug Report

## One-Line Summary

Calling `core.reset()` with satellite connected either bricks the core devices requiring power cycling (if called during startup kernel) or results in `ERROR(runtime::comms): unexpected message from core1: RtioInitRequest` & `WARN(runtime::comms): connection terminated: unrecognized packet`(if called in experiments).

## Issue Details

### Steps to Reproduce

1. Flash Kasli SoCs with latest firmware.
2. Attempt to call `self.core.reset()`

### Expected Behavior

Core reset without throwing an error.

### Actual (undesired) Behavior
Error `ERROR(runtime::comms): unexpected message from core1: RtioInitRequest` & `WARN(runtime::comms): connection terminated: unrecognized packet`

Logs when booting each device separately:

* master log message when booting **without** satellite connection:

> rabi@legacy0:~/experiment/legacy-trap$ artiq_coremgmt log
[     0.000067s]  INFO(runtime): NAR3/Zynq7000 starting...
[     0.005242s]  INFO(runtime): gateware ident: 8+6287e5b4;brittonlab-legacy-trap
[     0.017640s]  INFO(libboard_zynq::i2c): PCA9548 detected
[     0.181435s]  INFO(runtime::rtio_clocking): using 100MHz reference to make 125MHz RTIO clock with PLL
[     0.554578s]  INFO(libboard_artiq::si5324): waiting for Si5324 lock...
[     2.704952s]  INFO(libboard_artiq::si5324):   ...locked
[     2.760995s]  INFO(runtime::rtio_clocking): SYS CLK switched successfully
[     2.772960s]  INFO(libboard_zynq::i2c): PCA9548 detected
[     2.810626s]  INFO(runtime::comms): network addresses: MAC=fc-0f-e7-07-7b-df IPv4=192.168.1.76 IPv6-LL=fe80::fe0f:e7ff:fe07:7bdf IPv6: no configured address
[     2.842344s]  INFO(libboard_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; }
[     2.854581s]  INFO(runtime::rtio_mgt): SED spreading disabled by default
[     2.874256s]  INFO(runtime::comms): Loading startup kernel...
[     2.880047s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[     2.886695s]  INFO(runtime::comms): Starting startup kernel...
[     2.892560s]  INFO(ksupport::kernel::core1): kernel starting
[    64.755342s]  INFO(ksupport::kernel::core1): kernel finished
[    64.760990s]  INFO(runtime::comms): Startup kernel finished!
[    64.766781s]  INFO(runtime::comms): Loading idle kernel
[    64.772045s]  INFO(runtime::comms): Running idle kernel
[    64.777255s]  INFO(ksupport::kernel::core1): kernel starting
[    65.265975s]  INFO(libboard_zynq::eth): eth: got Link { speed: S1000, duplex: Full }
[   103.219630s]  INFO(runtime::mgmt): received connection

* satellite UART log message when booting **without** connection to the master:

> rabi@legacy0:~/experiment/legacy-trap$ cat /dev/ttyUSB9
> 
> 
>                      __________   __
>                     / ___/__  /  / /
>                     \__ \  / /  / /
>                    ___/ / / /__/ /___
>                   /____/ /____/_____/
> 
>                  (C) 2020-2022 M-Labs
> 
> [     0.019992s]  INFO(szl): Simple Zynq Loader starting...
> [     0.025199s] DEBUG(libboard_zynq::clocks::source): Set ARM_PLL to 2000000000 Hz
> [     0.007040s] DEBUG(libboard_zynq::clocks::source): Set IO_PLL to 1000000000 Hz
> [     0.016259s] DEBUG(libboard_zynq::clocks::source): Set DDR_PLL to 1066666666 Hz
> [     0.023608s] DEBUG(libboard_zynq::ddr): DDR 3x/2x clocks: 533333328/355555552
> [     0.030777s] DEBUG(libboard_zynq::ddr): DDR DCI clock: 10062892 Hz (divisors=2*53)
> [     0.042001s] DEBUG(libboard_zynq::sdio): Reset SDIO!
> [     0.046947s] DEBUG(libboard_zynq::sdio): Changing clock frequency to 400000
> [     0.053895s]  INFO(szl): Card inserted. Mounting file system.
[... 208 lines cut ...]
> [     1.424945s]  INFO(szl): executing payload
> [     0.000067s]  INFO(satman): ARTIQ satellite manager starting...
> [     0.006023s]  INFO(satman): gateware ident 8+6287e5b4;brittonlab-shared-DRTIO
> [     0.018332s]  INFO(libboard_zynq::i2c): PCA9548 detected
> [     0.532146s]  INFO(libboard_artiq::si5324): waiting for Si5324 lock...
> [     2.506615s]  INFO(libboard_artiq::si5324):   ...locked
> [     2.611821s]  INFO(satman): Switching SYS clocks...
> [     2.666684s]  INFO(satman): SYS CLK switched successfully
> [     2.692877s]  INFO(satman): SED spreading disabled by default
> [     2.698691s]  INFO(satman): uplink is up, switching to recovered clock
> [     2.744624s]  INFO(libboard_artiq::si5324): waiting for Si5324 lock...
> [     4.466444s]  INFO(libboard_artiq::si5324):   ...locked
> [     9.177119s]  INFO(libboard_artiq::si5324::siphaser): calibration successful, lead: 443, width: 439 (352deg)
> [     9.504615s]  INFO(satman): TSC loaded from uplink
> [     9.520634s]  INFO(satman): rank: 1
> [     9.524105s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; }
> [    10.130720s]  INFO(satman): resetting RTIO
> [   485.320605s]  INFO(satman): resetting RTIO

### Your System (omit irrelevant parts)

* Operating System: `Ubuntu 22.04.5 LTS`
* ARTIQ version: `ARTIQ v8.9006+17be223`
* Version of the gateware and runtime loaded in the core device: master: `gateware ident: 8+6287e5b4;brittonlab-legacy-trap`; satellite: `gateware ident 8+6287e5b4;brittonlab-shared-DRTIO`
* Hardware involved: both master & satellite are Kasli SoC v1.1.1

Spaqin, 2025-04-15:
Generally speaking, this seems very suspicious. That particular message shows up when the response for core1 is wrong... during loading of the kernel. The kernel is not running yet. And yet, it would suddenly pop up...? As if one of the core1 kernel loading messages would get lost for whatever reason.

Anyway, I tried to reproduce locally it based on newest [Hydra build](https://nixbld.m-labs.hk/build/178411). The experiment is very simple as well:
<details>
  <summary>Experiment</summary>

```python
from artiq.experiment import *

class ResetTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.core.reset()
```

</details>

I compiled this experiment and installed it as a startup kernel. See the master logs:

<details>
  <summary>Master logs</summary>

```
[     0.733321s]  INFO(szl): Preparing for runtime execution
[     0.738955s]  INFO(szl): executing payload
[     0.000067s]  INFO(runtime): NAR3/Zynq7000 starting...
[     0.005161s]  INFO(runtime): gateware ident: 8+6287e5b4;master
[     0.016243s]  INFO(libboard_zynq::i2c): PCA9548 detected
[     0.175139s]  WARN(runtime::rtio_clocking): error reading configuration. Falling back to default.
[     0.183992s]  WARN(runtime::rtio_clocking): Using default configuration - internal 125MHz RTIO clock.
[     0.193192s]  INFO(runtime::rtio_clocking): using internal 125MHz RTIO clock
[     0.583950s]  INFO(libboard_artiq::si5324): waiting for Si5324 lock...
[     3.746144s]  INFO(libboard_artiq::si5324):   ...locked
[     3.801993s]  INFO(runtime::rtio_clocking): SYS CLK switched successfully
[     3.813971s]  INFO(libboard_zynq::i2c): PCA9548 detected
[     3.849941s]  INFO(runtime::comms): network addresses: MAC=fc-0f-e7-10-18-73 IPv4=192.168.1.198 IPv6-LL=fe80::fe0f:e7ff:fe10:1873 IPv6: no configured address
[     3.878543s]  INFO(libboard_artiq::drtio_routing): could not read routing table from configuration, using default
[     3.888720s]  INFO(libboard_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; 3: 3 0; 4: 4 0; }
[     3.902234s]  INFO(runtime::rtio_mgt): SED spreading disabled by default
[     3.909759s]  WARN(ksupport): error reading device map (Configuration key `device_map` not found), device names will not be available in RTIO error messages
[     3.927765s]  INFO(runtime::comms): Loading startup kernel...
[     3.933541s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link RX became up, pinging
[     3.941058s]  INFO(runtime::comms): Starting startup kernel...
[     3.946885s]  INFO(ksupport::kernel::core1): kernel starting
[     3.952536s]  INFO(ksupport::kernel::core1): kernel finished
[    10.727980s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] remote replied after 36 packets
[    10.919305s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link initialization completed
[    10.927039s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[    10.934372s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[    10.941056s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 128
[    10.951726s]  INFO(runtime::comms): Startup kernel finished!
[    11.457082s]  INFO(libboard_zynq::eth): eth: got Link { speed: S1000, duplex: Full }
[    28.799200s]  INFO(ksupport::kernel::core1): kernel starting
[    28.804856s]  INFO(ksupport::kernel::core1): kernel finished
[    28.810688s]  INFO(runtime::comms): peer closed connection
```

</details>

Satellite is a Kasli 2.0 with ARTIQ-9, but given the origin of the error (and that the master is literally never aware of the satellite's board/firmware) it should not matter.

<details>
  <summary>Satellite logs</summary>

```
[     0.005734s]  INFO(satman): ARTIQ satellite manager starting...
[     0.010339s]  INFO(satman): software ident 9.0+unknown.beta;satellite
[     0.016771s]  INFO(satman): gateware ident 9.0+unknown.beta;satellite
[     0.023205s]  INFO(satman): log level set to INFO by default
[     0.028837s]  INFO(satman): UART log level set to INFO by default
[     0.045472s]  INFO(board_misoc::io_expander): MCP23017 io expander 0 not found. Checking for PCA9539.
[     0.074055s]  INFO(board_misoc::io_expander): MCP23017 io expander 1 not found. Checking for PCA9539.
[     0.216639s]  INFO(satman): Clocking has already been set up.
[     0.221040s]  INFO(satman): SED spreading disabled by default
[     0.227717s]  INFO(satman): uplink is up, switching to recovered clock
[     0.261077s]  INFO(board_artiq::si5324): waiting for Si5324 lock...
[     2.004955s]  INFO(board_artiq::si5324):   ...locked
[     4.274001s]  INFO(board_artiq::si5324::siphaser): calibration successful, lead: 6, width: 433 (347deg)
[     4.598507s]  INFO(satman): TSC loaded from uplink
[     4.770560s]  INFO(satman): rank: 1
[     4.772709s]  INFO(satman): routing table: RoutingTable { 0: 0; 1: 1 0; 2: 2 0; 3: 3 0; 4: 4 0; }
[    10.552344s]  INFO(satman): resetting RTIO
```

</details>

connorgoham, 2025-04-16:
The above core logs appear to be an issue with running the startup kernel (mine of which invokes core.reset() once or twice, depending upon if a satellite connection is established.)

I have verified that upon removing my startup kernel, I am able to boot the devices & the core.reset() now works as expected between the master & satellite without the aforementioned errors.

I will open a separate issue about the startup kernel if warranted upon further investigation.

closed 2025-04-16 by connorgoham

connorgoham, 2025-04-17:
Follow up issue #2729