m-labs/artiq #2729: Startup Kernel With DRTIO Causes Core Panic

connorgoham, 2025-04-17:
# Bug Report

## One-Line Summary

If a startup kernel is flashed on the master device & a satellite is connected, the device panics upon (re)booting.

## Issue Details

### Steps to Reproduce

Minimal startup kernel to reproduce:

```
from artiq.experiment import *

class StartupKernel(EnvExperiment):
    """Startup Kernel Minimal Bug Example"""

    def build(self):
        self.setattr_device("core")
        self.setattr_device("led1")

    @kernel
    def run(self):
        self.core.reset()
        self.led1.on()
        self.core.wait_until_mu(now_mu())
```

Compiled with `artiq_compile ...`

Flashed with `artiq_coremgmt config write -f startup_kernel ...`

### Expected Behavior

Startup kernel completes without error, irrespective of satellite connection.

### Actual (undesired) Behavior

If the satellite is _not_ connected to the master, the core device boots without error (logs of this case included at bottom).

However, when a satellite device _is_ connected, the master often panics & hangs, requiring the power to be cycled and the device to be allowed to reboot without the satellite connected. Rarely, it seems the master can successfully boot with the satellite connected & a startup kernel flashed, however I am not sure why sometimes it suceeds when often it fails.

If no startup kernel is flashed, master reboots without error, irrespective of satellite connection.

* **Master** UART log (satellite device is connected & startup kernel flashed):

> rabi@legacy0:~/experiment/legacy-trap$ cat /dev/ttyUSB2
> [	69.392343s]  INFO(runtime::mgmt): received connection
> [	69.397939s]  INFO(runtime::mgmt): rebooting
> 
> 
>                  	__________   __
>                 	/ ___/__  /  / /
>                 	\__ \  / /  / /
>                	___/ / / /__/ /___
>               	/____/ /____/_____/
> 
>              	(C) 2020-2022 M-Labs
> 
> [ 	0.019988s]  INFO(szl): Simple Zynq Loader starting...
> [ 	0.025194s] DEBUG(libboard_zynq::clocks::source): Set ARM_PLL to 2000000000 Hz
> [ 	0.007039s] DEBUG(libboard_zynq::clocks::source): Set IO_PLL to 1000000000 Hz
> [ 	0.016259s] DEBUG(libboard_zynq::clocks::source): Set DDR_PLL to 1066666666 Hz
> [ 	0.023608s] DEBUG(libboard_zynq::ddr): DDR 3x/2x clocks: 533333328/355555552
> [ 	0.030776s] DEBUG(libboard_zynq::ddr): DDR DCI clock: 10062892 Hz (divisors=2*53)
> [ 	0.042000s] DEBUG(libboard_zynq::sdio): Reset SDIO!
> [ 	0.046947s] DEBUG(libboard_zynq::sdio): Changing clock frequency to 400000
> [ 	0.053894s]  INFO(szl): Card inserted. Mounting file system.
> [ 	0.082002s] DEBUG(libboard_zynq::sdio): Changing clock frequency to 25000000
[... 124 lines cut ...]
> [ 	0.181319s]  INFO(runtime::rtio_clocking): using 100MHz reference to make 125MHz RTIO clock with PLL
> [ 	0.554511s]  INFO(libboard_artiq::si5324): waiting for Si5324 lock...
> [ 	2.727012s]  INFO(libboard_artiq::si5324):   ...locked
> [ 	2.782995s]  INFO(runtime::rtio_clocking): SYS CLK switched successfully
> [ 	2.794966s]  INFO(libboard_zynq::i2c): PCA9548 detected
> [ 	2.832576s]  INFO(runtime::comms): network addresses: MAC=fc-0f-e7-07-7b-df IPv4=192.168.1.76 IPv6-LL=fe80::fe0f:e7ff:fe07:7bdf IPv6: no configured address
> [ 	2.863356s]  INFO(libboard_artiq::drtio_routing): routing table: RoutingTable { 0: 0; 1: 1 0; }
> [ 	2.873479s]  INFO(runtime::rtio_mgt): SED spreading disabled by default
> [ 	2.886753s]  INFO(runtime::comms): Loading startup kernel...
> [ 	2.892511s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
> [ 	2.899237s]  INFO(runtime::comms): Starting startup kernel...
> [ 	2.905067s]  INFO(ksupport::kernel::core1): kernel starting
> [ 	2.910844s]  INFO(ksupport::kernel::core1): kernel finished
> [ 	2.916494s]  INFO(runtime::comms): Startup kernel finished!
> [ 	2.922248s]  INFO(runtime::comms): Loading idle kernel
> [ 	2.927519s]  INFO(runtime::comms): Running idle kernel
> [ 	2.932731s]  INFO(ksupport::kernel::core1): kernel starting
> [ 	6.922089s]  INFO(libboard_zynq::eth): eth: got Link { speed: S1000, duplex: Full }

### Your System (omit irrelevant parts)

* Operating System: `Ubuntu 22.04.5 LTS`
* ARTIQ version: `ARTIQ v8.9006+17be223`
* Version of the gateware and runtime loaded in the core device: master: `gateware ident: 8+6287e5b4;brittonlab-legacy-trap`; satellite: `gateware ident 8+6287e5b4;brittonlab-shared-DRTIO`
* Hardware involved: both master & satellite are Kasli SoC v1.1.1

jbqubit, 2025-04-24:
Thanks for the [patch](https://git.m-labs.hk/M-Labs/artiq-zynq/pulls/381) [mwojcik](https://git.m-labs.hk/mwojcik)! Please test this on an M-Labs setup where both master & satellite are Kasli SoC before closing.

Spaqin, 2025-04-25:
Tested with two Kasli-SoCs, this doesn't panic anymore.

```
[     3.912885s]  INFO(runtime::comms): Loading startup kernel...
[     3.918631s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link RX became up, pinging
[     3.926117s]  INFO(runtime::comms): Starting startup kernel...
[     3.931938s]  INFO(ksupport::kernel::core1): kernel starting
[    10.393981s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] remote replied after 32 packets
[    10.412461s] ERROR(runtime::comms): Uncaught kernel exceptions: [Some(Exception 2 from (Rust function) in libksupport/src/kernel/rtio_csr.rs:90:9, message: RTIO destination unreachable, output, at {0} mu, channel 0x0035:unknown)]
[    10.432779s]  INFO(runtime::comms): Startup kernel finished!
[    10.441769s]  INFO(runtime::comms): loading idle kernel
[    10.447013s]  INFO(runtime::comms): running idle kernel
[    10.452222s]  INFO(ksupport::kernel::core1): kernel starting
[    10.471839s]  INFO(runtime::rtio_mgt::drtio): [LINK#0] link initialization completed
[    10.479564s]  INFO(runtime::rtio_mgt::drtio): [DEST#0] destination is up
[    10.486268s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] destination is up
[    10.492949s]  INFO(runtime::rtio_mgt::drtio): [DEST#1] buffer space is 128
[    10.938081s]  INFO(libboard_zynq::eth): eth: got Link { speed: S1000, duplex: Full }
```

The RTIODestinationUnreachable exception is interesting in its own way but it's not part of this issue.

closed 2025-04-25 by Spaqin

jbqubit, 2025-04-25:
> The RTIODestinationUnreachable exception is interesting in its own way but it's not part of this issue.

How do you know it's not part of this issue? `RTIODestinationUnreachable` is only reported in one [other](https://github.com/m-labs/artiq/issues/1521) Issue so I would surmise it's rare.