m-labs/artiq #2731: Presence of idle kernel results in `unexpected message from core1: RtioInitRequest` upon interruption

connorgoham, 2025-04-18:
# Bug Report

## One-Line Summary

If there is an idle kernel flashed onto the device which runs perpetually, subsequent kernels raise an error.

## Issue Details

### Steps to Reproduce

1. Flash core device with minimal (perpetual) idle kernel:

```
from artiq.experiment import *

class IdleKernel(EnvExperiment):

    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.core.reset()
        while True:
            pass
```

2. Run a minimal experiment which interrupts the idle kernel on the core device:

```
from artiq.experiment import *

class InterruptIdleKernel(EnvExperiment):
    """Interrupt Idle Kernel Bug Example"""

    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        pass
```

### Expected Behavior

Idle kernel is interrupted without error.

### Actual (undesired) Behavior

Experiment fails, resulting in an `unexpected message from core1: RtioInitRequest` error and a `connection terminated: unrecognized packet` warning. UART log from master given two subsequent attempts to run InterruptIdleKernel via `artiq_run`:

> [   452.667443s] ERROR(runtime::comms): unexpected message from core1: RtioInitRequest
> [   452.675003s]  WARN(runtime::comms): connection terminated: unrecognized packet
> [   452.682213s]  INFO(runtime::comms): Loading idle kernel
> [   452.689272s]  INFO(runtime::comms): Running idle kernel
> [   452.694489s]  INFO(ksupport::kernel::core1): kernel starting
> [   462.890897s] ERROR(runtime::comms): unexpected message from core1: RtioInitRequest
> [   462.898455s]  WARN(runtime::comms): connection terminated: unrecognized packet
> [   462.905665s]  INFO(runtime::comms): Loading idle kernel
> [   462.910984s]  INFO(runtime::comms): Running idle kernel
> [   462.916195s]  INFO(ksupport::kernel::core1): kernel starting

Probably this is what I mistook for #2720 & related to #2729

### Your System (omit irrelevant parts)

* Operating System: `Ubuntu 22.04.5 LTS`
* ARTIQ version: `ARTIQ v8.9006+17be223`
* Version of the gateware and runtime loaded in the core device: master: `gateware ident: 8+6287e5b4;brittonlab-legacy-trap`; satellite: `gateware ident 8+6287e5b4;brittonlab-shared-DRTIO`
* Hardware involved: both master & satellite are Kasli SoC v1.1.1

``

jbqubit, 2025-04-24:
Thanks for the [patch](https://git.m-labs.hk/M-Labs/artiq-zynq/pulls/381), [mwojcik](https://git.m-labs.hk/mwojcik)! 

Please test this on an M-Labs setup where both master & satellite are Kasli SoC before closing. Is there a DRTIO hierarchy involving Kasli SoC that is part of your continuous integration testing?

Spaqin, 2025-04-25:
Also tested with two Kasli-SoCs.

For the CI, at the moment we have a RISC-V and a Zynq system, but it's zc706 and kc705 respectively. There's also no DRTIO involved. However this issue wouldn't be caught as the HITL tests do not touch on startup or idle kernels - the behavior wasn't exactly dependent on it being an idle/startup kernel, but rather the kernel itself being interrupted (it would've also happened with e.g. a normal kernel with an infinite loop).

closed 2025-04-25 by Spaqin

jbqubit, 2025-04-25:
@Spaqin Thank you for testing with the Kasli-SoC!

jbqubit, 2025-04-28:
In response to this bug you made [this](https://git.m-labs.hk/M-Labs/artiq-zynq/commit/209eba33f0e2aa58e33fcb01ded9f87d22794167) edit to Zynq. Now we see a ```proto_rev``` mismatch error when calling ```cpld.init()``` on the satellite. There is no ```proto_rev``` error when calling ```cpld.init()``` on the master.

jbqubit, 2025-04-28:
I suspect this is an artifact of the Urukul ```cpld.init()``` having an implementation error. https://github.com/m-labs/artiq/issues/2739