m-labs/artiq #2854: High Fastino jitter

atseml, 2025-09-05:
When compared to a TTL, there seems to be a cyclically reducing delay when setting Fastino voltages, which manifests as jitter at the macro level on the order of around ±200 ns, way higher than the Fastino gateware's [7 ns](https://github.com/quartiq/fastino/blob/91e6df4dcdfdcec67aaa606194bef245bbb716db/README.md?plain=1#L39).

Reproducing Experiment:

```python
class FastinoJitter(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")
        self.setattr_device("fastino0")

    @kernel
    def run(self):
        self.core.reset()
        self.fastino0.init()
 
        while True:
            self.ttl0.on()
            self.fastino0.set_dac(0, 9.9)
            delay(25*ms)

            self.ttl0.off()
            self.fastino0.set_dac(0, 0.0)
            delay(25*ms)

```

On an oscilloscope (with C1: `fastino0` channel 0, C3: `ttl0`):

https://github.com/user-attachments/assets/6a9633c3-e15b-4bb8-b677-30955eab469f

jordens, 2025-09-05:
That's how sampled systems work.

closed 2025-09-05 by jordens

atseml, 2025-09-18:
> That's how sampled systems work.

If by sampled system you mean the Fastino end then I don't think this has to do with that. I've added probes to the EEM0_1_P (C2) and EEM0_1_N (C4) lines on the Fastino, here it is on the oscilloscope along with the previous output:

https://github.com/user-attachments/assets/38fc82ba-fedb-42ce-91a4-bd2d2c7c9b10

It looks like the drifting (relative to the TTL) signals actually originate from the controlling core device, and is not the product of the Fastino itself. So, this is a gateware issue on the Kasli side, is it not?